MarlinFirmware/Marlin #5847: CUSTOM_MACHINE_NAME overflows to top of LCD (minor issue)

pearson, 2017-02-16:
This is a minor issue.

If the CUSTOM_MACHINE_NAME is too long, the word "ready" overflows to the top of the LCD panel, obscuring the extruder temperature. For example, the first attached image shows:
`#define CUSTOM_MACHINE_NAME "RCBugfix 16 Feb 2017"`

[image]

However, if the CUSTOM_MACHINE_NAME is absurdly long, it starts to corrupt the entire LCD panel. For example, the second attached image shows:
`#define CUSTOM_MACHINE_NAME "RCBugfix 16 Feb 2017 RCBugfix 16 Feb 2017 RCBugfix 16 Feb 2017"`

[image]

This is on a 20x4 LCD panel using:
`#define REPRAP_DISCOUNT_SMART_CONTROLLER`

Perhaps a sanity check for the CUSTOM_MACHINE_NAME length might be a good fix?

Roxy-3D, 2017-02-17:
> Perhaps a sanity check for the CUSTOM_MACHINE_NAME length might be a good fix?

Yes!  But I think we should also make the code more robust so it doesn't do bad things like that.   These type of problems are at the root of many memory corruption bugs.

Tannoo, 2017-02-17:
Maybe not checking but, how about truncating at 14 characters?

Blue-Marlin, 2017-02-17:
Please be aware of:
* we have multiple languages. The length of the "ready" equivalent may differ.
* we use utf8 here. The space used is not srtlen().

Tannoo, 2017-02-17:
Yeah...so truncate accordingly.

thinkyhead, 2017-02-18:
If we make our UTF-8-savvy string functions `constexpr` then the pre-compiler will be able to use them in `static_assert`.

thinkyhead, 2017-02-18:
A man goes to a doctor.
Man: "Doctor, it hurts when I do this."
Doctor: "Then don't do that. That'll be $75!"

closed 2017-10-31 by thinkyhead