MarlinFirmware/Marlin #5849: 32-Bit-RcBugFix Rebase

p3p, 2017-02-17:
I made a pull request to clarify the best way to go about bringing 32-Bit-RcBugFix inline with RcBugFix, but I'm inexperienced with remote rebasing and thought it would be best moved to an issue to ask how I should move forward.

closed 2017-02-18 by p3p

thinkyhead, 2017-02-19:
Try merge first, and if it's very tricky to figure out the resolutions then try rebase. If rebase doesn't work then the next method would be to hand-compare the 32 bit branch to its base and apply equivalent changes to the current `RCBugFix` code in a separate editor. That will probably be easiest.

Once in a while, if changes are not huge, and you know what to expect, you can scattershot the all differences, reverting a copy of the current code…
```
git checkout RCBugFix -b comp_32bit_to_now
git diff 32bit-RCBugFix comp_32bit_to_now | git apply
```
…then you use Github Desktop to cherry-pick the changes that apply only to 32-bit, and throw away all the code that reverts back to the past.

p3p, 2017-02-19:
yea I did some research after posting this and decided that going through the merge process would be best as rebasing should only really be done with a private branches, it wasn't actually as bad as I thought it would be. I already put the new pull request up that has no issues merging into 32-Bit-RcBugFix, I really need to get a Due I guess, when I can afford it, as I can only test it on my STM32F7 atm (works well though).