MarlinFirmware/Marlin #5851: RC8 Freezinh/crashing while using Filament Width Sensor

jmceara, 2017-02-17:
Hi.
Everytime time that I enable filament sensor (M405), my printing freezes and RAMPS/Arduino get unresponsive (nothing works, need to remove power). If I just don't enable filament sensor (M405), printer goes fine.
Not that all the time filament sensor still pluged and powered. I can see filament width on display, but without M405, correction factor is not applied and printer works fine. My guess is something related to memory leak. Also, the time that print freezes/stop is randon, but mostly about 30 minutes of printing.

Sebastianv650, 2017-02-18:
Are you using an rc version of marlin or the latest rcbugfix branch?

thinkyhead, 2017-02-18:
Could be related to a recently-fixed bug. Check `RCBugFix` and see if it's any better.

jmceara, 2017-02-19:
I was using RCBugFix from 2 days ago (the date when I opened this bug report). Any change from that date?

thinkyhead, 2017-02-26:
The patch to the sensor-reading code might have been that same day. The fix might have worked for your case, but we can't be sure. It may be related to the planner logic or ring buffer rather than the sensor reading code.

jmceara, 2017-03-01:
I didn't used any external patch, only RCBugFix branch and just to clarify, it didn't worked, still with same crash/freeze bug when enabling sensor code with M405.

jmceara, 2017-04-12:
Any neww on this? May I help?

Scratchbuild52, 2017-04-16:
Hi There,

       @thinkyhead , @jmceara is this problem addressed or do you still have the problem? I have tried now almost any configuration but the above described problem still exists. I have even monitored, per layer, the memory usage with M100 F and as far as I could see no memory problems. The controller just randomly decides to crash. As far as I could, I ruled out shorts on the filament sensor that might crash the controller, but then if it was something on the sensor it should also crash when the sensor is disabled, with M406, but still coupled. 

I also had a chat with Flipper, the designer, and he said that he also had the problem, and it was too little memory. He said he forfeited the LCD and it worked fine from there. I just cannot understand if M100 reports more than 2K memory buffer free, that the memory could be the problem. Or am I missing something with the free memory report from M100. On the other hand to forfeit my LCD controller to use the Filament Width Sensor would be going 500 steps back in the evolution of the 3D industry.

Configuration:
                    1) Ramps 1.4
                    2) Marlin V1.1.0-RC8 - 06/12/2016
                    3) Full Graphics Controller
                    4) Automatic Bed Leveling
                    5) Filament Width Sensor
                    6) Print from SD card

I did not try to use it without the LCD controller yet, but I will give that a try today, and see what happens.

Regards

Nic Scheepers

Roxy-3D, 2017-04-16:
> I just cannot understand if M100 reports more than 2K memory buffer free, that the memory could be the problem. 

Agreed.

> Or am I missing something with the free memory report from M100. 

No.  And the new M100 has more information and abilities.  The reason is, it is being used to chase down a memory corruption problem.   Depending upon what is in memory, that moves other things around in memory.  If you push something critical into the blast zone, you get bad results.

> On the other hand to forfeit my LCD controller to use the Filament Width Sensor would be going 500 steps back in the evolution of the 3D industry.

We need to find the cause of the data corruption.  And then you can use what ever you want so long as you don't totally fill up memory.

Please see:  https://github.com/MarlinFirmware/Marlin/issues/6331

Scratchbuild52, 2017-04-16:
Thanks for the quick reply, I will go through the other calls and see if I can pick up anything that might help. I will also load the latest RCbugfix and add the code provided in #6331 and #6334 and see if I find anything. Problem is random, and may only happen hours into the print.

Scratchbuild52, 2017-04-16:
Just some info: I never had G28 or G29 crash on me, but maybe they cause memory to be corrupted that is also used by the filament width sensor. Will also put a M100 D on every layer to see if and when memory is corrupted.

Scratchbuild52, 2017-04-17:
Hi there,

Unfortunately I do not have good news. I did not implement the changes in #6331 because I did not use UBL. So I figured I would just load the rcbugfix of yesterday and see if I get anywhere, without changing too much to keep the parameters that changed between tests the minimum. 

In any case I have added a M100 D and M100 F to every layer change and the stackpointer, free memory etc. stayed as constant as I could see up to the last layer it crashed in.  So a slow memory leak is not the perpetrator.

This was unfortunately hours into the print and was almost finished. I just thought the new rcbugfix build solved my problem the crash happened with nothing else in the log to give me a clue as to why. Can someone please just have a quick look through my configuration.h and see if anything is set up wrongly to make this happen.
[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/925141/Configuration.txt)

Scratchbuild52, 2017-04-17:
Hi There,

I have now also disabled the display in the configuration.h and the print still crashed. And it crashed very early into the print. I have checked the free memory again between each layer and still no corruption up to the layer before the crash.

Do you have anything specific for me to have a look at, or to try out. I will start going in depth into the code tomorrow but the problem is, that the crash is not consistently reproducible at a specific point of the print and would therefore be a challenge to catch.

Regards

Nic Scheepers

Roxy-3D, 2017-04-18:
And incidentally...  The graphical LCD Panels eat a lot more memory than the 20x4 panels.

Scratchbuild52, 2017-04-19:
Hi Roxy-3D,

Yes, it does take up a lot of memory, but as you can  see above, I disabled the Graphical screen and it still crashed. I did not test any further yet because I had to use the printer for real work :-) But just to report back, I have disabled the filament width sensor and I did not have a crash in 4 batches of printing.

But obviously my prints does not look as nice as they should.

Regards

Nic

Scratchbuild52, 2017-04-23:
Hi there,

         I was so hope-full when I saw the crashes with the UBL bed levelling has been solved, but unfortunately no change, when the filament width sensor is coupled and  enabled with M405, it crashes somewhere along the line.  I do not have to disable it in the config, I must just not enable it with M405. So it measures and displays the width on the LCD but does not use it, then it is happy and would not crash. 

I have now swopped out the filament width sensor with another one and it still crashes. There is no specific single point at which the crash happens so debugging it is very difficult because up to now I could not trace the problem to something specific.

So I have ruled out the following:
      1) Free memory corruption , checked with M100 on each level of printing.
      2) Free Memory leaks, checked with M100 on each level of printing.
      3) Graphical LCD - I have disabled it all together and it still crashed.
       4)Physical device - I have changed it with another one.
      5) Used the newest RCBugFix - 22Apr2017 - and still crashed after the memory corruption for UBL has been fixed. Also tested RC8 and a RCBugFix of 6 days ago and still crashing randomly.

And the worst is, it seems like I am the only one with this problem as there is no one chiming in.     

Regards

Nic

thinkyhead, 2017-04-26:
A "memory leak" implies use of `malloc` then forgetting to use `free`. But we don't do that. What we have here sounds more like a buffer overrun.

So, please post your `Configuration.h` and `Configuration_adv.h` files for testing. You may need to add `.txt` file extension before Github will allow the files to be dropped on the textfield.

> still crashing randomly

Only during printing, or at other times also?

Scratchbuild52, 2017-04-28:
Hi @thinkyhead,

Here is my configuration as I have it now. I must say I have changed one thing from the previous post I had, and that was the MEASUREMENT_DELAY_CM that I changed from 12.5 to 12. I thought it might have something to do with the resolution of the filament width buffer being 1cm but I am still testing, I had a crash right at the beginning this morning but I am not sure if it was the Filament Width Sensor or not. So I will see if it crashes again.

Herewith the configuration attached.

How do you set up debugging for this code without actually printing because dryrun does not produce any log entries if I do not actually print? Most probably because the delta e- counter in mm does not advance.

Regards

Nic Scheepers
[Configuration_adv.txt](https://github.com/MarlinFirmware/Marlin/files/963528/Configuration_adv.txt)

[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/963527/Configuration.txt)

Scratchbuild52, 2017-04-28:
Oh yes, it crashes only during printing with M405 i.e. filament sensor enabled. Otherwise it is happy. So I gathered that the problem occurs in Temperature.cpp but might have been a symptom of the planner.cpp code.

Scratchbuild52, 2017-04-28:
Unfortunately the change on the MEASUREMENT_DELAY_CM did not work it crashed twice this morning.

closed 2017-11-01 by thinkyhead