MarlinFirmware/Marlin #5852: Can't change the bed default settings pronterface

cookiz1, 2017-02-17:
Hello,
My settings don't allow me to change the default bed settings, I don't have the possibility to write my printer's bed sizes.
can someone help me ? I am a beginner with pronterface
Thank you

Roxy-3D, 2017-02-18:
I ran into this on a machine that was running Windows-10.    I was trying to setup my gMax 1.5+ which has a bed slightly smaller than a tennis court.  I really needed to change the default bed size.    I never got the menu to work.  But you can pull C:\Users\your-user-name\printrunconf.ini into your favorite text editor and zap what ever values you need into the file.   

I think you (and I) are see a problem with Python not doing edit windows correctly on Windows 10.

cookiz1, 2017-02-18:
Thank you ! I actually have a new issue now and I don't know if it's related, no my motors won't move and just make noise

Roxy-3D, 2017-02-19:
I am not aware of anything that would cause that in PronterFace.   But are you doing something to get the feed rate too high for your motors?   What else have you changed on your printer recently?

jfpion, 2017-02-21:
did you try to run pronterface as admin ? sometimes it solve problem

closed 2017-10-31 by thinkyhead

PJMoore, 2020-10-20:
> Hello,
> My settings don't allow me to change the default bed settings, I don't have the possibility to write my printer's bed sizes.
> can someone help me ? I am a beginner with pronterface
> Thank you
> [image]

Find printrunconf.ini, and edit the bed size with a text editor.  It worked fine for me on Win 10.