MarlinFirmware/Marlin #5853: M163, M164, T0 not working - MIXING_EXTRUDER

rjbalsi, 2017-02-18:
Hi

My config is a modified TRONXY X3A - changed the board to a RUMBA, and setup a diamond hot end.  

Using the Marin 1.1.0 RC7 - If I compile for a normal single extruder I can print fine (using the first extruder).

Then if I recompile and add MIXING_EXTRUDER - the prints don't work.  The LCD menu move EXTRUDER works - but GCODE will not.

Manually inserting M163, M164 and T0 command into the GCODE still results in the extruders not working.

But if I alter the GCODE G1 commands to include A1.0 - then the print will work fine.

Sample GCODE that does work
G1 X86.681 Y89.262 A1 E10

Sample GCODE that does not work
M163 S0 P1
M163 S1 P0
M163 S2 P0
M164 S0
T0
G1 X86.681 Y89.262 E10

thinkyhead, 2017-02-26:
Have you tested the current `RCBugFix` branch?

rjbalsi, 2017-03-19:
Is Marin 1.1.0 RC7 current?

JohnOCFII, 2017-03-20:
> Is Marin 1.1.0 RC7 current?

No.  RC8 was released in early December, and many, many continued improvements are in the branch known as RCBugFix.

thinkyhead, 2017-04-02:
@rjbalsi Did you test `1.1.0-RC8` or `RCBugFix` and is this issue now resolved?

closed 2018-02-26 by thinkyhead