MarlinFirmware/Marlin #5857: Pause in Movement with Bilinear Mesh leveling (RCbugFix from today)

mgroel, 2017-02-18:
Hi,
just updated to the current RCBugFix from today. Works OK so far, but I noticed a short pause in movement if having automatic bilinear bed leveling enabled. Movement stops if moving over the edges of the mesh it seems. Anybody having such an behaviour, too? I think printed parts will get marks on the perimeters if the extruder halts each and every moment.

If you need additional information, please let me know.

Roxy-3D, 2017-02-19:
I have not used the bi-linear mesh leveling.  But I can tell you the UBL system has an initial stutter on the very first move if it is a large move.   (Both Bi-Linear and UBL split the move up and do a Bi-Linear interpolation of the required Z correction as the path crosses mesh boundaries.)   That stutter is not there during an actual print.   What is happening is the large initial move is getting split.  And the when the first split happens, there is nothing else in the command queue.  So the planner calculates the move as if no more movement is going to happen.  But then the rest of initial line movement pieces get sent to the planner.   At that point the first piece is still in process and so subsequent pieces to the 2nd piece get smoothly joined and you don't see the stutter.

cozierad, 2017-02-19:
Are you using a servo probe like a BLTouch? I noticed this too but I think it is the servo delay working. At least lowering the servo delay lowered the stuttering.

mgroel, 2017-02-20:
Ah, okay, not thought about the queue planner and disabled bi-linear-leveling for now. So it seems not to be a problem I will try again.

@cozierad Yes, I use the original BLtouch. Probing the bed has some newly introduced short delays now, too. Didn't notice the servo-delay variable, so everything seems to be right.

closed 2017-02-20 by mgroel