MarlinFirmware/Marlin #5860: Feature Request:  Hotend configuration switcher

Roxy-3D, 2017-02-19:
@studiodyne requested this feature in another thread.   Just to keep things orderly, I've moved it to its own thread:

Can you make something special for me ' and for us , for all '

Named : The Hotend configuration switcher
What : It's some new code , to give the possibility to switch ' by LCD or Gcode ' , between some predefined configuration in firmware source

For example : Switch from Chimera ' or any dual ' , to cyclops , needs to ' REFIRM' , because the thermal process of hotends must be changed. For chimera , 2 hotends = 2 heaters + 2 sensors + offset + own extruders motors direction +gear steps , but for cyclops 2 hotends + 1 sensor +1 heater .+ own extruders motors direction + gear steps '
Gears steps because , for cyclops/chimera we use mk8 gear ' 7 mm ' to have the maxi torque
Gears steps because if we use V6hotends , and flexion extruder ' 11mm gears' for flexible filament

How ?:
For the moment , in lcd , it's possible to
Change extruder offset , via Gcode in a lcd menu ' easy to make ' ... Ok for that
Change Gear steps ... via ' tune menu'
But IMPOSSIBLE to set the motors directions , and to trick the software by giving the same ' Heater and sensor ' for all hotends ' CYCLOPS config '

So , the firmware , need some modification
Declare ' Heater 2 +3 +4 ' abd ' sensor 2 + 3 + 4' as VARIABLES to give us , the possibility to change it in real time , of to just reboot after settings
declare number of hotends as variables
declare motors direction as variable

What 's in the predefined configuration
Hotend configuration 1
Name ' to use for lcd menu , it's better to have a name and not a number '
Numbers of hotends ' mono , dual , 3 or 4 '
Pins for sensors
Pins for heaters
Gcode intrusction to change offset or offset to be applied
Gear steps
Motors direction

AND if you are a good guys , please insert the config pack in ' EEPROM ' to never have to click when booting loll

It's not an easy job , you have to modify , all constant by variables , you have to make ' messages of the menu in all language or include it to be tweaked for each language '

But , you can be sure , this code will be immediatly inserted in release candidate , because of , the little risk of create bugs , it's just a configurator , not a deep code7

Do you want to make this , you can , you have made the proof of your ability to contribute very seriously for marlin

If success of this code , i'm ok to never say it's my ideal LOLLLLLL

tHKS
---------------------
I can make the code , but , to make it , i need to learn the language , i need to become good and write some line without THE COMPILATOR WAR that never want , what i write , i will need help , to just declare an external float never accepted by the compilator because of the million rules of programming

Please , to make it , i need to become the old developer i was , and i'm not today
I don't have the forces and the brain for that
Sorry
---------------------
If we declare pins as ' array' , like this heater_xxx[x] and sensor_xxx[x] and use this in the code , x is the number of the selected config in lcd and eeprom
and we create a ' class' for all the configuration switcher that can be evolutive for all ideas in the future and all possible evolution of extrusions systems
And in configadv.h , we will have each static definition of each configuration we want to make avalaible in lcd
The simple way , is just to trick the firmware and , have two configurations , pins for each ' case of dual extruder ' , and all pins the same ' cyclops ' , but in reality , the job to make a complete and zero limit configurator is the same job , to make a simple switcher for cyclops or other
So , for this reason , it's better to make a complete switcher , pins declared as 'array' and pins configurations always possible in realtime , of just by a click and a reboot ' may be '

My problem is to know , if , changing the writing/reading of pins is a problem , may be , these value have to be applied at the boot , or may be it's possible to change these anytime we want

But , if we need to reboot , it's perfect too , it's not a problem , one click , one reset and we have the config we need
I decide to make a perfect dual extrusion , i mount and connect my cyclops extrusion system , i click , i reboot , and the machine is ready to start
I decide to make Polycarbonate at high temp , i mount the v6 mono extrusion system , i click i reboot , and go

I need this , all the world need this
It's incredible to , find , a table , the cable , to place our laptop on the table , to plug usb , to refirm the board just to switch between extrusion system

I shut off this discussion and i forget this project if you are not ok to make it
I have posted on reprap , and here , nobody have enough knowledge to make it
On reprap , the communauty want i leave marlin , to use their 32bits ' decent ' hardware
I have told them , that a BOARD WITH ETHERNET and ability to use internet , but THAT CAN NOT MAKE A PUSH ALARM ON MY PHONE to inform me about a problem of printing , is a board that HAVE ZERO INTEREST FOR ME

Incredible ...ethernet connection , 32 bits processor , and just a toy firmware ,that can just tweak some setting
The FIRST THING I WANT if i have the power of web communication of my printer , is to SEND ALARM NOTIFICATIONS when problems , the reprap firmware have forgotten this idea , for me it's the first function I make if i work on a printer with wifi and internet api

End of discussion , i will pray , you accept to work on my idea

Thks for all

studiodyne, 2017-02-19:
// steeve   HOTEND CONFIGURATION SWITCHER
#if ENABLED(HOTENDS_CONFIGURATION_SWITCHER)
    // easy ! we active all extruders , to avoid to make more code 
    // All extruders are defined , and we just play with offsets and pins
    // The slicer/gcode instruction will use only the extruders needed
    // Ex : For cyclops , we set all hotends at the same offsets , motors direction , same heater/sensor pin as The first and it's all
    // i know it's not a serious code , and use ressources for nothing , but , if marlin works for 4 hotends , the ressources exist !
    #undef EXTRUDERS
    #define EXTRUDERS 4 // ;-) 
    #undef HEATER_1_PIN , TEMP_1_PIN 
    // transformation  in ARRAY to change it by LCD
    // define the second heater and sensor as ' macro ' 
    // to give the same pins for all heaters/sensors when 2 or + extruders but only one nozzle
    // Because of  intelligent update of the firmware , the only way is to redefine the constants as macros and arrays
    #define HEATER_1_PIN { SELECTED_BY_CONFIG_HEATER1[lcd_selected_hotend_config];}
    #define   TEMP_1_PIN { SELECTED_BY_CONFIG_TEMP1[lcd_selected_hotend_config];}
    // functions that applied config when lcd command
        extern int lcd_selected_hotend_config=0 // 0= default selection , original firmware pin config and others
        void hotends_selected_cfg(int selected_cfg)
    {// Will apply 'Motors direction' + Gears steps + Hotend Offsets +  all problems to solve that i don't know for the moment lllolll }
  
// now we can make our own configuration  
#define MAX_HOTENDS_CFG   // Our number of differents cfg , to make the loop of LCD dynamic item creation
#define HOTENDS_CONFIGURATION_SWITCHER[0]// default ' dual  V6 for me , 2 heaters / 2 sensors / gears steps / motors direction / hotend offsets 30
//#define HOTENDS_CONFIGURATION_SWITCHER[2]// MONO V6 HIGH TEMP  ,1 EXTRUDER, one heater , one sensor , gears steps , motors direction , hotend  0
#define HOTENDS_CONFIGURATION_SWITCHER[1]// Cyclops ' 2 EXTRUDERS ,one nozzle , one heater , one sensor , gears steps , motors direction , hotend offsets SAME FOR THE 2 
//#define HOTENDS_CONFIGURATION_SWITCHER[3]// Chimera ' 2 EXTRUDERS , one heater , one sensor , gears steps , motors direction , hotend offsets 0
#endif
// ***********************************  DATAS ********************************************************************
#ifdef HOTENDS_CONFIGURATION_SWITCHER[0] // the default original firmware settings
      #define SELECTED_BY_CONFIG_NAME[0]= " DUAL V6 " // FOR LCD ITEM MENU 
      #define SELECTED_BY_CONFIG_HEATER1[0]  // pin of the heater 1 
      #define SELECTED_BY_CONFIG_TEMP1[0]    // pin of the sensor 1
      #define SELECTED_BY_CONFIG_GEAR_STEPS[0]  // GEAR STEPS
      #define SELECTED_BY_CONFIG_E0_DIR[0] //  directions of motors 
      #define SELECTED_BY_CONFIG_E1_DIR[0] //  directions of motors 
      #define SELECTED_BY_CONFIG_E2_DIR[0] //  directions of motors 
      #define SELECTED_BY_CONFIG_E3_DIR[0] //  directions of motors 
      #define SELECTED_BY_CONFIG_HOTENDS_X_OFFSETS[0] {30.0,0.0,0.0} //the 3 other nozzles if exist
      #define SELECTED_BY_CONFIG_HOTENDS_Y_OFFSETS[0] {0.0,0.0,0.0} //the 3 other nozzles if exist
    #endif
#ifdef HOTENDS_CONFIGURATION_SWITCHER[1] // OUR PERSONNAL WISH OF CONFIG ' in my case it's cyclops
      #define SELECTED_BY_CONFIG_NAME[0] = "CYCLOPS" // FOR LCD ITEM MENU
      #define SELECTED_BY_CONFIG_HEATER1[1]  // Same pin of the heater 0 
      #define SELECTED_BY_CONFIG_TEMP1[1]    // pin of the sensor 0
      #define SELECTED_BY_CONFIG_GEAR_STEPS[1]  // GEAR STEPS
      #define SELECTED_BY_CONFIG_E0_DIR[1] //  directions of motors 
      #define SELECTED_BY_CONFIG_E1_DIR[1] //  directions of motors 
      #define SELECTED_BY_CONFIG_E2_DIR[1] //  directions of motors 
      #define SELECTED_BY_CONFIG_E3_DIR[1] //  directions of motors 
      #define SELECTED_BY_CONFIG_HOTENDS_X_OFFSETS[1] {0.0,0.0,0.0} //the 3 other nozzles if exist
      #define SELECTED_BY_CONFIG_HOTENDS_Y_OFFSETS[1] {0.0,0.0,0.0} //the 3 other nozzles if exist
    #endif

tximy, 2017-05-27:
Hi, this is an great work, I need it too ;)
I combine hard and soft and create a gcode configuration file for any head, load and it's save to eeprom, but need upgrade the M851 with X and Y offset of probe https://github.com/MarlinFirmware/Marlin/issues/6713
I has configure 3 extruders and use 1, 2 or 3 , on the single extruder put a resistor to T1 and T2, thi wiring of steppers are inverted for any and did not need mod the software.

Great job
Thx.

boelle, 2019-02-18:
is this one still relavant?

revilor, 2019-02-19:
The crucial point here is: when you swap a Chimera for a Cyclops it's dual-extruder single nozzle vs. dual-extruder dual nozzle, and those are compile time settings at the moment that take effect all over the source code. Therefore it would require quite some effort to make this possible. But I like the idea, could be part of #13130.

boelle, 2019-02-19:
i will agree, @revilor good  example you gave

closed 2022-01-23 by thinkyhead