MarlinFirmware/Marlin #5861: Issue compiling in Linux

W8KDB, 2017-02-19:
I am trying to upload a version of Marlin (it is included with Skynet but is setup for the Anet A8 I have) and they included an Arduino setup for Windows to compile and upload. My laptop is Ubuntu Linux, so the Arduino they included does not work for me. When I try and compile Skynet with Ubuntu Ardunio, I get an error:

exec: "generate_version_header_for_marlin": executable file not found in $PATH

While I know Skynet isn't straight Marlin, the creator mentioned it is mainly Marlin with code setup for the Anet A8 (and possibly other printers, but I am interested in the Anet A8) when I mentioned an issues about the prepare menu options disappearing while it was preheating for PLA. Since the error also reference something with Marlin, I though I would see if there was a known issue about compiling on Linux.

Ubuntu: 16.4.1
Arduino: 1.8.1
Marlin Version: Unknown
Skynet Version: 2.3.1

thinkyhead, 2017-02-19:
You can try putting a copy of `generate_version_header_for_marlin` from the current version of Marlin somewhere in your $PATH, and perhaps it will get past that error.

W8KDB, 2017-02-19:
Where is it located? I did not see a file with that name.

Roxy-3D, 2017-02-19:
@w8kdb   RepRap requires a certain amount of being willing to figure things out.   Please be willing to spend an the time to work through problems as far as you can on your own.

mtowara, 2017-02-24:
in your Arduino hardware/anet/avr/ folder rename or delete the platform.local.txt file.
With that modification it builds for me. (Arduino nightly, Ubuntu 16.10, Skynet 2.3.1 from Facebook (sic!))

the generate_version_header_for_marlin is in buildroot/bin but won't help you as the Skynet thing is not correctly derived from a git repository for that to work.

W8KDB, 2017-02-24:
Thank you @mtowara. Once I removed the file it allows me to compile the code. I as thinking it was something in the code, not the hardware part.

closed 2017-02-24 by W8KDB