MarlinFirmware/Marlin #5863: How do I create an Analog Output Pin (QUESTION)

kcheeeung, 2017-02-19:
How do I create an analog OUTPUT pin from Arduino boards?
**I have an Azteeg X3 Pro (ATMEGA2560 and RAMPS 1.4), and I see that it only has analog input pins.**

How would I transform one of these input to OUTPUT pins and is there a way I can vary voltage with the gcode? Use pinMode()?

One of our devices require an analog input signal, and we want to be able to vary it in voltage.

Bob-the-Kuhn, 2017-02-19:
Sounds like you need to use a hardware PWM port so that you can have an analog output.  A hardware PWM will produce a 1k-15kHz signal that varies in duty cycle.  To a slow device this looks like an analog signal.  The frequency of the signal varies with duty cycle.

You'll need to initialize it as an output by using this:
`SET_OUTPUT(YOUR_PIN);`

To set the output voltage use this:
`analogWrite(YOUR_PIN, value);  //where value can range from 0 through 255`

If you’re using one of the on-board FETS then a value of 0 will give you maximum output.

Here’s the useable hardware PWMs on your board:
``` 
Pin  normally assigned to
2     X_MAX                       
3     X_MIN                       
4     HEATER_4
5     HEATER_5
6     HEATER_6
7     LCD
8     HOTBED
9     HEATER_1
10    HEATER_0
44    LCD                         
45    LCD
```                         

All the HEATER_x and HOTBED signals in the above list are hardwired to power FETS.  Don't use them if you want just a 0-5V voltage. 

Because of the way pins are defined for the Azteeg X3 Pro it’s hard to tell what is assigned to a particular pin.  There may be multiple functions assigned to one pin.  To determine what’s assigned in your system do the following:
1.	In **Configuration_adv.h** enable **PINS_DEBUGGING**
2.	Issue **M43 I1** command 
3.	Note the name of the signal assigned to the pin you want to use
4.	Add **#undef signal name** to the end of your pins_**AZTEEG_X3_PRO.h** file
5.	Repeat steps 2-4 until **unused** shows up
6.	Add **#define YOUR_PIN xx** to the end of your **pins_AZTEEG_X3_PRO.h** file

Bob-the-Kuhn, 2017-02-19:
I forgot about your **gcode** comment.

The command **M42 PYOUR_PIN Svalue** will set **YOUR_PIN** as an output and set the duty cycle to **value** where value can range from 0 to 255.

kcheeeung, 2017-02-19:
@Bob-the-Kuhn 

What file should I put this code in? And is this right? I feel like this is incomplete or I'm missing something.
```
void setup(){
	SET_OUTPUT(YOUR_PIN);
}

void loop(){
	analogWrite(YOUR_PIN, ###);
}
```

Bob-the-Kuhn, 2017-02-19:
Exactly what do you want your system to do?

If you want to manually issue a command to set JUST this analog value then the gcode command **M42 Pyour_pin Svalue** will do.

----
You should also decide what this pin will do on powerup and reset.  That'll determine where to put the initialization.

kcheeeung, 2017-02-20:
@Bob-the-Kuhn 
I just want to send an analog voltage (that can vary) out from the available ports and that's it. This other device needs an analog signal.

To confirm, if I had this:
_start gcode_
...
...
M42 PYOUR_PIN Svalue
G0 XYZ
...

YOUR_PIN would provide a constant analog signal (from 0-5volt)? (Say I happened to use the HEATER_X pin). Yeah I know it'd still be pwm-like, but more or less it'll seem analog right?

Bob-the-Kuhn, 2017-02-20:
Sounds like using the M42 gcode command is a reasonable way to do it.

One BIG caveat is if your device will accept a pulsing digital signal as an analog input.  I suggest checking out how your remote device reacts to the PWM signal by issuing M42 commands with a range of Sxxx values.  You may end up building a table with Sxxx values vs. what your remote device does.  Worst case is you build a simple RC filter to turn the PWM into a true analog signal.

If all you want is a low power (5mA or less)  0-5V signal then do NOT use the heaters.  It's much simpler to use the voltage only PWMs.  All the heaters have +12V/+24V tied to the outputs via an LED in series with a resistor.  You'd need to add circuitry to make sure you don't accidentally kill your remote device by applying too much voltage.

If you point me to your analog device then I can tell you what PWM outputs are compatible with it.

kcheeeung, 2017-02-20:
@Bob-the-Kuhn 
For this issue, our remote device is a pressure regulator. The other (in the previous issue) is a microvalve, which we solved by issuing a pulse. That pulse is supposed to be the signal to open/close the valve.

**Pressure regulator (We have ITV0050-3N)**
http://www.ebay.com/itm/LOT-of-2-SMC-ITV0050-3N-Electro-Pneumatic-Regulators-with-cables-/142035525801?hash=item2111fa20a9:g:ZpIAAOSwMNxXbWOo
**Datasheet**
http://content.smcetech.com/pdf/ITV0000_EU.pdf
It says the input signal type is: Voltage type 0 to 10VDC. The input signal voltage is linearly proportional to the output pressure. We would want some way of being able to vary pressure, by changing voltage.

I previously tested by connecting the signal end to the FAN_PIN (also a HEATER_X pin) on the Azteeg X3 Pro. I tried testing with M106 SXXX (fan on), and it seemed like the pressure regulator didn't really like it.

The last scenario is to build a RC filter, but I wanted to see if there's alternatives on the board. Where would be these voltage only PWMs be as well?

Bob-the-Kuhn, 2017-02-20:
The voltage only (0-5V) hardware PWMs on the Azteeg pro are:
```
2     X_MAX                       
3     X_MIN                       
7     LCD
44    LCD                         
45    LCD
```

What voltage are you running your hotends & fans at?  If it's 12V then a 0-10V signal is actually semi-easy to create.  If it's 24V then it takes a couple more components.

>  seemed like the pressure regulator didn't really like it

What exactly does this mean?  Was the pressure always zero, did the pressure pulse, ...?

A couple of things come to mind that might have interfered:
- Control is reversed when using a heater/fan because of the FET.  M106 S255 will make the fan spin fastest but the output signal will be zero volts so the pressure regulator will be at minimum pressure.  M106 S0 will turn the fan off and the pressure regulator will be at maximum.
- The fan control signal can be really slow in which case the pressure regulator might be trying to swing from max to min & back all the time.

If you're running 12V heaters/fans then see if this works.  It basically isolates logical heater 6 (HE 7 on the silkscreen) from all the Marlin thermal controls.
- In  pins_AZTEEG_X3_PRO.h comment out **#define HEATER_6_PIN  6** 
- In  pins_AZTEEG_X3_PRO.h add this:  **#define PRESSURE_PIN  6** 
- Connect the "-" terminal on **HE7** (as shown on the silkscreen) to the control signal on the pressure regulator
- M42 P6 S128 should give you about 6V, S255 about 0V and S42 about 10V

kcheeeung, 2017-02-20:
Very interesting. I'll try this out and let you know of the results. Our power source is currently at 12V.

> What exactly does this mean? Was the pressure always zero, did the pressure pulse, ...?

There was a red light on the pressure regulator meaning an error (nothing happens).

> The fan control signal can be really slow in which case the pressure regulator might be trying to swing from max to min & back all the time.

How we'll be using the pressure regulator is just initializing it once, at the very start start of the print so we just need a single constant voltage/pressure.

> Connect the "-" terminal on HE7 (as shown on the silkscreen) to the control signal on the pressure regulator

We've been connecting the control signal wire to the "+" terminal of the heater/fans so that might have been also why we received an error and nothing happened.

kcheeeung, 2017-02-24:
@Bob-the-Kuhn 
> If you're running 12V heaters/fans then see if this works. It basically isolates logical heater 6 (HE 7 on the silkscreen) from all the Marlin thermal controls.

You are right. I was able to generate the voltages as you said. Is the "+" terminal in this case not to be considered the ground? When I measured the voltage, I measured the "-" and ground COM on the power supply, and it seems to be right.

If this is true for just generating a voltage from the heater/fan, should I be only using the "-" terminal as the signal end only? The pulse on the microvalve, where this voltage is supposed to be opening the valve.
```
inline void gcode_M430() {
  #define DEFAULT_PULS_WIDTH  200
  #define CHECK_DELAY         100
  uint16_t pulse_usec = code_seen('S') ? code_value_int() : DEFAULT_PULS_WIDTH;
  delayMicroseconds(CHECK_DELAY);
  WRITE(FAN_PIN, HIGH);
  delayMicroseconds(pulse_usec);
  WRITE(FAN_PIN, LOW);
  delayMicroseconds(CHECK_DELAY);
}
```

```
PIN:  9        FAN_PIN                     Output = 0    TIMER2B    WGM: 1    TIMSK2: 0   can be used as PWM
PIN: 10        HEATER_0_PIN                Output = 0    TIMER2A    WGM: 1    TIMSK2: 0   can be used as PWM
```
What does it mean that a pin is protected when I sent a M42 P# S# command? I commented out FAN_PIN in pins_azteeg_x3_pro.h. Am I missing places that HEATER_0_PIN or FAN_PIN might exist and be in use? I want to be able to utilize these two heater/fan ports as well.

kcheeeung, 2017-02-24:
This is what I had 
```
//
// Heaters / Fans
//
// #define HEATER_2_PIN       16 //AZTEEG X3 PRO
// #define HEATER_3_PIN       17 //AZTEEG X3 PRO
// #define HEATER_4_PIN        4 //AZTEEG X3 PRO
// #define HEATER_5_PIN        5 //AZTEEG X3 PRO
// #define HEATER_6_PIN        6 //AZTEEG X3 PRO
// #define HEATER_7_PIN       11 //AZTEEG X3 PRO

#define TEST_0_PIN       10 // HE1 protected USED BY HEATER_0_PIN SOMEWHERE
#define TEST_1_PIN        9 // HE2 protected USED BY FAN_PIN SOMEWHERE
#define TEST_2_PIN       16 // HE3 NOT PWM
#define TEST_3_PIN       17 // HE4 NOT PWM
#define TEST_4_PIN        4 // HE5 WORKS
#define TEST_5_PIN        5 // HE6 WORKS
#define TEST_6_PIN        6 // HE7 WORKS
#define TEST_7_PIN       11 // HE8 PWM BUT CAN'T USED BC OF COUNTER MODE

// #undef FAN_PIN
// #define FAN_PIN             6 // Part Cooling System   AZTEEG X3 PRO

PIN: 11 <unused> Input = 0 TIMER1A WGM: 4    TIMSK1: 2   Can't be used as a PWM because of counter mode

```

Also what would it mean that "PWM because of counter mode"?

Bob-the-Kuhn, 2017-02-24:
The + terminals on HE1-HE8 are tied to the power connector **INPUT 2** through fuse F4.  If you're reading 0V on them then either you haven't tied anything to INPUT 2 or F4 is open.

The - terminals are the drains of the power FETs.  

Normally one wire from a fan or a heater is tied to the + terminal and the other wire to the - terminal.  

In your case you just want a voltage signal so ...
- The + signal from your valve control input is tied to the - terminal.
- The - signal from your valve control input is tied to ground.
-  A pull up from the - terminal to +12 is needed.  The pull up probably should be in the 1K - 10K range.  There is already a weak pull up on board in the form of a red LED in series with a resistor tied between the + terminal and the - terminal.  If you command a 10V output but are only reading 9V then you'll need to add an external 1K resistor between the + terminal and the - terminal.

----

>  "Can't be used as a PWM because of counter mode"

Mostly yes - Timer 1 is used to generate interrupts that are used to count time, to run the thermal manager periodically and to step the motors.

Because of the way Timer 1 is set up the hardware PWM functions of that timer are not available.  They're basically just like any other general purpose digital I/O pin.  Sometimes these pins are used as SOFTWARE PWMs (usually for slow things like fans and heaters). A bit confusing.

----

**HEATER_0_PIN** and **HEATER_1_PIN** are defined in **pins_RAMPS.h**.  They can be re-defined in **pins_AZTEEG_X3_PRO.h AFTER** the **#include "pins_RAMPS.h"** line.

----

The heaters and extruder fans should be controlled **ONLY** by the thermal manager.  To enforce that, all the pins that have FAN or HEATER in their name are put on a "protected" list.  The M42 command consults this list before taking any action.  

----

Marlin insists on having one heater and one fan defined for each extruder and insists that there must be at least one extruder.  Because of this you MUST have **HEATER_0_PIN** and **FAN_PIN** defined.  You can assign them to any pins you want (even non-existent ones like 78).

----

Putting it all together and assuming that you want to use pin 6 to control the valve your **pins_AZTEEG_X3_PRO.h** file should have the following in it.  
```
//
// Heaters / Fans
//
// #define HEATER_2_PIN       16 //AZTEEG X3 PRO
// #define HEATER_3_PIN       17 //AZTEEG X3 PRO
// #define HEATER_4_PIN        4 //AZTEEG X3 PRO
// #define HEATER_5_PIN        5 //AZTEEG X3 PRO
// #define HEATER_6_PIN        6 //AZTEEG X3 PRO
// #define HEATER_7_PIN       11 //AZTEEG X3 PRO

//#define TEST_0_PIN       10 // HE1 protected USED BY HEATER_0_PIN SOMEWHERE
//#define TEST_1_PIN        9 // HE2 protected USED BY FAN_PIN SOMEWHERE
#define TEST_2_PIN       16 // HE3 NOT PWM
#define TEST_3_PIN       17 // HE4 NOT PWM
#define TEST_4_PIN        4 // HE5 WORKS
#define TEST_5_PIN        5 // HE6 WORKS
//#define TEST_6_PIN        6 // HE7 WORKS
//#define TEST_7_PIN       11 // HE8 PWM BUT CAN'T USED BC OF COUNTER MODE

#undef FAN_PIN
#define FAN_PIN             11   // Part Cooling System
#undef SERVO1_PIN     // also assigned to pin 6
#define VALVE_CONTROL_PIN  6  // pressure valve control signal
```

----

FYI - here's the pin assignments for an unmodified AzteegX3pro system.  Note that some pins have multiple functions tied to them.
[temp.txt](https://github.com/MarlinFirmware/Marlin/files/798442/temp.txt)

kcheeeung, 2017-02-25:
@Bob-the-Kuhn 
I would like to thank you for all your invaluable knowledge about the Marlin firmware and 3d printer boards. I am a researcher from the Lawrence Berkeley National Laboratory, and we are actually building a custom 3d printer to extrude experimental materials using this firmware. We couldn't have make significant progress and utilize our peripherals without your help. Thank you so much again.

The Marlin firmware is very intelligent and says there's non-existent pins so I just defined the HEATER_0_PIN and HEATER_1_PIN as pin 11.

closed 2017-02-25 by kcheeeung

Bob-the-Kuhn, 2017-02-25:
You're welcome.  Glad to be of help.