MarlinFirmware/Marlin #5865: CoreYX homing ignores endstops (RCBugFIX )

Brandano, 2017-02-19:
The printer is a K8200 modified for a CoreYX layout (note YX, not XY). Movement direction is fine, and so is homing direction, and the endstops read correctly on M119, but when the homing sequence starts the axis reaches the endstops and stall against it without ever backing off. I think this might be the same issue as https://github.com/MarlinFirmware/Marlin/issues/3964 , that appears to have been closed without an actual followup. I am attaching my configuration files.

(renamed as txt because Github) 
[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/786054/Configuration.txt)
[Configuration_adv.txt](https://github.com/MarlinFirmware/Marlin/files/786055/Configuration_adv.txt)

Bob-the-Kuhn, 2017-02-19:
By any chance is X_HOME_BUMP_MM, Y_HOME_BUMP_MM and/or Z_HOME_BUMP_MM set to zero?

What do you mean by "stall against it"?  Does it come up to the stop and stop at the correct location or does it try to continue on until it hits the physical stop and starts grinding away?

Brandano, 2017-02-20:
X_HOME_BUMP_MM and Y_HOME_BUMP_MM are both set to 5, Z_HOME_BUMP_MM is set to 2, but I have not yet tried to home Z for fear of actually damaging the printer. The printer just grinds against the endstop, or better, reaches it and starts skipping steps because I lowered the motor current for safety, but the endstop is clearly triggered.

Bob-the-Kuhn, 2017-02-20:
How do you know the endstop is triggered?

I like to use the M43 E1 command when playing with endstops.  This routine sits in the background, polls the endstops and reports whenever an endstop changes state.  It doesn't care if endstops are enabled or not and it doesn't care if the head is moving. It also toggles the system LED so it makes physically setting the endstops much easier.

To use the M43 E1 command you'll need to enable PINS_DEBUGGING near the end of Configuration_adv.h

If you issue the M43 E1 command and then do a G28 X you should see X_MIN popup when the carriage hits the X MIN endstop  (assuming you're homing in the negative direction).  If not then either you need to adjust your sensor or fix your wiring or specify the correct pin in your pins_YOUR_BOARD.h file.

Just for grins & giggles issue a M43 I1 command.  That one will report what signals are assigned to each pin.

Blue-Marlin, 2017-02-20:
Check if the 'right' endstop is used.
If homing to min the min-endstop has to be in use. If that is not working ehter your min endstop is the max endstop, or the direction of the motor is wrong.

Brandano, 2017-02-20:
The axes move and home in the right direction, towards the correct endstop, and the endstops read correctly when queried with M119. If I manually place the bed to the origin and set the position with G92 the printer will behave correctly. I will have to enable M43 and repeat the tests, as well as trying to increase the bump distance, though the smaller pinions and higher step count per mm should in theory mean that 5mm is plenty.  I guess I could also reduce the homing speed and see if it makes a difference.

Blue-Marlin, 2017-02-20:
Pfff.
Maybe your and the machines opinion where the origin is, is different.

Blue-Marlin, 2017-02-20:
@thinkyhead 
Has `CORESIGN()` to be used in `Endstops::update()`, in one of the direction-ifs?

Brandano, 2017-02-20:
The machine is a K8200, so the bed moves in X and Y and the extruder moves in Z. Home is set as X0, Y0 and Z0, with the nozzle of the hotend touching the lower left corner of the bed (ie, the bed moved to the upper right range of the movement) and in this situation all three MIN endstops are triggered if queried with M119. The coordinates system is the same as before the CoreYX conversion, right handed or at least matching that of Slic3r. I had a similar configuration working with RC6, except for the specification of CoreYX movement and the recalculated steps per mm on X and Y due to using 16T pulleys on GT2 instead of the original hardware on T2.5 belts. I am not near the printer right now, I will experiment a bit more on it in a few hours. Thanks for your time, though.

Bob-the-Kuhn, 2017-02-20:
It's important that you issue the M43 E1 command before doing a home command.  That's the only way to know what the machine is seeing.   Without M43 E1 we're all guessing.

Brandano, 2017-02-20:
I am not having much luck with M43.
Here's the response after sending the command via octoprint:
```
Send: M43 E1
Recv: endstop monitor enabled
Recv: ok
Recv: X_MIN:1
Recv: 
Recv: X_MIN:0
[...]
Send: G28 X0
Recv: echo:busy: processing
Recv: X_MIN:1
Recv: 
Recv: X_MIN:0
Recv: 
Recv: echo:busy: processing
Recv: start
Recv: echo: External Reset
```
The first trigger is me testing the endstop manually, then I started the G28 command, triggered the endstop by hand before the movement could reach it, then released it and resetted the printer to prevent the movement from pushing against the endstop uselessly. So it seems like the endstop reads properly, but for some reason the printer just won't aknowledge it and bump.

Bob-the-Kuhn, 2017-02-20:
Yes, Marlin commands are all uppercase.  My Repetier Host automatically converts everything to uppercase before sending it to the printer.

>  started the G28 command, triggered the endstop by hand before the movement could reach it

Did the motor stop, reverse a little and then go back in the - direction?  

If it did then the X_MIN system is working correctly.  That implies the X_MIN sensor is in the wrong position.  If you want to confirm then issue the M43 E1 & G28 X, let the head stall for a short time  and see if X_MIN 1 pops up.  If X_MIN 1 doesn't pop up then for sure the sensor is at the wrong location.

To adjust the X MIN sensor I suggest the following:
1. Move the head to where you want the X 0 position to be
2. Issue the M43 E1 command
3. Enable interrupts M120 
4. Move the sensor away from the physical end stop toward the head until you see the system LED change state
5. Confirm that X_MIN 1 has popped up on the screen
6. Issue G0 X5 command and then a G0 X-2 command
7. The screen should show X_MIN going to 0 and then back to 1 and it should say that the X_MIN endstop has been hit
8. If the head is NOT at an acceptable position then redo steps 4 - 7
9. If the head is at an acceptable position tighten down the sensor, redo steps 6 & 7 to confirm it didn't move during the tightening process and then declare victory.

Brandano, 2017-02-20:
Nope, no pause, just kept travelling while the endstop was triggered. I did triple check that the endstops were in the correct position, and that the movements were in the correct directions. as I was saying, I could actually jog the bed in all directions if I homed it manually and set G92 X0 Y0, and M114 returns the expected position after a G1 move. I am fairly sure that the homing sequence is just ignoring the fact that the endstop is being triggered and carrying along as if nothing had happened.

Bob-the-Kuhn, 2017-02-21:
We're making a false assumption somewhere.  Time to start from scratch & assume everything is wrong until proven otherwise.

I REALLY want you to try to hit the endstop.  PLEASE, PLEASE, PLEASE do the following:
1. move the head near X0
2. issue M119
3. issue M43 E1
4. issue M120
5. keep moving the head towards X0 until either you hit the physical stop, the screen says endstop hit or one of the endstops shows up on the screen.
6. issue M119

Brandano, 2017-02-21:
Here is the output from that sequence of opreations (the [...] sections are suppressed temperature queries and responses from octoprint):
```
[...]
Send: M119
Recv: Reporting endstop status
Recv: x_min: open
Recv: y_min: open
Recv: z_min: open
Recv: ok
[...]
Send: M43 E1
Recv: endstop monitor enabled
Recv: ok
[...]
Send: M120
Recv: ok
[...]
Recv: X_MIN:1
Recv: 
[...]
Send: M119
Recv: Reporting endstop status
Recv: x_min: TRIGGERED
Recv: y_min: open
Recv: z_min: open
Recv: ok
```
I am quite sure that this is not a configuration issue, that's why I did post the configuration files in my first post.

Bob-the-Kuhn, 2017-02-21:
The code that does the detection & starts the terminate move sequence is in endstops.cpp,

An endstop is checked only if there is movement on that axis in the correct direction.

I'll look at the code.

Bob-the-Kuhn, 2017-02-21:
Yes, Y and X endstops are checked for CoreYX layout.

Could the logical Y_MIN endstop be the one checked when going in the -X direction?  Maybe as a result of using YX instead of XY?

Try switching X & Y endstops at the controller or switching the pin numbers in pins_YOUR_BOARD.h.

Brandano, 2017-02-21:
I will have to do that later this evening when I come back from work, but in the meantime I tried triggering the MIN_Y endstop manually while homing X, with no change in behaviour:
``` M105 Recv: ok T:18.7 /0.0 B:18.7 /0.0 @:0 B@:0 Send: G28 X0 Recv: echo:busy: processing Recv: Y_MIN:1 Recv: Recv: Y_MIN:0 Recv: Recv: Y_MIN:1 Recv: Recv: echo:busy: processing Recv: Y_MIN:0 Recv: Recv: X_MIN:1 Recv: Recv: start 
```

Brandano, 2017-02-21:
Specifically, i think that the 2nd part of the OR statement might not be evaluated properly considered the definition of #define CORESIGN(n) (-(n)) in line 63 of Conditionals_post.h . Is it possible that the test should be reversed for CoreYX?

Bob-the-Kuhn, 2017-02-21:
Can't help you on the "why" questions.  Just don't know enough about core_XY, ... type machines.

Either we have a bug or there's a lack of documentation.

Let's see what direction endstops.cpp thinks it's going in.  I've added some print statements to the attached.

Please do the following:
rename your current chop endstops.cpp
chop off the ".txt" from the attached and drop this into your directory
do a G28 X  (probably will result in the printer going through reset)
do a G28 Y  (probably will result in the printer going through reset)

[endstops.cpp.txt](https://github.com/MarlinFirmware/Marlin/files/790831/endstops.cpp.txt)

Brandano, 2017-02-21:
The printer doesn't reset, but ends looping through the print statements several times before apparently freezing. I had to power cycle the controller to connect to it again, as even the reset button was ignored. I am getting the same output for both G28 X0 and G28 Y0:
```
[...]
Recv: NOT_SAME_NUM_STEPS is false
Recv: SAME_DIRECTION is true
Recv: X_HEAD is false
Recv: Y_HEAD is false
Recv: +X direction
Recv: NOT_SAME_NUM_STEPS is false
Recv: SAME_DIRECTION is true
Recv: X_HEAD is false
Recv: Y_HEAD is false
Recv: +X direction
[...]
```
This was repeated several times, enough to exceed octoprint's terminal buffer, so I just chopped it.

Bob-the-Kuhn, 2017-02-21:
Very definitely at least one bug in the software.

G28 X results in the endstop routine thinking it's going in the +X direction.  Semi-believable

G28 Y **ALSO** results in the endstop routine thinking it's going in the +X direction.  Not believable.

I don't see a quick fix to get both G28 X and G28 Y working properly.

Got to go help a neighbor.  It'll also give me some time to think about it.

Bob-the-Kuhn, 2017-02-21:
As I understand it on core XY & core YX machines
- same number of steps means only moving in one direction 
- steppers rotating in same direction - table moving in direction 1
- steppers rotating in opposite directions - table moving in direction 2

In your system when doing G28 X0 or G28 Y0:
- the steps are the same  - as is expected
- pure X direction and pure Y direction **BOTH** have the steppers turning in the same direction - **NOT** as expected

I'd like more info.  Can you point me to the equations for core YX?  I have the ones for core XY.

Here's a new version of the endstop code.  I've added print statements so we can see which direction the steppers are turning.

Please drop the new code into your directory and then do pure X & pure Y moves in both directions.
G0 X10, G0 X0, G0 Y10, G0 Y0

Bob-the-Kuhn, 2017-02-21:
and enable endstops

Brandano, 2017-02-21:
CoreYX is exactly the same as CoreXY, but with one stepper reversed as they are on the side of the printer instead of the front or back. Turning both steppers in the same direction causes one axis to move, turning them in opposite directions causes the other to move. Turning only one stepper causes the effector, or in my case the print bed, to move along a diagonal. In any case, the fact that in both cases the test detects the steppers turning in the same direction is obviously false, though the number of steps will be the same for both pure X or Y movements. Here is a rather blurry picture of my  setup: 
[Imgur](http://i.imgur.com/duUNGo8.jpg)
It can help thinking of it as an upside-down corexy printer with the print surface stuck to the hotend carriage and viceversa.
I don't see the updated code, possibly a markdown tag is missing?

Brandano, 2017-02-21:
This is an OpenSCAD animation of the movement. 
[Imgur](http://i.imgur.com/Xnio3Fd.gifv)
I suppose I could just switch to CoreXY and live with the idea that my build volume is rotated 90 degrees, though probably I'll also have to reverse one or both axes then.

Bob-the-Kuhn, 2017-02-21:
OK - basically the same equations govern both.

Yes, I did forget to attach the file.
[endstops.cpp.txt](https://github.com/MarlinFirmware/Marlin/files/791680/endstops.cpp.txt)

This'll report the direction that endstops.cpp sees and the settings of the direction pins.

Brandano, 2017-02-21:
I will probably have to remove some of those outputs, because I think that it's causing the serial buffer to overload and freezes the printer to a level where it has to be cold cycled to reset

Brandano, 2017-02-21:
Ok, I have 100 steps per mmset, so a 10mm movement will generate quite a long sequence of steps that will cause the printer to bog down. I can move by 0.07mm at a time, it seems, without causing too much distress. the interesting thing is the return from M114:
```
Send: M114
Recv: X:0.07 Y:0.00 Z:0.00 E:0.00 Count A:7 B:-7 Z:0
```
Where that A and B are probably the cound of steps for the A and B motor. This is after a 0.7mm move in X, so the output from above still applies. 
So, let's move X back to "zero":
```
Send: G0 X0
Recv: ok
Recv: NOT_SAME_NUM_STEPS is false
Recv: SAME_DIRECTION is false
Recv: X_HEAD is true
Recv: Y_HEAD is false
Recv: CORE_AXIS_1 direction: 1
Recv: CORE_AXIS_2 direction: 0
Recv: X stepper direction: 0
Recv: Y stepper direction: 1
Recv: +Y direction
```
Hmm, why +Y direction? something here is wrong. And yet the movement is correct. And M114:
```
Send: M114
Recv: X:0.00 Y:0.00 Z:0.00 E:0.00 Count A:0 B:0 Z:0
```
Let's try Y:
```
Send: G0 Y0.07
Recv: ok
Recv: NOT_SAME_NUM_STEPS is false
Recv: SAME_DIRECTION is true
Recv: X_HEAD is false
Recv: Y_HEAD is false
Recv: CORE_AXIS_1 direction: 0
Recv: CORE_AXIS_2 direction: 0
Recv: X stepper direction: 1
Recv: Y stepper direction: 1
Recv: +X direction
[...]
Send: M114
Recv: X:0.00 Y:0.07 Z:0.00 E:0.00 Count A:7 B:7 Z:0
```
Aaand now it thinks it's moving X. Moving back?
```
Send: G0 Y0
Recv: okNOT_SAME_NUM_STEPS is false
Recv: SAME_DIRECTION is true
Recv: X_HEAD is false
Recv: Y_HEAD is true
Recv: CORE_AXIS_1 direction: 1
Recv: CORE_AXIS_2 direction: 1
Recv: X stepper direction: 0
Recv: Y stepper direction: 0
Recv: +X direction
[...]
Send: M114
Recv: X:0.00 Y:0.00 Z:0.00 E:0.00 Count A:0 B:0 Z:0
```
I'd say that the logic that decides the direction as set in endstops.cpp is broken, even though it appears to be fine in the planner.

Brandano, 2017-02-21:
Looking at the values of X_head and Y_head it seems to me that those alone should be enough to decide the direction, though not necessarily which axis is moving.

Bob-the-Kuhn, 2017-02-21:
Could you repeat that using G28 X0 and G28 Y0?  I have a suspicion that it's different.

Brandano, 2017-02-21:
The problem with G28 is that I can't limit the movement, so I have to cold cycle after every attempt, since the printer will freeze. 
```
Send: G28 X0
Recv: NOT_SAME_NUM_STEPS is false
Recv: SAME_DIRECTION is true
Recv: X_HEAD is false
Recv: Y_HEAD is false
Recv: CORE_AXIS_1 direction: 0
Recv: CORE_AXIS_2 direction: 0
Recv: X stepper direction: 1
Recv: Y stepper direction: 1
Recv: +X direction
```
Just repeats those lines over and over and leaves the system LED blinking fast.
```
Send: G28 Y0
Recv: NOT_SAME_NUM_STEPS is false
Recv: SAME_DIRECTION is true
Recv: X_HEAD is false
Recv: Y_HEAD is false
Recv: CORE_AXIS_1 direction: 0
Recv: CORE_AXIS_2 direction: 0
Recv: X stepper direction: 1
Recv: Y stepper direction: 1
Recv: +X direction
```
Hmm, so it looks like G28 doesn't quite set the movement with the same logic as G0? In any case the test is not useful to detect which axis is moving.

Bob-the-Kuhn, 2017-02-21:
You're driving me to drink!  My consumption of diet Dr. Pepper is **WAY** up.

The G0 results are believable.  The G28 are not.

Wait a minute - doesn't G28 do a Z raise before doing the X or Y movement?  If that's the case then what we're seeing is the info from the Z raise movement.

Brandano, 2017-02-21:
Yes, it indeed does. I'll have to disable that

Bob-the-Kuhn, 2017-02-22:
Here's what I think is the fix.  I've modified the logic to agree with the G0 data.

The X_MAX and X_MIN directions may not be correct yet.

[endstops.cpp.fix.txt](https://github.com/MarlinFirmware/Marlin/files/791871/endstops.cpp.fix.txt)

Bob-the-Kuhn, 2017-02-22:
Here's the fix with the X_MAX and X_MIN directions also flipped

[endstops.cpp.fix2.txt](https://github.com/MarlinFirmware/Marlin/files/791882/endstops.cpp.fix2.txt)

Brandano, 2017-02-22:
Thanks again for your patience, Bob-the-Kuhn. I am testing the first fix. There's a typo at line 322, a === check in place of a == check. I tried homing X, but the debug message flooded the serial buffer and caused the controller to freeze. However it was reporting correctly `Recv: -X direction` . I am re-trying with the debug info turned off, leaving in the messages for the Y axis, but I will have to go to bed soon.

Brandano, 2017-02-22:
And the direction appears to be correct on the first fix too. Oddly enough while homing X it would output `Recv: +Y direction`, but this doesn't seem to affect the homing sequence for X. Homing Y would correctly print `Recv: -Y direction` . I'll remove the debug output and try an actual homing, but this seems to have fixed it, at least for me.

Bob-the-Kuhn, 2017-02-22:
Did you put your Y and X endstops cables back to normal?

Brandano, 2017-02-22:
Huh? I never changed the cabling, I just manually triggered the "wrong" endstop for tests.

Bob-the-Kuhn, 2017-02-22:
Fixed & happy - great news!

I'm going to submit the fix.