MarlinFirmware/Marlin #5867: Abort print issue while print sd card

cyptech, 2017-02-19:
I'm using latest Marlin-RCBugFix with Prusa I3 clone and working fine except while sd card printing when i select "stop print" printer stop printing but heaters doesn't stop and continue heating extruder and heatbed. Then i have to select manually cooldown option from menus for shutdown heaters.  Is someone saw that issue before?

Bob-the-Kuhn, 2017-02-19:
That's the way my printer works.

The shut down / cool down commands are automatically appended to the end of the gcode file by the slicer.  When you stop the print  part way through that means the remaining gcode commands are not executed (including the shut down / cool down commands).

I'm not aware of a means of automatically running a set of gcode commands when stopping a print part way through.

A partial solution is to create a gcode file that has the shut down / cool down commands in it and then execute it manually after you stop the print.

cyptech, 2017-02-20:
Instead of struggling with menu or gcode, i follow lcd_cooldown() function at ultralcd.cpp and found a function to disable all heaters on printer. Then i copied thermalManager.autotempShutdown(); function into lcd_sdcard_stop(). Now stop print working as expected. It's dirty fix for me so i hope developers fix code near soon.

closed 2017-10-31 by thinkyhead