MarlinFirmware/Marlin #5868: Bilinear bed leveling w/BLTouch not working

Psyc0Flyer, 2017-02-20:
I recently configured RCBugFix and uploaded to my Azteeg X3 Pro board and I can not get the autolvl to work! My probe works fine and probes the defined points correctly but when it prints, the nozzle still digs in a little on one side. What do I have to do to get this working properly?? I have tried M420 S1 to enable auto leveling, M500 after probing and everything. I get the readout after probe but nothing is compensated for the bed irregularities.. PLEASE help!

Here is what I get when finished probing
READ: Bilinear Leveling Grid:
READ:       0     1     2     3
READ:  0 +0.28 +0.24 +0.36 +0.21
READ:  1 +0.25 +0.20 +0.16 +0.07
READ:  2 +0.10 +0.07 +0.08 +0.05
READ:  3 +0.23 +0.20 +0.18 +0.14

This is my starting script in S3D
G21                     ;metric values
M107                  ;start with the fan off
G28 
M420 S1
M420 Z0.7
G29                     ;AUTO PROBE
M500                  ; SAVE AUTO PROBE RESULTS
G1 X2 Y2 Z0
M109 S245        ;set extruder temp and wait
M116                 ;wait for temps to stabilize
G92 E0                  ;zero the extruded length
G1 F200 E20              ;extrude 10mm of feed stock
G92 E0                  ;zero the extruded length again

cozierad, 2017-02-20:
M420 is for Mesh leveling, which is totally different than bilinear leveling. You should only need a G28 followed by a G29. I would remove the M420 commands and see if that helps. Make sure you have your Z probe offset set correctly. 

M851 Zxxx followed by an M500 is used to set the offset in EEPROM. You only need to do this once.

Psyc0Flyer, 2017-02-20:
I used that script on the last try to see if it helped. It has already been deleted. I have done the M851 z-2.8/M500 but what I am saying is that after everything is done, it does not apply the compensations for the bed so it still acts as if it never probed in the first place. If I were to print without probing, I would get the exact same issue so therefore the probe calibration is not being applied.

Psyc0Flyer, 2017-02-20:
Here is my config file. Is there issues with it?

`/**
 * Marlin 3D Printer Firmware
 * Copyright (C) 2016 MarlinFirmware [https://github.com/MarlinFirmware/Marlin]
 *
 * Based on Sprinter and grbl.
 * Copyright (C) 2011 Camiel Gubbels / Erik van der Zalm
 *
 * This program is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, either version 3 of the License, or
 * (at your option) any later version.
 *
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program.  If not, see <http://www.gnu.org/licenses/>.
 *
 */

/**
 * Configuration.h
 *
 * Basic settings such as:
 *
 * - Type of electronics
 * - Type of temperature sensor
 * - Printer geometry
 * - Endstop configuration
 * - LCD controller
 * - Extra features
 *
 * Advanced settings can be found in Configuration_adv.h
 *
 */
#ifndef CONFIGURATION_H
#define CONFIGURATION_H

/**
 *
 *  ***********************************
 *  **  ATTENTION TO ALL DEVELOPERS  **
 *  ***********************************
 *
 * You must increment this version number for every significant change such as,
 * but not limited to: ADD, DELETE RENAME OR REPURPOSE any directive/option.
 *
 * Note: Update also Version.h !
 */
#define CONFIGURATION_H_VERSION 010100

//===========================================================================
//============================= Getting Started =============================
//===========================================================================

/**
 * Here are some standard links for getting your machine calibrated:
 *
 * http://reprap.org/wiki/Calibration
 * http://youtu.be/wAL9d7FgInk
 * http://calculator.josefprusa.cz
 * http://reprap.org/wiki/Triffid_Hunter%27s_Calibration_Guide
 * http://www.thingiverse.com/thing:5573
 * https://sites.google.com/site/repraplogphase/calibration-of-your-reprap
 * http://www.thingiverse.com/thing:298812
 */
[... 1368 lines cut ...]
 * 34 - RAMPS1.4 - uses Analog input 5 on the AUX2 connector
 * 81 - Printrboard - Uses Analog input 2 on the Exp1 connector (version B,C,D,E)
 * 301 - Rambo  - uses Analog input 3
 * Note may require analog pins to be defined for different motherboards
 **********************************************************************/
// Uncomment below to enable
//#define FILAMENT_WIDTH_SENSOR

#define DEFAULT_NOMINAL_FILAMENT_DIA 3.00  //Enter the diameter (in mm) of the filament generally used (3.0 mm or 1.75 mm) - this is then used in the slicer software.  Used for sensor reading validation

#if ENABLED(FILAMENT_WIDTH_SENSOR)
  #define FILAMENT_SENSOR_EXTRUDER_NUM 0   //The number of the extruder that has the filament sensor (0,1,2)
  #define MEASUREMENT_DELAY_CM        14   //measurement delay in cm.  This is the distance from filament sensor to middle of barrel

  #define MEASURED_UPPER_LIMIT         3.30  //upper limit factor used for sensor reading validation in mm
  #define MEASURED_LOWER_LIMIT         1.90  //lower limit factor for sensor reading validation in mm
  #define MAX_MEASUREMENT_DELAY       20     //delay buffer size in bytes (1 byte = 1cm)- limits maximum measurement delay allowable (must be larger than MEASUREMENT_DELAY_CM  and lower number saves RAM)

  #define DEFAULT_MEASURED_FILAMENT_DIA  DEFAULT_NOMINAL_FILAMENT_DIA  //set measured to nominal initially

  //When using an LCD, uncomment the line below to display the Filament sensor data on the last line instead of status.  Status will appear for 5 sec.
  //#define FILAMENT_LCD_DISPLAY
#endif

#endif // CONFIGURATION_H`

cozierad, 2017-02-20:
From what I can tell your firmware config looks correct. I just uploaded the RCbugFix today and configured my Delta(Ramps 1.4) to use a BLTouch.  

I verified that the compensation is working by triggering the probe, in several spots, several mm above the print bed; then i printed a part. It is actively compensating. Maybe you could try this? 

It would rule out a hardware or offset issue. 

The other option is to disable EEPROM settings and plug your probe offset and any other EEPROM settings directly into the config file. That is how I just setup mine. Maybe the EEPROM offsets are not being read correctly for some reason.

cozierad, 2017-02-20:
The only other thing I can think of is that bilinear leveling used to only work with an odd number of points. 

ie 
#define ABL_GRID_MAX_POINTS_X 5

instead of
#define ABL_GRID_MAX_POINTS_X 4

This one is a bit of a shot in the dark...

Psyc0Flyer, 2017-02-20:
What about this section?

`// Experimental Subdivision of the grid by Catmull-Rom method.
    // Synthesizes intermediate points to produce a more detailed mesh.
    //
    //#define ABL_BILINEAR_SUBDIVISION
    #if ENABLED(ABL_BILINEAR_SUBDIVISION)
      // Number of subdivisions between probe points
      #define BILINEAR_SUBDIVISIONS 3`

Bob-the-Kuhn, 2017-02-20:
Actually, M420 is for all the bed leveling systems in RCBugFix.  Sections are added/deleted based on if mesh leveling is enabled or if one of the Auto Bed Leveling schemes is selected.

I wouldn't recommend using M500 unless something has changed.  The EEPROM literally wears out a little every time it's written.

Definitely disable the EEPROM until you have things working as desired.  The settings in EEPROM override a lot of the settings in your configuration files.  You can find examples in this forum of people pulling out their hair because they'd make a change in their config file but the EEPROM would keep on overriding it with the old value.

To disable the EEPROM just comment out EEPROM_SETTINGS in configuration.h.

Sanity check - go to each of your G29 probe points and use the M48 command to measure the repeatability of your BLTouch.  Your compensated Z height is at best as accurate as the readings you get from M48.

FYI - don't do a G28 after you've done a G29.  G28 turns off bilinear bed leveling.

Increasing the number probe points definitely improves the results of G29.  You're running 3x4 now.  Try 5x5.

Any idea how far off the compensation is after doing the G29 command?

Psyc0Flyer, 2017-02-20:
Looking at the printer, let's say the left 1/2 prints good and the right is
into plate. I am using buildtak so you can imagine my frustration having
gouge marks in my build material..

I have 1st layer settings at 100% and my probe offset is -2.75.

I am not a very comfortable coder so is there any way you can kindly walk
me through what I need to do? I can handle commenting out the line in code
but that is about as far as it goes.. Lol.

Bob-the-Kuhn, 2017-02-20:
I just looked up buildtak.  Nice but not cheap.

You're not actually changing code when you set/enable/disable items in Configuration.h and Configuration_adv.h.  You're just setting parameters and enabling/disabling flags used by the code and the compiler.  You're expected to modify these files and pins_AZTEEG_X3_PRO.h to match your machine.

To disable the EEPROM open the file configuration.h, do a search for EEPROM_SETTINGS and then add **//** at the beginning of the line.

Have you tried changing your probe offset?  Any error in it directly affects what the system thinks Z0 is after G29.

Try using the "paper between nozzle and build surface" method to look for high/low spots with leveling enabled and then disabled.  There should be an obvious improvement.

Psyc0Flyer, 2017-02-20:
I have already done all of that (enable/disable/set) specific lines in code since your post. I have had my printer working for quite some time now and I am an original backer for the BLTouch but it never seemed as though it was working properly since I was using Repetier. I just recently switched to Marlin because I was told that it supports the probe and Rep does not.

When I first load the code, I do the paper/nozzle calibration so I know what the offset would be (M851 Zxxx) but I only have done it in 1 spot (Middle). Is there something that I should do differently? I do not understand how I would do the paper test in multiple spots and only enter 1 # for the offset..

Bob-the-Kuhn, 2017-02-20:
I hope you're using an RCBugFix that's less than a week old.  Older ones had a minimum speed limit where if you probed too slow the BLTouch signal could be missed which results in the head being pushed into the build plate until you hit your kill switch.

So you're the one to blame for that 5mS pulse from the BLTouch!  Not the best decision they made.

The Z offset moves everything up/down at once.  Based on your printing experience I suggest modifying it by 0.05 or 0.1 and see what happens.  Sorry, too late at night and I don't remember if going from -2.75 to -2.65 raises or lowers it.

I'm a little surprised at the M48 results.  They imply that each step pulse (when micro stepping is set to 1) results in about 0.025mm movement.  I'm not used to seeing dedicated 3D printers with that course of a minimum physical movement. 

The more I think about it the more I'm convinced your path to better prints is to go to a 5x5 grid and to adjust your Z offset.  Don't forget to put your final offset back into your config file.

Psyc0Flyer, 2017-02-20:
The -2.65 would raise the nozzle away from the bed. The higher the #, the more "into" the bed it goes. I do not think I have micro stepping activated unless it is on by default. You say not to forget to put it back into the config file. What exactly are you speaking of and where?

Bob-the-Kuhn, 2017-02-20:
Z_PROBE_OFFSET_FROM_EXTRUDER in Configuration.h is the variable/setting you need to update.  Once you've set this then you no longer need to manually set it with M851.

At power up or after a reset the offset is loaded from EEPROM if that is enabled.  If the EEPROM is not enabled then it uses the offset from the Configuration.h file.  M851 changes the currently used offset.  A M851 specified value is lost after a reset/powerup unless saved to the EEPROM via the M500 command or manually copied into the config file.

Psyc0Flyer, 2017-02-20:
Here is what I got with 5x5 grid and -2.95 as my offset (center)

READ: Bilinear Leveling Grid:
READ:       0     1     2     3     4     5
READ:  0 +0.17 +0.11 +0.16 +0.19 +0.17 +0.13
READ:  1 +0.20 +0.18 +0.17 +0.18 +0.25 +0.02
READ:  2 +0.16 +0.09 -0.06 -0.11 -0.05 -0.07
READ:  3 +0.28 +0.26 +0.12 +0.03 +0.02 -0.08
READ:  4 +0.10 +0.03 -0.07 -0.06 +0.01 -0.04
READ:  5 +0.27 +0.19 +0.09 +0.10 +0.15 +0.07
READ: 
READ: Subdivided with CATMULL ROM Leveling Grid:
READ:         0        1        2        3        4        5        6        7        8        9       10       11       12       13       14       15
READ:  0 +0.16875 +0.14653 +0.12431 +0.11375 +0.12352 +0.14495 +0.16375 +0.17611 +0.18583 +0.19000 +0.18639 +0.17722 +0.16625 +0.15412 +0.14019 +0.12625
READ:  1 +0.18282 +0.16598 +0.14913 +0.14069 +0.14664 +0.16099 +0.17421 +0.18327 +0.19119 +0.19755 +0.20627 +0.21342 +0.20671 +0.17742 +0.13427 +0.09111
READ:  2 +0.19690 +0.18543 +0.17396 +0.16764 +0.16976 +0.17703 +0.18468 +0.19043 +0.19656 +0.20509 +0.22616 +0.24963 +0.24718 +0.20072 +0.12835 +0.05597
READ:  3 +0.20250 +0.19356 +0.18463 +0.17750 +0.17259 +0.16949 +0.16875 +0.16940 +0.17241 +0.18125 +0.20972 +0.24403 +0.24625 +0.19306 +0.10778 +0.02250
READ:  4 +0.18810 +0.17579 +0.16348 +0.14926 +0.12994 +0.10871 +0.09324 +0.08513 +0.08278 +0.08912 +0.11722 +0.15401 +0.16319 +0.12349 +0.05619 -0.01111
READ:  5 +0.16523 +0.14670 +0.12817 +0.10393 +0.06699 +0.02434 -0.00866 -0.02733 -0.03636 -0.03440 -0.01160 +0.02218 +0.03875 +0.02296 -0.01005 -0.04306
READ:  6 +0.16000 +0.13875 +0.11750 +0.08750 +0.03903 -0.01819 -0.06375 -0.09102 -0.10662 -0.11000 -0.09380 -0.06537 -0.04625 -0.04671 -0.05648 -0.06625
READ:  7 +0.19764 +0.18197 +0.16630 +0.13958 +0.09116 +0.03169 -0.01792 -0.05132 -0.07486 -0.08662 -0.07964 -0.06088 -0.04935 -0.05380 -0.06549 -0.07718
READ:  8 +0.25292 +0.24633 +0.23975 +0.22056 +0.17828 +0.12340 +0.07472 +0.03715 +0.00578 -0.01532 -0.01885 -0.01212 -0.01301 -0.03004 -0.05469 -0.07935
READ:  9 +0.27625 +0.27421 +0.27218 +0.25750 +0.22037 +0.17060 +0.12500 +0.08755 +0.05426 +0.03000 +0.02231 +0.02366 +0.01625 -0.00875 -0.04250 -0.07625
READ: 10 +0.22806 +0.22110 +0.21414 +0.19718 +0.16165 +0.11611 +0.07625 +0.04624 +0.02189 +0.00634 +0.00708 +0.01661 +0.01560 -0.00589 -0.03792 -0.06995
READ: 11 +0.14792 +0.13150 +0.11508 +0.09282 +0.05789 +0.01711 -0.01486 -0.03316 -0.04264 -0.04329 -0.02778 -0.00343 +0.00787 -0.00552 -0.03195 -0.05838
READ: 12 +0.10500 +0.08134 +0.05769 +0.03125 -0.00361 -0.04125 -0.06750 -0.07662 -0.07435 -0.06375 -0.03815 -0.00421 +0.01500 +0.00690 -0.01593 -0.03875
READ: 13 +0.13477 +0.10908 +0.08339 +0.05565 +0.02033 -0.01703 -0.04190 -0.04752 -0.04063 -0.02694 -0.00116 +0.03144 +0.04926 +0.04020 +0.01637 -0.00745
READ: 14 +0.20176 +0.17625 +0.15074 +0.12282 +0.08653 +0.04783 +0.02218 +0.01752 +0.02591 +0.03903 +0.06032 +0.08634 +0.09838 +0.08554 +0.05872 +0.03190
READ: 15 +0.26875 +0.24343 +0.21810 +0.19000 +0.15273 +0.11268 +0.08625 +0.08255 +0.09245 +0.10500 +0.12181 +0.14125 +0.14750 +0.13088 +0.10106 +0.07125

m600x, 2017-02-23:
I have the same problem (briefly mentioned in issue #5817 ) but the fact that I didn't own a legit BLTouch lead me to rethink if it was a sensor problem or not. You seems to have a legit one and the same problem as me. Maybe I screwed up Marlin and it's not the sensor's fault!

The `G29`would effectively compensate for the bed leveling but not enough in one side of the printer. I do know it do his job (stepper whining) and with a heavily tilted bed (-10mm in one side), you can see it working. But still like 0.2mm error in one side (X max in my case).

I tryed with linear, bilinear. 2, 3 or 4 points. With and without EEPROM. With the lastest RCBugFix. No luck.

Here is the picture of mine, I wonder where your error occur (which axis):
[[image]](http://reho.st/view/self/1d200d4d5935551509c8e80ea1697e57012bd41d.jpg)

Bob-the-Kuhn, 2017-02-23:
@m600x 

I'm guessing here ... but is the 0.2mm error section outside the probe area?  Can you modify the **xxx_PROBE_BED_POSITION** settings to include more of the error section?   With a BLTouch type of probe there'll always be a strip along two edges that the probe can't get to.

You could try a 5x5 bilinear grid.  That was the magic bullet on a couple of issues but those were for errors within the probed area.

Really long shot - try enabling **ABL_BILINEAR_SUBDIVISION** in **configuration.h**.

By chance is the bed tilt primarily in the X direction and are you running **Z_DUAL_STEPPER_DRIVERS**?  If yes to both then you could consider using **Z_DUAL_ENDSTOPS**.  This allows you to tilt the X carriage.  Both of these are in **configuration_adv.h**.

----

I'm curious about your BLTouch type probe.

If it's a Geeetech 3D Touch then you have no worries.  They did a good job of emulating the BLTouch.

If it's not a Geeetech then I have a utility that I'd like you to run.  I've proposed a Z servo probe utility that will tell you if the probe is BLTouch compatible.

m600x, 2017-02-23:
The probing points are is within the printing zone. I modified the max X and Y of the printer to compensate the offset of the probe (`X -16 Y -78`) so the probe will still reach the whole printable bed.

I'm currently set as:
```
#define AUTO_BED_LEVELING_BILINEAR

#if ENABLED(AUTO_BED_LEVELING_LINEAR) || ENABLED(AUTO_BED_LEVELING_BILINEAR)
  #define ABL_GRID_MAX_POINTS_X 4
  #define ABL_GRID_MAX_POINTS_Y 4
  #define LEFT_PROBE_BED_POSITION 0
  #define RIGHT_PROBE_BED_POSITION 180
  #define FRONT_PROBE_BED_POSITION 10
  #define BACK_PROBE_BED_POSITION 150
  #define MIN_PROBE_EDGE 10
  #if ENABLED(AUTO_BED_LEVELING_BILINEAR)
    #define ENABLE_LEVELING_FADE_HEIGHT
    #define ABL_BILINEAR_SUBDIVISION
    #if ENABLED(ABL_BILINEAR_SUBDIVISION)
      #define BILINEAR_SUBDIVISIONS 3
    #endif
#endif
```

So yes, the subdivision is already activated. I'll try increasing the grid to 5x5 and update the result. And no, there's no dual stepper set in the firmware.

The printer is a Leapfrog Creatr where all the electronics have been replaced to RAMPS.
I just updated to the lastest RCBugFix (22feb #f9e4717, was on 11feb #0369f97). And see if it change something.

---
As for the probe type, mine seems to behavior like a retarded clone: deployed pin is always triggered, there's no pulsing involving. And sure, I can run your utility!

m600x, 2017-02-23:
**TL;DR: 5x5 doesn't solve the problem.<br>**
- Lastest RCBugFix<br>
- Bilinear with subdivision<br>
- No EEPROM
- Probe inside print area
- Firmware at: https://github.com/m600x/Leapfrog-Creatr-on-RAMPS-with-Marlin/tree/RCBugFix/Marlin

I did what you suggest me to do: increasing the grid point from 4 to 5 in bilinear leveling with subdivision activate.<br>
Here are the result:<br>

**4X4**<br>
Measured (in mm):<br>
0,23	0,24	0,30<br>
0,23	0,21	0,26<br>
0,22	0,22	0,22<br>
[[image]](http://reho.st/view/self/860da71ba454bf974b9b403af8a4c8248f214eec.jpg)

**5x5**<br>
Measured (in mm):<br>
0,22	0,20	0,33<br>
0,23	0,25	0,30<br>
0,23	0,23	0,22<br>
[[image]](http://reho.st/view/self/80b8b38bc07283f3253d42fefac827cf47802824.jpg)

So it is roughly the same result. Increasing did not solve the problem unfortunately.

The following code is used to produce the result below:
```
G28
G29 V4 T
```

**Bilinear 4x4 with subdivision 3**

```
Bilinear Leveling Grid:
      0     1     2     3
 0 -0.07 -0.05 -0.08 -0.18
 1 +0.08 +0.07 +0.01 -0.09
 2 +0.07 +0.09 +0.02 -0.14
 3 +0.08 +0.13 +0.07 -0.13

Subdivided with CATMULL ROM Leveling Grid:
        0        1        2        3        4        5        6        7        8        9
 0 -0.07448 -0.06246 -0.05044 -0.04583 -0.05152 -0.06462 -0.08385 -0.11084 -0.14396 -0.17708
 1 -0.01599 -0.01019 -0.00439 -0.00469 -0.01333 -0.02808 -0.04830 -0.07560 -0.10837 -0.14115
 2 +0.04250 +0.04208 +0.04167 +0.03646 +0.02486 +0.00845 -0.01275 -0.04037 -0.07279 -0.10521
 3 +0.08177 +0.07838 +0.07498 +0.06719 +0.05370 +0.03582 +0.01302 -0.01651 -0.05096 -0.08542
 4 +0.08798 +0.08710 +0.08622 +0.07975 +0.06608 +0.04681 +0.02116 -0.01327 -0.05408 -0.09489
 5 +0.07498 +0.07985 +0.08472 +0.08189 +0.06911 +0.04863 +0.01952 -0.02140 -0.07095 -0.12051
 6 +0.06510 +0.07541 +0.08571 +0.08646 +0.07487 +0.05372 +0.02188 -0.02463 -0.08185 -0.13906
 7 +0.06684 +0.08097 +0.09510 +0.09855 +0.08829 +0.06734 +0.03414 -0.01590 -0.07818 -0.14047
 8 +0.07170 +0.08935 +0.10699 +0.11308 +0.10443 +0.08423 +0.05041 -0.00226 -0.06854 -0.13482
 9 +0.07656 +0.09772 +0.11889 +0.12760 +0.12058 +0.10112 +0.06667 +0.01138 -0.05889 -0.12917
```

Heatmap version: https://plot.ly/~M600/3/
[[image]](https://plot.ly/~M600/3/)

```
Bilinear Leveling Grid:
      0     1     2     3     4
 0 -0.15 -0.07 -0.05 -0.08 -0.12
 1 +0.12 +0.10 +0.12 +0.06 -0.06
 2 +0.12 +0.10 +0.08 +0.04 -0.07
 3 +0.07 +0.16 +0.16 +0.06 -0.09
 4 +0.13 +0.19 +0.19 +0.12 -0.05

Subdivided with CATMULL ROM Leveling Grid:
        0        1        2        3        4        5        6        7        8        9       10       11       12
 0 -0.14688 -0.12091 -0.09495 -0.07448 -0.06136 -0.05374 -0.05156 -0.05594 -0.06576 -0.07760 -0.09101 -0.10644 -0.12187
 1 -0.04697 -0.03409 -0.02121 -0.00982 +0.00150 +0.01134 +0.01397 +0.00619 -0.00879 -0.02714 -0.04870 -0.07363 -0.09855
 2 +0.05293 +0.05273 +0.05252 +0.05484 +0.06437 +0.07642 +0.07949 +0.06833 +0.04818 +0.02332 -0.00639 -0.04081 -0.07523
 3 +0.12240 +0.11331 +0.10422 +0.10000 +0.10706 +0.11898 +0.12135 +0.10812 +0.08534 +0.05677 +0.02174 -0.01908 -0.05990
 4 +0.14263 +0.13040 +0.11818 +0.11051 +0.11297 +0.11998 +0.11944 +0.10699 +0.08698 +0.06042 +0.02544 -0.01609 -0.05762
 5 +0.13243 +0.12126 +0.11010 +0.10152 +0.09870 +0.09846 +0.09387 +0.08351 +0.06881 +0.04705 +0.01496 -0.02418 -0.06333
 6 +0.11771 +0.11142 +0.10513 +0.09896 +0.09392 +0.08900 +0.08125 +0.07112 +0.05816 +0.03802 +0.00687 -0.03146 -0.06979
 7 +0.10177 +0.10694 +0.11210 +0.11426 +0.11305 +0.10883 +0.09965 +0.08534 +0.06607 +0.04035 +0.00557 -0.03566 -0.07689
 8 +0.08133 +0.10175 +0.12218 +0.13598 +0.14167 +0.14073 +0.13102 +0.11065 +0.08149 +0.04703 +0.00665 -0.03905 -0.08474
 9 +0.07240 +0.10318 +0.13397 +0.15573 +0.16620 +0.16765 +0.15781 +0.13385 +0.09861 +0.05833 +0.01343 -0.03652 -0.08646
10 +0.08416 +0.11588 +0.14760 +0.17006 +0.18078 +0.18224 +0.17267 +0.14988 +0.11605 +0.07591 +0.02892 -0.02439 -0.07770
11 +0.10745 +0.13520 +0.16295 +0.18243 +0.19128 +0.19185 +0.18295 +0.16381 +0.13519 +0.09811 +0.05011 -0.00635 -0.06281
12 +0.13073 +0.15451 +0.17830 +0.19479 +0.20177 +0.20147 +0.19323 +0.17774 +0.15432 +0.12031 +0.07130 +0.01169 -0.04792
```

Heatmap version: https://plot.ly/~M600/1/
[[image]](https://plot.ly/~M600/1/)

Bob-the-Kuhn, 2017-02-24:
Do  M48 at a few of the "good" spots and a few of the "bad" spots.  The spread should be no more than the movement 2 steps of the motor would produce.  

In my case I have 200 step motors.  1 step results in 0.027mm movement.  I normally see my spread between 0 & 0.054mm.

At best your grid will only be as accurate as the spread you see with M48.

Bob-the-Kuhn, 2017-02-24:
I've got to learn to put my machine back to normal after troubleshooting a problem.  Nothing like a few moved pins and an open connection to drive you crazy.

I've finally been able to test my utility with your branch.  Attached is the modified file.

To run the test issue a **M43 S1** command.  You'll need to enable **PINS_DEBUGGING** near the end of **Configuration_adv.h**.

[Marlin_main.zip](https://github.com/MarlinFirmware/Marlin/files/798716/Marlin_main.zip)

m600x, 2017-02-25:
Your utility seems to work (can't confirm, ain't got a real one):

```
Send: M43 S1
Recv: Servo probe test
Recv: .  using index:  0
Recv: .  deploy angle: 10
Recv: .  stow angle:   90
Recv: . probe uses Z_MIN pin: 18
Recv: . uses Z_MIN_ENDSTOP_INVERTING (ignores Z_MIN_PROBE_ENDSTOP_INVERTING)
Recv: . Z_MIN_ENDSTOP_INVERTING: true
Recv: . deploy & stow 4 times
Recv: CLONE detected
Recv: .  DEPLOYED state: HIGH (logic 1)
Recv: .  STOWED (triggered) state: LOW (logic 0)
Recv: ok
```

As pointed out by playlet, his workaround seems to be disabling `ENABLE_LEVELING_FADE_HEIGHT`which I just did to see if it does improve things.

Will keep updated.

cozierad, 2017-02-25:
@m600x did disabling the fade height make any difference for you?

I just realized my leveling was not working in a small increments as well. 

It does not appear that disabling the fade height made a difference for me.

closed 2017-10-31 by thinkyhead