MarlinFirmware/Marlin #5869: M420 S1 "forgets" Z

MoonshineSG, 2017-02-20:
I activate/de-activate the bed levelling matrix at the beginning and at the end of a print, respectively (as well as if a print is canceled)

After issuing the `M420 S0`, the Z axis gets reseted to 0. 

This behaviour can easily be replicated:

```
[...]
Send: G28
Recv: echo:Active Extruder: 0
[...]
Recv: echo:Active Extruder: 0
Recv: X:51.00 Y:8.00 Z:6.85 E:0.00 Count X: 4080 Y:640 Z:2740
[...]
Send: G1 Z10
[...]
Send: M114
Recv: X:51.00 Y:8.00 Z:10.00 E:0.00 Count X: 4080 Y:640 Z:4000
[...]
Send: M420 S1
Recv: echo:Bed Leveling On
[...]
Send: M114
Recv: X:51.00 Y:8.00 Z:0.00 E:0.00 Count X: 4080 Y:640 Z:4000
[...]

```

I'm not sure if this is normal, or needs to be looked into...

FHeilmann, 2017-03-01:
As a quick fix, make sure you use `M420` **before** you do your Z-Probing. That way your Z-Position doesn't get messed up

FHeilmann, 2017-03-01:
Thats not what I mean,
i mean doing this:
```
G28 XY
M420 S1
G28 Z
```
or
```
M420 S1
G28
```
instead of
```
G28
M420 S1
```
The "homed z-position" is lost when a `M420` is issued, so home your Z-Axis after the `M420` command

MoonshineSG, 2017-03-01:
Ah... ya... that's exactly what I am trying to avoid. (obviously that's what I am doing now, as the Z position is lost)

Let me explain. As I am using an induction probe, if I do a a Z probe after a cancel or a consecutive print, while everything is hot the readings are off. Only by just a little, but sufficient to screw up the first layer.

I am not sure why M420 forgets the Z value, but I don't think that should happen.

FHeilmann, 2017-03-01:
Hm, would it make sense to always probe when your bed is hot?

At least thats what I am doing with my nozzle probe right now. Home XY -> move to purge position -> heat up -> purge -> home Z

That way you could adjust your first layer for the hot state which would eliminate the differences you're seing between your first print and consecutive prints/resumes.

MoonshineSG, 2017-03-01:
yes, i used to do something similar... if only the bed would be always at the same temperature... but... PLA 60C... ASA 110C.... big difference.

FHeilmann, 2017-03-02:
I retested this on my machine and found the following issues:

- G28 turns off bed leveling for whatever reason, even if it was enabled before isuing the G28 command
- Issuing: 
```
G28 Z
M420 S1
M114
``` 
 Reports Z as zero, but you can still move "below Z", so in my case, after G28 my nozzle is at 2mm and I can move downwards this far even though M114 reports Z=0 
- The Z-Offset is messed up after this procedure: Mine is .5mm too low which will cause the nozzle to crash into the bed if I attempt to lower the nozzle by 2mm. Looking at where this .5mm discrepancy is coming from, the only "similar" value I found was from my ABL Bilinear correction matrix, which reads -.5mm at my G28 Z homing point.
@thinkyhead meaybe this is related to the other leveling issues we are seing?

closed 2017-10-31 by thinkyhead