MarlinFirmware/Marlin #5878: Z offset isn't working W/ BLTouch

noobStatus, 2017-02-20:
I noticed some odd behavior in RC8 regarding Z offset.  

Here's my test:  

M851 Z0

G28

G1 Z0

Adjust Z height with paper.  Mine ended up at -0.11.

M851 Z-0.11

M500.  I confirmed that the setting was saved in the EEROM.

My starting script:

G28
G29

The issue is, no matter what the offset is, it starts printing at 0 no matter what.  I attached my configuration file.  Any idea on this?
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/788611/Configuration.zip)

ts4s, 2017-02-21:
Try just doing a:

G28
G29
M500
G1 Z0
Then Dial in your offset
M851 Z-0.11(update as needed)
M500

Startup script should have the first 3 lines above in it.

I just went through this on my printer. Change the config several time and then started all over before I got it.

Later I can look at the config if someone else doesn't do it be for then.

noobStatus, 2017-02-21:
I've been auto leveling every print for now.  Still need the m500 in the startup script?

mperdue, 2017-02-21:
I wouldn't put M500 in your startup script. That would cause an EEPROM write every time you start up which will toast your EEPROM in fairly short order. There's a limit to the number of times you can write to EEPROM.

Roxy-3D, 2017-02-21:
> There's a limit to the number of times you can write to EEPROM.

Some of the AVR's are spec'ed at 10,000 writes per location.  But I think most of the parts are at 100,000 writes now.

But yeah...   You don't want to be pounding on the EEPROM all the time.

ts4s, 2017-02-21:
Well, I stand corrected... 🙄... Guess that's due to me working with 2 different build plates. Sorry, but good to know 😉

noobStatus, 2017-02-21:
It's working! 

I ran M502 and started the process over again.  This time when I G1 Z0 it actually used the offset!  No more display lag either, which is awesome.

I have no idea what fixed it, though.  

Great work on the firmware, everyone!

closed 2017-02-21 by noobStatus