MarlinFirmware/Marlin #5879: EEPROM backup issue ?

spilz87, 2017-02-20:
Hi,
I didn't a lot of tests, but I feel like there is an issue with EEPROM settings backup when I load a new version of Marlin, specially from RC7 to RC8, on max acceration maybe other parametters

is it normal ?
does someone else observed this ?

As I need to print some part, I can't test it more right now

Bob-the-Kuhn, 2017-02-20:
The EEPROM layout changed from V24 on RC7 to V27 on RC8 to V29 on the latest RCBugFix.

Looks like when an older version of the EEPROM is detected it is automatically overwritten with values from the current config files.  

Looks like your old EEPROM config settings are gone.

KevMags, 2017-02-21:
When uploading a new version number the EEPROM gets erased.  When uploading the same version the EEPROM is unchanged.

spilz87, 2017-02-22:
Actually I just update the same version and PID values where lost whereas I saved them before with LCD menu

psavva, 2017-02-23:
Suggest to cost this incident if you're happy with the result

spilz87, 2017-02-23:
Actually I'm not, the issue is still here as I lost saved values

psavva, 2017-02-23:
By any chance, have you disabled the EEPROM settings in the config file?
Did you re-upload the same version of the firmware, or an older version with a change EEPROM version?

Can you provide extract steps to replicate, and exactly what values were set, and lost?
Were all values lost?

spilz87, 2017-02-23:
I didn't disable EEPROM settings
I load the same firmware version, I was working on an issue with langage file and accent.
I didn't check all values, just I saw that PID values have changed 

I will try to replicate it to give you exact steps

Bob-the-Kuhn, 2017-02-23:
Please include the EEPROM version of the old and new when you try to replicate.

You'll find it in the file **configuration_store.cpp**.   At about line 40 you'll find a statement similar to **#define EEPROM_VERSION "V29"**.  

Don't bother doing anything more if the EEPROM versions are different.  If version XX powers up and finds the EEPROM is version YY then it automatically erases all the EEPROM and writes the values from the **configuration.h** and **configuration_adv.h** files into EEPROM.  Some things like the bilinear autolevel grid aren't in the config files so those items are gone.

At the very beginning of this thread you said something about going from RC7 to RC8.  There definitely was an EEPROM version change between the two.

Bob-the-Kuhn, 2017-02-24:
I was wrong.  

When the EEPROM version changes the EEPROM is **NOT** automatically overwritten.  What happens is:
- Most settings are copied from the config files into RAM.
- The RAM settings that aren't in the config files are set to either an OFF state or a benign state.

Only a M500 command will change the EEPROM.

That means your settings are still in EEPROM **BUT** you can't see or use them unless you're running the code that uses that version.  Unfortunately there's currently no way to know what version is in the EEPROM.

psavva, 2017-02-24:
There is one thing which I can still think of they may erase the EEPROM...

If your EEPROM checksum gets corrupted, Marlin will re-read the configuration files.

I'm guessing it may be the case here as well...

It would also be very difficult to replicate.

thinkyhead, 2017-03-01:
Upon a new Marlin install, we recommend doing `M502` followed by `M500` to save the default settings to EEPROM. You can save/restore your old EEPROM settings by using `M503 S0` and copying the GCode output. Re-play this GCode to restore your old saved settings.

closed 2017-10-31 by thinkyhead