MarlinFirmware/Marlin #5880: END report issue

spilz87, 2017-02-20:
Hi,

at the end of a print, the "time report" shows only minutes and seconds, not hours.
maybe I did Something wrong with a personal setting.

can someone tell me where is the line which write this report at the end of print (for REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER), I don't find it.

thanks for your help

regards

Bob-the-Kuhn, 2017-02-21:
I'll try to learn something new and hopefully help you out.

Is there room for at least two more digits at the end of the display (for hours and a separator)?

Exactly what is the message you see?  Most of the time the beginning of the message is a predefined string that I can use Astrogrep to search for.

spilz87, 2017-02-21:
yes, there is lot of room for this, it worked with previous version.
I see somthing like "01m23s" on the bottom left.

I don't find where is the line of code which is at the end of a SD print.

Bob-the-Kuhn, 2017-02-21:
> worked with previous version

Which version works & which doesn't?

Please try the latest RCBugFix.  It has a lot of fixes in it.

spilz87, 2017-02-21:
it worked on RC6 but not in RC8
i took the RCBugFix this night...
But I had a specific part do display the time left and not the time past, but it didn't have effect on RC6, so I would like to know where is the line at the end of a SD printig to understand if it comes from my code or not

Bob-the-Kuhn, 2017-02-21:
Is it working as you want with RCBugFix?

If yes then I'll go through the changes and see if I can find the one that fixed it.

I'm a little confused on what you're doing.  Are there two items/issues?
1. At end of a print, the LCD shows only elapsed minutes & seconds but it should also show hours.  Might have been fixed by going to RCBugFix.
2.  During a print you want to display the time remaining to the end of the print.  You had custom code in RC6 that implemented this.  Now you want to apply that custom code to the new Marlin release but you can't find where the new Marlin generates the report seen in the bottom left of the screen.

spilz87, 2017-02-21:
No there is only one issue : 
at the end of a print, the LCD don't show hours.

The custom code I added to display the time remaining works well on all version (I modify it at each version to be close to the code of the version).
But as there is this issue at the end of a print, I want to understand where is this code to find if there is interaction with mine

Bob-the-Kuhn, 2017-02-21:
As best I can tell the "toString(" function in duration_t.h returns the elapsed time.  It automatically formats the string based on the total time.  It stops building the string when it hits one with a zero value.

spilz87, 2017-02-21:
I think you are right, the string is created in "toString(" function in duration_t.h
it looks like it's used in "gcode_M31()" , it's the only part where I see it for the use in end of a print, but I don't have M31 in my gcode, so I don't rely understand where it's called ...

thanks for helping me

Bob-the-Kuhn, 2017-02-21:
Doing a Grep on "toString(" shows that it's called in ultralcd.cpp

spilz87, 2017-02-21:
Yes, but only to display statistic datas, isn't it ?

Blue-Marlin, 2017-02-21:
May be a bit of a strange question.
Are you sure the message is produced by Marlin - or is there a chance the message is produced by the host or slicer and sent via a M117 command?

Bob-the-Kuhn, 2017-02-22:
Does the elapsed time update when not connected to a host?

I doubt that the slicer is generating M117 commands.  Mine doesn't.  The slicer would have to know how fast your machine moves before it could calculate elapsed time and if there is a speed override in place.

spilz87, 2017-02-22:
it's not product by the host, the printer is not connected to a computer, I only use SD card
there is no M117 in the GCODE (only one in the start gcode which works well).

Actually, before when I worked, I shown to me the real time of the print (not the estimation calculed by the slicer), so it comes from Marlin

spilz87, 2017-02-22:
end of my last print, I got on the bottom left : 
"3m 33s"

I guess I found the issue
It comes from my additionnal code to display the time remaining, I comment
 `// printer_job_timer.duration()` line 509 in ultralcd_impl_DOGM.h 
So maybe it doen't increase the counter and display only the time before the start of the print (when hotend is heating)

is it possible ?

Bob-the-Kuhn, 2017-02-22:
As best I can tell the print timer is started when the "print from SD card" routine is kicked off.

Try adding the M75 start print timer command to the beginning of the g-code file & see if that gives you the desired results.

You might also try adding **#define DEBUG_STOPWATCH** to your configuration.h file.  That'll print out when the print timer starts, pauses & stops.  It may print it out the serial port rather than the LCD.

spilz87, 2017-02-23:
Just did a test, that was not the solution :(

Bob-the-Kuhn, 2017-02-23:
By chance is **#define PRINTJOB_TIMER_AUTOSTART** commented out in **configuration.h**?

thinkyhead, 2017-02-26:
> I don't really understand where it's called

In `CardReader::printingHasFinished`…
```cpp
enqueue_and_echo_commands_P(PSTR("M31"));
```

spilz87, 2017-02-26:
I found it thanks :)
I think the issue is from my additional code to read comment lines (to get total printing time and display remaining time instead of past time)

thinkyhead, 2017-03-01:
@spilz87 Cool. If the code works okay without your modification, you can close this issue.

spilz87, 2017-03-01:
The thing is that it works well for small print (<1h) but I don't know if it works for longer print.
I close the issue and will reopen it if I still have the issue next time I have a long print.
Thanks for your help
Regards

spilz87, 2017-08-19:
Actually I did test and debbug and found where the issue is from : 

During the start of the print, gcode_M109() is called , specially : `#if ENABLED(PRINTJOB_TIMER_AUTOSTART)
      /**
       * Use half EXTRUDE_MINTEMP to allow nozzles to be put into hot
       * standby mode, (e.g., in a dual extruder setup) without affecting
       * the running print timer.
       */
      if (parser.value_celsius() <= (EXTRUDE_MINTEMP) / 2) {
        print_job_timer.stop();
        LCD_MESSAGEPGM(WELCOME_MSG);
      }
      else
        print_job_timer.start();
    #endif`
but `parser.value_celsius() ` return 0 so print_job_timer is stopped :( 

this is the origin of the issue, but I don't understand how parser.value_celsius() works. 

it is call during my start gcode, somewhere in G90 or M82

closed 2017-11-14 by thinkyhead