MarlinFirmware/Marlin #5881: Feature Request: Dual X and Y endstops.

paulpenney, 2017-02-21:
I'll try and make this short, but I believe this is a very much needed feature.

There is a large and growing community of makers who are making the MPCNC from Vicious1.com. If you are not familiar, it is a low cost CNC that runs primarily Marlin, using dual X and Y gantry motors and EMT conduit driven through belts. 

Because the gantry is using belts and independent, it is easy for it to get "out of square" in between cuts unless you go through a testy calibration process. Dual endstops on X and Y (that mimicked the existing dual Z end stop functionality) would solve this immediately. 

This would also solve any issue where the end user needed to square up two motors on the same axis that aren't connected and are independent of one another.

Let me know what you think

Paul

Bob-the-Kuhn, 2017-02-21:
To implement this the controller would need to have 6 independent stepper channels.  I'm not aware of a low cost controller with that many.

Do you think that people that build a low cost machine will buy a medium cost controller?

paulpenney, 2017-02-21:
Bob,

Not 6 independent stepper channels necessarily. The MPCNC only uses 5 stepper motors (two on X axis, two on Y axis, and one in Z axis. I am using an MKS base on mine, with all 5 channels of the board running all 5 steppers using the new dual x and dual y stepper functionality in marlin. Any controller with 5 stepper drivers will do (MKS base / gen, not sure if Ramps 1.4 has 5?)

What I am requesting is for the software to replicate the Z_DUAL_ENDSTOPS functionality on the x and y axis. I have tried this but am not a coder and don't understand enough to do it successfully. This would allow you to use 4 endstops on the MPCNC x and y axis... one endstop for each motor. That way all the motors could be squared with the frame for every operation. 

Paul

Bob-the-Kuhn, 2017-02-21:
I keep thinking "3D printer".

Yes, the RAMPS1.4 based solutions have 5 independent channels and are fairly low cost.

I have dual Z endstops enabled on my printer.  I love it.

You're right that all the base code is in place.  All that's need is someone to make the mods and someone to test them.  That's the harder part.

paulpenney, 2017-02-22:
Yes! No worries! It is a very popular machine, hundreds of builds, and I am sure this would be a worthwhile mod.. I've tried it once, I got the extra end stops registering but was never able to get the axis working the way I wanted. I think a good place to start might be to take the original Z2 code, change it to Y2 or X2, and see if it could even be adapted for an axis other than z? I wish I was a programming wizard like thingy!

boelle, 2019-02-18:
since there was a PR that made this possible and its merged should this one not be closed?

closed 2019-03-17 by p3p