MarlinFirmware/Marlin #5882: design proposal

averydarkwhite, 2017-02-21:
Hi guys, 
thanks for the great work until now!

I had my Kossel Machine running on an earlier version of marlin for a long time, until I wanted to implement a bed levelling mechanism. This led me to trying out several different versions of Marlin, until I ended up with the RCBugFix version, which runs fine. The annoying thing was to enter all my values again every time I try a new version. Having nearly all those collected in the configuration.h makes it a lot easier, but since the configuration.h changes with every version, it's not possible just enter my printer's values once and then copy it everytime I install a new Version. would something like this be possible?

thanks for reading!

fiveangle, 2017-02-21:
Configuration.h is designed to be migrated manually btw. Ideally you compare the original version of your current build with the original version of the build you're moving to in order to highlight any major configuration changes (using something like Beyond Compare, diff, or even github web UI); if none, drop your Configuration.h in and compile; if some, hand migrate them to your Configuration.h. 

Regarding saving all bed leveling detail into EEPROM across upgrades, that sounds like a potential troubleshooting disaster if anything goes wrong with the mechanism due to regression or otherwise — only after resetting to factory defaults would one be able to find if there's a problem (which is typically viewed as a last resort). For MBL it would probably be fine though since it would be easy to compare an S0 report to your known values, assuming you knew them. Although even that could be problematic if people began to rely this new feature to keep their MBL values safe rather that having them saved elsewhere.

Are you sure this would be worth it in a the long run given the potential for problems ? One thing that should be central to robotic design philosophy is ensuring a bot doesn't damage itself, and operations around bed leveling  seems to be the primary culprit for bed gouging of FDM printers.

christianh17, 2017-02-21:
Hi!
In the past I used "meld" to synchronize the different config-files. This helped a lot. Now I use github (the desktop app), have my own local repository. Every time I reflash my printer I create a new branch (tags are not supported within the gui and I want to have it easy) and the working repository is updated from rcbugfix. I do this since about 4 month and there was only once a patch which could not be merged automatically. So normaly I have no effort to switch to a new rcbugfix version.

Even my small modifications survived...

Regards
Christian

teemuatlut, 2017-02-21:
Something you could consider is making your own myConfiguration.h file.
Inside you would write something like
```cpp
#undef HEATER_0_MAXTEMP
#undef X_PROBE_OFFSET_FROM_EXTRUDER
#define HEATER_0_MAXTEMP 250
#define X_PROBE_OFFSET_FROM_EXTRUDER 0
```
etc...
Also you would have to write the line "#include "myConfiguration.h" to the end of Configuration.h

Bob-the-Kuhn, 2017-02-21:
@christianh17 - could you supply the details on how you did the update without loosing the existing info?

I just tried to do an update & failed to keep my customizations. Idid the following:
1. created a new repository
2. copied all my files into it as the master
3. created a branch from it
4. went to Gitshell
5. git remote -v
6. git remote add upstream https://github.com/MarlinFirmware/Marlin.git
7. git fetch upstream
8. git rebase upstream/RCBugFix

ruggb, 2017-02-22:
@teemuatlut I think I will try that, though I do not think the variables need to be undefined. Doesn't redefining them suffice? One of the issues is that there are a lot more variables that can be tweaked that are not even talked about, and they are buried in files not designated as part of a configuration, like LCD timeout in the ultralcd.h file. Another issue is that new builds do not come with enough information RE configuration changes, so that it is necessary to do an optical compare to find anything that might affect you. Another issue is that the config files contain code, which make things more confusing. I am sure a file could be created that implemented all those IF code lines that a user would not have to be concerned with and the only thing in the config files were #define.... statements.
SO, in conclusion:
1. Put ALL hardware specific #define statements in one file - or 2, one for absolutely necessary and 1 for optional.
2. Detail changes that affect the config file with the release
3. Put all the IF statements in a separate file.

WHY should this be done - developer would get a lot more people testing their builds if it were a lot simpler to upgrade.

christianh17, 2017-02-22:
@Bob-the-Kuhn:
I am not the fittest in using git via the shell. Usually I like things like using the command line but git "thinks" another way than me so I decided to use the app even knowing there are some limitations.

My way: I cloned the repository into a local git. Than I made a branch and called it "mybranch". In this branch I did all the changes I had done before (modifications like quick access to babystepping and of course the configuration.h). These changes were checked in (into my local repository).

Next step was flashing marlin into the printer, I did this using my repository. Of course there were some minor changes but as soon as I was satisfied I had a "version" for me which I used some time. For knowing what I had done to this version I created a new branch "mybranch-<actualdate>". Of course I did not touch this branch any more. 

For the next update: I synchronized (right word??) my local rcbugfix from the official website and than into my "mybranch". Now the good point: Only when e.g. names of used (!) defines in the config.h are changed I have to look at it. I tested it today, there was the following change in config.h.

- //#define MANUAL_Z_HOME_POS 0 // Distance between the nozzle to printbed after homing
+ //#define MANUAL_Z_HOME_POS 0

As I did not change anything on my own at this line this merge took place without any effort.

I hope I described my way good enough and again: sorry: I am sure there are hundreds of better ways to do this but at least this works for me.

Regards
Christian

Bob-the-Kuhn, 2017-02-22:
@christianh17 - thanks for the response.

I've been doing basically the same thing.  I use Notepad++ with the compare plugin to look for changes in the config files.  Best compare tool I've ever used.

teemuatlut, 2017-02-22:
> I think I will try that, though I do not think the variables need to be undefined. Doesn't redefining them suffice?

I think it will. You'll just get some redefine warnings where you changed a value, but I believe it should work alright.

closed 2017-10-31 by thinkyhead