MarlinFirmware/Marlin #5886: Issue Hotbed thermal runaway

TAz00, 2017-02-21:
Hi,

The power to my printbed gets locked to ON after the initial brimming layer. 
And then proceeds to get as hot as possible. It seems the issue does not start until gcode brimming proces has finished.

[image]

Its a 300x200mm plate so I havnt seen it go over 80c, with the 12v I'm feeding it, and the test print ends before it goes any further.

If I set a bed temp before printing, it will hold that temp just fine. 
[image]

Once the print has started, it makes no difference if I turn the bed temp down with the lcd menu, it will continue to climb as best as possible.

Ive also found that disabling the bed intially, and then turning it on manually midway through the print, leads to the same constantly on runaway.

Slicer: Cura 15.04.6

Tested Firmwares
Marlin-1.0.2-2
Marlin-1.1.0-RC7
Marlin-1.1.0-RC8    - Attached Configuration
Marlin-RCBugFix

Board: RAMPS 1.4

[20mmX20mmX20mm-hollow.gcode.zip](https://github.com/MarlinFirmware/Marlin/files/791107/20mmX20mmX20mm-hollow.gcode.zip)
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/791108/Configuration.zip)

Tannoo, 2017-02-22:
What slicer did you use for that? That is printing really fast. Too much filament in the corners and not enough on the side. So, it not coming out very good.

Tannoo, 2017-02-22:
Mid-way through the print, I set my bed temp (from LCD) to 30, and my bed temp is dropping.

Bob-the-Kuhn, 2017-02-22:
I didn't see any issues in your config files or in the gcode program.

The slicer issues a M190 S50 at the beginning and M140 S0 at the end and nothing else that would affect the bed temperature.

What happens when you manually issue a M190 S50?  That should override any Pronterface settings and keep the bed near 50C.  After the bed has been at 50C for 5 minutes try turning on the fan and the extruder heater and see if the bed stays under control.

Do you have a way to independently verify the temperature of the bed?

Is the following how you have the heaters and the fan connected?
```
PIN:  8   Port: H5        HEATER_BED_PIN              Output = 0
.                         RAMPS_D8_PIN                Output = 0
PIN:  9   Port: H6        FAN_PIN                     Output = 0 
.                         RAMPS_D9_PIN                Output = 0
PIN: 10   Port: B4        HEATER_0_PIN                Output = 0 
.                         RAMPS_D10_PIN               Output = 0
```

TAz00, 2017-02-22:
Thanks for your replies.

Im not sure the bed thermistor is type 11, but this is all the info from the ebay seller 'NTC 3950 Thermistor 100K 1%'

Here I've set the bed temp, and then turned on the extruder. I have the extruder fan connected to the 12v rail, so its always on. But other than that, yes, Port 8 is extruder heater and 10 is the bed.
[image]

To verify the temperatures, when the bed and extruder have been off for a while, they show the same ambient temperature. I havn't actually put a thermometer to the bet, but 30c and 70c and feels right enough. 

150 m/s with a bowden extruder maybe a bit too fast, but you cant argue with the results :P 
[image]

TAz00, 2017-02-22:
Thanks for testing :)

Here's a better test where I've set the bed to 50 for 5 mins, then turned on the extruder, bed temperature is holding steady. 
[image]
[image]
[image]

Tannoo, 2017-02-22:
Can you try a type 1 thermistor? If so and you have PID enabled... tune it again.

 I'm not sure if that's the issue being that you can hold temp at idle. AND that it works fine if you preheat before printing.

I'm at a loss otherwise.

TAz00, 2017-02-22:
Scratch your heads no more

After removing M106 from both LAYER1 and LAYER2 in the gcode, 
M106 S127
M106 S255

My bed no longer gets locked to fully on, but holds temp during print.

M106 controls fan speed, and 255 sets max power, so I suspect there's a short (since you guys checked the config) between the pins for mosfet 9 & 10 

I will check that and report back once the print finishes.
But so far everything's great.

Tannoo, 2017-02-22:
Well, that's bitter-sweet news.

TAz00, 2017-02-22:
Turns out, the heatsinks for 9&10 were touching. Apparently they're live for whatever reason :D 

Pried them apart, and now I'm printing.

closed 2017-02-22 by TAz00

Tannoo, 2017-02-22:
Glad you got it.