MarlinFirmware/Marlin #5891: BLTouch delay time question

mperdue, 2017-02-22:
In Marlin_main.cpp at or near line 1987 is the following:

  #if ENABLED(BLTOUCH)
    void bltouch_command(int angle) {
      servo[Z_ENDSTOP_SERVO_NR].move(angle);  // Give the BL-Touch the command and wait
      safe_delay(375);
    }

I'm curious about where the 375 comes from. It seems to create a relatively long pause during homing and bed leveling for each probe action.

Roxy-3D, 2017-02-22:
I think this number can be reduced.   But I have seen 'dirty' BL-Touch probes deploy and stow more slowly.     If you care to experiment and find a lower bound for that number that works 100% of the time, I would be interested to know what you find.   (We won't be able to shift all the way to your number without getting validation from other BL-Touch users that it works 100% of the time for them too...)

mperdue, 2017-02-22:
It might be nice to make that a define so it can be set in the configuration file in the future. I'm currently running 50 with no problems. I may try reducing it further for testing but that delay isn't too obnoxious. I'm running the BLTouch on to machines so I can test it for repeatability between them. Both are genuine BLTouch probes.

Bob-the-Kuhn, 2017-02-22:
The real question is if you want the BLTouch to fully deploy before starting the probe move or in parallel with the move.

If in parallel then the probe height and probe speed needs to result in a move of at least the 375 mS before getting to the trigger point.

My probe speed is 1000 mm/minute so a 375mS move duration results in a 6.25mm move.  Add in my Z offset of -2.75mm and my minimum probe height would be 10mm.

mperdue, 2017-02-22:
Mine didn't seem to be doing any kind of parallel motion. It would move into position, stop, wait and then probe. This was repeated at each location and or the homing point.

Bob-the-Kuhn, 2017-02-22:
Agreed - it waits for the 375mS delay period and then moves.

If you drop the delay to 50mS then it should start moving before the probe is fully deployed.

mperdue, 2017-02-22:
At 50 it seems to be pretty close to simultaneous. If the probing motion starts first it doesn't seem to be by much.

closed 2017-05-26 by mperdue