MarlinFirmware/Marlin #5894: r restarting) on display after bed heats

G-Forge, 2017-02-23:
I have been running RC8 bf from December, all OK no promlens.
Today I loaded current bugfix. Prints OK, no problem - except
after the bed heats up the 12864 displays **_r restarting)_** and then starts heating the extruder. Print starts OK when it gets to temp and there are no problems at all.

For some reason it looks like it displayed this string 
#define MSG_ERR_STOPPED                     "Printer stopped due to errors. Fix the error and use M999 to restart. (Temperature is reset. Set it after restarting)"
but continued on.

thinkyhead, 2017-02-26:
That's pretty strange. It sounds like a buffer overrun or stack overflow because it's printing strings from random places. The "r restarting)" string comes from here:
```cpp
#define MSG_ERR_STOPPED "Printer stopped due to errors.
                         Fix the error and use M999 to restart.
                         (Temperature is reset. Set it after restarting)"
```
What version of Arduino did you use to do the build and upload?

Blue-Marlin, 2017-02-26:
More intersting to me would be - what should be printed?
Looks to me as if it could be a progmem vs. ram -string problem.

G-Forge, 2017-02-26:
The version of Arduino was 1.6.7 and the Marlin RC8 bugfix of 3/22/17.

thinkyhead, 2017-03-01:
> More intersting to me would be - what should be printed?

Perhaps `MSG_HEATING_COMPLETE` ("Heating done.")

@G-Forge
- Do you have any `M117` commands in your starting GCode?
- Please test `RCBugFix` because this bug may already be gone.

G-Forge, 2017-03-01:
>More intersting to me would be - what should be printed?
Perhaps MSG_HEATING_COMPLETE ("Heating done.")

Yes, I appears when _heating done_ would normally appear. I hae no M117 commands.

Blue-Marlin, 2017-03-01:
What is `const char* const message` in 
```
void lcd_setstatuspgm(const char* const message, uint8_t level) {
```
in ultralcd.cpp and why is it different from
```
void lcd_setstatuspgm(const char* message, const uint8_t level=0);
```
in ultralcd.h?

benlye, 2017-03-02:
Not sure if it's the same issue, but I see a similar type of behaviour after updating to the latest RCBugFix.

When issuing M104 S180 to heat my extruder, my LCD (a variant of the RepRapDiscount Full Graphic Smart Controller) displays "LLED." in the status line.  M104 S0 takes it back to "3D Printer Ready".

benlye, 2017-03-02:
M109 has the same effect - 'LLED.' on the screen until it reaches the specified temperature, then the message changes to 'Heating done.'

I suppose 'LLED.' might come from here:
```
#define MSG_KILLED                          "KILLED. "
```

benlye, 2017-03-03:
Related to #5944 and #5939?

thinkyhead, 2017-03-03:
Should be fixed by #5948.

closed 2017-03-03 by thinkyhead