MarlinFirmware/Marlin #5895: Request: Best routine for filament change/runout

Bubolz, 2017-02-23:
I enabled and running automatic filament switching on my machine (great job) but would like to suggest a more complete routine	
	
01 - Beeps and displays the message on the screen indicating the start of the filament exchange process	
02 - Pause printing	
03 - Retract fast 2mm (to avoid blobs)	
04 - Raises the nozzle 3mm	
05 - Move o X e Y para 0	
06 - Waiting for nozzle temperature decay to 160° (maintains this temperature) displays standby message on the screen with the current temperature	
07 - Slowly retract 50mm (extracts the filament at the temperature of 160° from inside the extruder Cleanly)	
08 - Retracts fast 600mm (extracts The filament of the bowden tube)	
09 - Beep and display message on screen to insert filament	
10 - Heats the nozzle at the printing temperature	
11 - Slow feed 50mm (slow feeding speed until passing through the extruder)	
12 - Feed fast 550 (quickly inserts the filament through the bowden tube)	
13 - Slow feed 50mm (slowly insert the filament to the nozzle)	
14 - Emits a beep warning the end of the routine and asks if the user wants to restart printing or extrude more 50mm	
15 - The nozzle returns to the point where the print has stopped and the process continues.	
	
All values would be configurable parameters in firmware

Roxy-3D, 2017-02-23:
> 06 - Waiting for nozzle temperature decay to 160° (maintains this temperature) displays standby message on the screen with the current temperature
> 07 - Slowly retract 50mm (extracts the filament at the temperature of 160° from inside the extruder Cleanly)

These parts make me especially uncomfortable.  I don't think my PLA filament is going to retract at this temperature.

Bubolz, 2017-02-23:
170º may be more appropriate... It would be nice if all length, speed and temperature values can be configured in the firmware.

Bubolz, 2017-02-23:
It would also be interesting add in LCD menu options with the same routine for "Load" and "Unload" filament , independently of the filament exchange command.

Tannoo, 2017-02-24:
Filament change option doesn't show on the LCD has temp. So, pre-heat the hotend and the filament change will work, even outside of printing.

Tannoo, 2017-02-24:
>It would be nice if all length, speed and temperature values can be configured in the firmware.

It looks like they are.

```
  #define FILAMENT_CHANGE_X_POS 3             // X position of hotend
  #define FILAMENT_CHANGE_Y_POS 3             // Y position of hotend
  #define FILAMENT_CHANGE_Z_ADD 10            // Z addition of hotend (lift)
  #define FILAMENT_CHANGE_XY_FEEDRATE 100     // X and Y axes feedrate in mm/s (also used for delta printers Z axis)
  #define FILAMENT_CHANGE_Z_FEEDRATE 5        // Z axis feedrate in mm/s (not used for delta printers)
  #define FILAMENT_CHANGE_RETRACT_FEEDRATE 60 // Initial retract feedrate in mm/s
  #define FILAMENT_CHANGE_RETRACT_LENGTH 2    // Initial retract in mm
                                              // It is a short retract used immediately after print interrupt before move to filament exchange position
  #define FILAMENT_CHANGE_UNLOAD_FEEDRATE 10  // Unload filament feedrate in mm/s - filament unloading can be fast
  #define FILAMENT_CHANGE_UNLOAD_LENGTH 100   // Unload filament length from hotend in mm
                                              // Longer length for bowden printers to unload filament from whole bowden tube,
                                              // shorter length for printers without bowden to unload filament from extruder only,
                                              // 0 to disable unloading for manual unloading
  #define FILAMENT_CHANGE_LOAD_FEEDRATE 6     // Load filament feedrate in mm/s - filament loading into the bowden tube can be fast
  #define FILAMENT_CHANGE_LOAD_LENGTH 0       // Load filament length over hotend in mm
                                              // Longer length for bowden printers to fast load filament into whole bowden tube over the hotend,
                                              // Short or zero length for printers without bowden where loading is not used
  #define FILAMENT_CHANGE_EXTRUDE_FEEDRATE 3  // Extrude filament feedrate in mm/s - must be slower than load feedrate
  #define FILAMENT_CHANGE_EXTRUDE_LENGTH 50   // Extrude filament length in mm after filament is loaded over the hotend,
                                              // 0 to disable for manual extrusion
                                              // Filament can be extruded repeatedly from the filament exchange menu to fill the hotend,
                                              // or until outcoming filament color is not clear for filament color change
  #define FILAMENT_CHANGE_NOZZLE_TIMEOUT 45L  // Turn off nozzle if user doesn't change filament within this time limit in seconds
  #define FILAMENT_CHANGE_NUMBER_OF_ALERT_BEEPS  5L  // Number of alert beeps before printer goes quiet
  #define FILAMENT_CHANGE_NO_STEPPER_TIMEOUT         // Enable to have stepper motors hold position during filament change
```

Tannoo, 2017-02-24:
Except for temperature.

Bubolz, 2017-02-24:
In addition to the ideal temperature setting to remove filament from the nozzle, and subsequent reheating to load the filament, there is also the issue of slowing down (Both input and output on the filament) as it passes through the interior of the extruder and the nozzle. High speeds would only be used while the filament passing through the bowden tube.

Bubolz, 2017-02-24:
User "CoderSquirrel" has implemented something following this line of thinking about the variable speed of the filament when passing through areas like the extruder and the hotend. The implementation can be seen here:

[https://github.com/MarlinFirmware/Marlin/pull/3605/commits/9f5c7ce908a0a20be55fbfc09b87944724b6560c](url)

But my suggestion is just a bit more complicated ;-)

closed 2018-02-06 by thinkyhead