MarlinFirmware/Marlin #5897: G38.3 with Z Probe Pin

Phife54, 2017-02-23:
Im trying to use the G38.3 command with an endstop connected to X Max. It seems that the marlin firmware im using (RC8) only responds to Z_MIN pin for G38.3 commands, even if I have another pin defined for the Z probe. I am using the Z_MIN pin for homing.

Can anyone help me out and change the G38.3 code to respond to the Z_Probe pin instead of Z_Min?

Bob-the-Kuhn, 2017-02-24:
See pull request  #5678 for the fix.

Not only does it only use Z_MIN but it is only checked in the -Z direction.

Bob-the-Kuhn, 2017-02-24:
> Awesome Bob, Thanks for this. Only Question, you mentioned to me that this only works in the -Z direction, can this be changed to work in the +Z direction but still use the Z Probe Pin?
> Z Min endstop is my home switch, probing takes place when moving away from home. probe is on X_MAX and is defined in configuration.h as #define Z_MIN_PROBE_PIN X_MAX_PIN

It works in all directions on all axis.

Attached is the updated file.  

[endstops.zip](https://github.com/MarlinFirmware/Marlin/files/799948/endstops.zip)

closed 2017-07-07 by thinkyhead