MarlinFirmware/Marlin #5905: Issues with knob on LCD2004

Jolly2015, 2017-02-25:
Hello, I defined REPRAP_DISCOUNT_SMART_CONTROLLER  and use on my delta(motherboard=33,RC8 version),sometimes like I click "stop print",but it run to "pause print" and let machine pause(I am sure I didn't turn the knob and appeared with high probability and not only occuring on "stop print"), sometimes I click "auto home" and it return to main interface(appeared with high probability).I don't know where I can fix it  in marlin,anybody can help me?Thx

Tannoo, 2017-02-25:
If I understand your issue correctly:
```
// This option overrides the default number of encoder pulses needed to
// produce one step. Should be increased for high-resolution encoders.
//
//#define ENCODER_PULSES_PER_STEP 1

//
// Use this option to override the number of step signals required to
// move between next/prev menu items.
//
//#define ENCODER_STEPS_PER_MENU_ITEM 5
```
Enabled these and adjust as needed.

Jolly2015, 2017-02-25:
Wow, thx for your fast answer,it fixed. Thank you .

Tannoo, 2017-02-25:
Good!

closed 2017-02-25 by Jolly2015