MarlinFirmware/Marlin #5906: Cannot compile Current RC and RCbugfix homing_feedrate_mm_s

natan115, 2017-02-25:
hi to all,
need help plz 
Cannot compile Current RC and RCbugfix homing_feedrate_mm_s

arduino 1.8.0 from arduino.cc 

Arduino: 1.8.0 (Windows 7), Board: "Arduino Mega or Mega 2560, ATmega2560 (Mega 2560)"

WARNING: Category '' in library EEPROM is not valid. Setting to 'Uncategorized'
WARNING: Category '' in library SPI is not valid. Setting to 'Uncategorized'
WARNING: Category '' in library SoftwareSerial is not valid. Setting to 'Uncategorized'
WARNING: Category '' in library Wire is not valid. Setting to 'Uncategorized'
Warning: platform.txt from core 'Arduino AVR Boards' contains deprecated recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{build.path}/{archive_file}" "{object_file}", automatically converted to recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{archive_file_path}" "{object_file}". Consider upgrading this core.
Marlin_main.cpp:376: error: expected initializer before 'homing_feedrate_mm_s'

 float constexpr homing_feedrate_mm_s[] = {

                 ^

sketch\Marlin_main.cpp: In function 'float get_homing_bump_feedrate(AxisEnum)':

Marlin_main.cpp:1528: error: expected initializer before 'homing_bump_divisor'

   int constexpr homing_bump_divisor[] = HOMING_BUMP_DIVISOR;

                 ^

Marlin_main.cpp:1529: error: 'homing_bump_divisor' was not declared in this scope

   int hbd = homing_bump_divisor[axis];

             ^

Marlin_main.cpp:1535: error: 'homing_feedrate_mm_s' was not declared in this scope

   return homing_feedrate_mm_s[axis] / hbd;

          ^

sketch\Marlin_main.cpp: In function 'void do_blocking_move_to(const float&, const float&, const float&, const float&)':

Marlin_main.cpp:1669: error: 'homing_feedrate_mm_s' was not declared in this scope

       feedrate_mm_s = fr_mm_s ? fr_mm_s : homing_feedrate_mm_s[Z_AXIS];

                                           ^

Marlin_main.cpp:1681: error: 'homing_feedrate_mm_s' was not declared in this scope

       feedrate_mm_s = fr_mm_s ? fr_mm_s : homing_feedrate_mm_s[Z_AXIS];

                                           ^

sketch\Marlin_main.cpp: In function 'void do_homing_move(AxisEnum, float, float)':

Marlin_main.cpp:2604: error: 'homing_feedrate_mm_s' was not declared in this scope

     planner.buffer_line(current_position[X_AXIS], current_position[Y_AXIS], current_position[Z_AXIS], current_position[E_AXIS], fr_mm_s ? fr_mm_s : homing_feedrate_mm_s[axis], active_extruder);

                                                                                                                                                     ^

exit status 1
expected initializer before 'homing_feedrate_mm_s'

Tannoo, 2017-02-25:
I'd uninstall the Arduino IDE and reinstall the latest version.. 1.8.1

natan115, 2017-02-25:
did it, didn't help.

Bob-the-Kuhn, 2017-02-25:
Try the following:
1. download a new copy of RCBugFix
2. unzip it
3. compile it as is (no changes to anything)

If it compiles then do your changes a few at a time.  Make sure you use the **Configuration.h** and **Configuration_adv.h** files that came with the new download. 

If it doesn't compile then try 1.6.12 or 1.6.13.

If it still doesn't compile then remove Arduino completely and re-install it.  To remove Arduino completely do the following:
1. Go to program manager and uninstall if Arduino shows up there.
2. Delete the **C:\Program Files (x86)\Arduino** directory and all similarly named directories.
3. Delete the **Arduino15** directory.  On my Win10 machine it's at  *C:\Users\Bob\AppData\Local\Arduino15**.

How have you been installing Arduino - **Windows Installer** or **Windows ZIP** ?   This time use the other method.

If you end up using the ZIP method then try this:
1. Unzip the package to a convenient location
2. Drill down until you find **arduino.exe** and then double click it.

If it still doesn't compile then I'm out of ideas.

Tannoo, 2017-02-26:
Also try to compile any example like `blink`.

Tannoo, 2017-02-26:
These:
>WARNING: Category '' in library EEPROM is not valid. Setting to 'Uncategorized'
WARNING: Category '' in library SPI is not valid. Setting to 'Uncategorized'
WARNING: Category '' in library SoftwareSerial is not valid. Setting to 'Uncategorized'
WARNING: Category '' in library Wire is not valid. Setting to 'Uncategorized'

Are signs of a bad IDE install.

Tannoo, 2017-02-26:
https://forum.arduino.cc/index.php?topic=347693.0

>by adafruit_support_mike on Sun Jun 12, 2016 10:35 pm 
Sorry.. multi-issue threads like this get confusing.

Digging specifically into the "Setting to 'Uncategorized'" error, that seems to be related to the 'library.properties' file used by recent versions of the IDE.

It looks like you have an older version of the board support files. Open the Boards Manager, update the Arduino AVR Boards package, and see if that helps.

sergiomajluf, 2017-03-12:
I can replicate this error, brand new installation of Arduino.
Blink won't compile. 

the problem is with within Marlin AVR Board. If you change to the default boards section, both blink and Marlin compile and uploads fine

Bob-the-Kuhn, 2017-03-12:
I don't understand.  

What's a "Marlin AVR Board"?  I see an Arduino AVR section in the Arduino IDE's board selection menu. 

Can you explain what you mean by "If you change to the default boards section"?  Is this in the Arduino IDE?

What board did you have selected before and what board did you finally end up with?

sergiomajluf, 2017-03-12:
When I downloaded Marlin, there was a folder for the firmware itself, and also some libraries and hardware folders for the Arduino environment, specific to the Arduino version you are using, that you copy into ~/Arduino/Libraries and ~/Arduino/hardware

Now that I think about it, maybe those are legacy files from Marlin < 1.0???

Anyway, after copying those files, within Arduino Tool menu I get this:
![Arduino Tool Menu, Then boards]
(https://dl.dropboxusercontent.com/s/1q8fetlcpuzntrv/Screenshot%202017-03-12%2000.07.18.png)

As you can see, there is a section below. I'm using the default Arduino Mega and things are apparently working.

Bob-the-Kuhn, 2017-03-12:
"also some libraries and hardware folders for the Arduino environment, specific to the Arduino version you are using"

Yes, I see that in Marlin 1.0.2.  All those are for old versions of the Arduino IDE.  They are not needed for 1.6.x & 1.8.x Arduino.

matt448, 2017-04-26:
omg I fought with this for two days. I did the same thing... copied the IDE add-ons into 1.8.1 and got all the same compile errors above. Once I started with clean install of 1.8.1 and was able to compile 1.1.0-RC8. Though now the X,Y and Z steppers seem to be mixed up. Probably a configuration issue I need to sort out.

closed 2017-04-26 by thinkyhead