MarlinFirmware/Marlin #5907: Cannot compile because this error.

Rhonal89, 2017-02-25:
/Users/Logan-Mac/Documents/Arduino/libraries/U8glib/utility/u8g_rot.c:48:1: warning: initialization from incompatible pointer type u8g_dev_t u8g_dev_rot = { u8g_dev_rot_dummy_fn, NULL, NULL }; ^/Users/Logan-Mac/Documents/Arduino/libraries/U8glib/utility/u8g_rot.c:48:1: warning: (near initialization for 'u8g_dev_rot.dev_fn')

Its on Mac latest u8glib. RC8

Blue-Marlin, 2017-02-25:
This error is not an error but a warning. It does not prevent from compiling. Warnings in u8glib are completely normal. If zou feel disturbed bz them / repot them there.
If zour Marlin does not compile look for the first line wit an "error".

thinkyhead, 2017-02-26:
See if you can determine the actual error preventing the compile. If there is none, you can close this issue.

Rhonal89, 2017-02-26:
Thanks I was trying to help someone over the internet with marlin. It was compiling but his bltouch is not working. I think is the way he has it wired.

closed 2017-02-26 by Rhonal89