MarlinFirmware/Marlin #5909: can i g29 without first g28?

lacosa83, 2017-02-26:
Hello, i don't have endstops at all on my router but i should like do g29. Can I? Regards :)

Tannoo, 2017-02-26:
Yes. It will initiate a G28 on it's own, if it hasn't been homed.

lacosa83, 2017-02-26:
thanks,  but i will never mount endstops on the machine :)
My question is, there is a method for bypassing the g28 launching g29? Usually i put on 0ish in manual without any problem...

Tannoo, 2017-02-26:
I don't think so. The machine needs to know where it is. Otherwise, it will run past the physical limits and mess things up.

lacosa83, 2017-02-26:
no (obv), but i see my similar problem solved in an other post ( https://github.com/MarlinFirmware/Marlin/issues/4831 ), but i don't understand where add the cmd line LOOP_XYZ(i) axis_known_position[i] = axis_homed[i] = true; linked by thinkyhead :)

Tannoo, 2017-02-26:
AHH..
Then, this is where the manual positions come into play. Would it help to set these in `configuration.h`?
```
// Manually set the home position. Leave these undefined for automatic settings.
// For DELTA this is the top-center of the Cartesian print volume.
//#define MANUAL_X_HOME_POS 0
//#define MANUAL_Y_HOME_POS 0
//#define MANUAL_Z_HOME_POS 0
```

Tannoo, 2017-02-26:
The manual positions may or may not do much for you. I don't really know.

lacosa83, 2017-02-26:
i only find void setup ()

[image]

Tannoo, 2017-02-26:
I would put it at the end of that function just to make sure all else is loaded first.

lacosa83, 2017-02-26:
Okay thanks :)

Tannoo, 2017-02-26:
I would think that the manual position definitions would eliminate the need for a G92 before printing. (In your case)

lacosa83, 2017-02-26:
bump doesn't work

Tannoo, 2017-02-26:
Sorry, but I am at a loss on that then.

closed 2017-02-26 by lacosa83

thinkyhead, 2017-02-28:
I use this for one of my client projects:
```cpp
  /**
   * G8: Simulate homing
   */
  inline void gcode_G8() {
    current_position[X_AXIS] = LOGICAL_Z_POSITION(X_HOME_POS);
    current_position[Y_AXIS] = LOGICAL_Z_POSITION(Y_HOME_POS);
    current_position[Z_AXIS] = LOGICAL_Z_POSITION(Z_HOME_POS);
    LOOP_XYZ(i) axis_homed[i] = axis_known_position[i] = true;
    SYNC_PLAN_POSITION_KINEMATIC();
    SERIAL_ECHO_START;
    SERIAL_ECHOLNPGM("At Home " STRINGIFY(X_HOME_POS) " " STRINGIFY(Y_HOME_POS) " " STRINGIFY(Z_HOME_POS));
    report_current_position();
  }
```

lacosa83, 2017-02-28:
Thanks, i will try it tomorrow