MarlinFirmware/Marlin #5912: DUAL_X_CARRIAGE and MBL adjusted between tools?

Pipshag, 2017-02-26:
I've just built a Dual X Carriage and have a question about what's intended to happen when using MBL with a Dual X system.

Using T0 I've done an MBL and saved to EEPROM with M500.
After doing a G28, MBL activates when moving T0 over the bed (G1 X0 Y0) and puts the bed at the correct distance from the nozzle.
If I instead try G28 followed by T1 and move over the bed (G1 X300 Y0), the bed never gets adjusted to the values set in MBL.
It doesn't activate either if I move in T0 over the bed, move it out and then try to move T1 in over the bed again.

I have set a Z-offset for T1 since it's at a different height (M218 T1 Z-0.2). The bed is not completely leveled and T1-nozzle hits the side of the bed.

Running on commit f9e4717f from RCBugFix.

My question is - is MBL intended to work for only one tool? Or is this a bug or feature that's not there not? Since there is a Z-offset set, the data would be there to adjust for the uneven bed for T1 as well.

oysteinkrog, 2017-02-26:
I have this problem too, it's been like this for a long time, you can find earlier issues on it, but they were closed due to inactivity (my fault, I was rebuilding my printer).

epatel, 2017-02-26:
Hi guys. Yes, I am sorry about that. But when I added MBL https://github.com/MarlinFirmware/Marlin/pull/1619 I didn't have a probe or multi-nozzles. And getting that first version in was worse than I expected. But! @Roxy-3D is doing a great job of merging MBL and ABL in the UBL branch I think. Maybe that works better for multi-nozzles? I see that @Pipshag is probably using a BigBox, I have had a little contact with @Alex9779 that might know more about doing this on that machine? If I had a similar hardware setup I could probably help out better. @Pipshag Du bor inte nära Stockholm?

Pipshag, 2017-02-26:
#1619 is a great contribution and have worked very reliably in the past. Thank you for developing it!

It's true that I use a BigBox, now using the probe only as an endstop. Alex9779 has switched from Marlin and is now running other hardware. It seems from discussions on the E3D forums that the currently accepted workaround is to simply make sure your bed is level and not at an angle. Somehow the MBL is still going to work.

> MBL operates on a single printhead, T0. Calcaulating the offsets corrects any difference between the two. Technically there will be a slight miss-alignment between the actual bed matrix and the position of T1 as the M218 command has not be used to offset the head [Clarification: in X and Y], but it will be so small as to not matter.
> 
> The difference in height between the two nozzles should be no more than 0.5mm. If you are getting collisions I recommend checking the bed is actually level and not off at an angle. 

@epatel Unfortunately, I live near Malmö. I'll check out where UBL stands but do believe that it's only a short-term solution and that we need to focus on the main branch. Don't get me wrong, it looks like a great development, but it's going to drive testers away until if/when it's merged.

closed 2017-10-31 by thinkyhead