MarlinFirmware/Marlin #5913: Modifying config files

ruggb, 2017-02-26:
@thinkyhead - When I update to a new build I do an optical compare between the old file and new file. Having old eyes, this requires multiple scans. But I am sure I am not alone. To add to the confusion, commented out #defines look the same as text especially when some text uses /*,,,,*/ and other uses //.

When things change in the config files a new stumbling block is added. I have gone thru the h and adv.h files and made some mods but I see another blocks. Sections are labeled, then the label appears multiple times in both files. Other sections do not seem to be relevant to what is in them. It now becomes more difficult to modify something and justify all the variables associated with that mod.

I am one for being organized and simply clear explanations and some of the comments still are not understandable to me - BUT I must say it has gotten significantly better in the years since I became involved with 3D printing - thanks.

I would like to propose some changes to the information, and structure of these files but I am not sure if there is some reason for multiple "@section temperature" and the difference between that and banners with "==================temperature===========" and other items.

I have already made some significant edits of the two config files but would like to get some feedback on this b4 continuing. Plus I also do not know how to present this. I suppose I would make a branch, but I have newer done that. I am a hardware engineer but I have done some coding and understand programmers. I just don't see the whole picture of Marlin. That's a thought -- is there a flow chart for Marlin???

To see if there is any desire for this, please comment on these ideas:

1. All comment lines will start with /* and end at the end of a line with */
 - this is to make commented out code lines more obvious.
 - comments at the end of a code line will remain with //.

2. What is a banner mark vs section label used for?
 - I am obviously unclear about this.

3. Make comments clear to the user
 - I have to go back and look but there is at least one that is a total mystery to me.

4. There are "user configurable" items in files other than the config files that should be placed in the config files or at least listed in the config files as something a user might want to look at - like the language files (reference) or the utltralcd files - I'm thinking of the LCD timeout here (move to config).

5. Place config items for functions together - like temp items, or safety items, or mechanical items, or LCD items, or even babystepping.
 - setting temp items or configuring for thermal runaway are really two different sections.
 - having multiple sections for the same topic is also not good even if one has required settings and one has optional settings. That can be designated under the same section heading.

Just trying to help
thanks

Dreamloom, 2017-02-26:
I use Notepad++ with its Compare plugin, to merge customizations and configuration settings with new Release Candidates and Bugfixes.  It brings up the new file and your customized version side by side highlighting the differences, so you can copy and paste your code into the new Release.

Download Notepad++, install. Click plugins on the menu bar, then plugin manager and select Compare.

To use, open (for example) config.h in the New Release, then open your current config.h - they will open in different tabs. With Compare installed and YOUR config.h file's tab selected., near the right end of the menu bar, click "Set as first to Compare", then select "Compare" and you'll get the two files opened with the differences highlighted.

ruggb, 2017-02-26:
I am using Ultraedit. Same function. I did just notice in going thru this and trying to make it simpler - which I think I am succeeding at - that the definitions for the endstops changed from a const bool to a #define between RC7 and RC8 and I never noticed that. Could be why it doesn't work for me, but I haven't tried it yet.
I have separated out things from the config files into individual files and used includes to add them if needed. I don't use the drivers defined in c_adv.h or bed leveling, servos, etc. The lcd section is just too big to not add confusion and it works nicely in a separate file.
Hopefully, someone who can do something about the current confusion will agree and we can make Marlin better - easier to use and easier to modify and fix.

Roxy-3D, 2017-02-27:
> Hopefully, someone who can do something about the current confusion will agree and we can make Marlin better - easier to use and easier to modify and fix.

That is the intent with all of the changes to configuration.h   The name changes and re-organization are aimed at making things more intuitive and easy to setup.   Unfortunately...  People that have working systems will have to do a little bit editing to map over their working configuration to what is going to be a 'Stable Branch' of Marlin.

ruggb, 2017-02-27:
The changes that have occurred are definitely good. I would just like to propose a few things that may/should make things even clearer and easier - of course IMHO. 
In going thru this I have found a few things I don't understand - maybe you can clarify them.
@@@@@@@@@@@@@
configuration.h
102 /* Enable if your E steppers or extruder gear ratios are not identical. */
103 //#define DISTINCT_E_FACTORS
Where is the gear ratios set?
@@@@@@@@@@@
/* A dual extruder that uses a single stepper motor
 * Don't forget to set SSDE_SERVO_ANGLES and HOTEND_OFFSET_X/Y/Z
//#define SWITCHING_EXTRUDER
Where is SSDE_SERVO_ANGLES??
@@@@@@@@@@@@@@@@
What is this feedrate for? - from c_adv.h
#define DEFAULT_MINIMUMFEEDRATE       0.0     // minimum feedrate
#define DEFAULT_MINTRAVELFEEDRATE     0.0
@@@@@@@@@@@@@@@@@@@@
What does this mean? Is default (undefined) to show warning or not?
/* Warn on display about possibly reduced accuracy */
//#define DISABLE_REDUCED_ACCURACY_WARNING
@@@@@@@@@@@@
There are only 2 hotend preheat sections. Doesn't marlin support 3 or 4 hotends now?
@@@@@@@@@@@@@@@@@
I can't separate out the Filament sensor code into a separate file because DEFAULT_NOMINAL_FILAMENT_DIA 3.00 is used elsewhere even though the sensor is not used. Does that need to be defined all the time?
@@@@@@@@@@@@@@
MOVEMENT
 * If fewer factors are given, the last will apply to the rest.
what exactly does this mean?
 * Note that if EEPROM is enabled, saved values will override these.
Are you saying that EEPROM settings are not changed with a new upload of firmware?
@@@@@@@@@@@
The defines for the endstops changed, I assume between RC7 and RC8 and I totally missed that.
From const bool to just #define. Maybe that is why RC8 doesn't work for me - and maybe others....
I have not gone back to verify yet.
@@@@@@@@@@@@@

What I am basically doing is trying to get items that relate in the same place, set groups of code for specialty items like bed leveling and special motor support into separate files and link them in with includes, use /* & */ for comments - this makes //#define ... stand out a lot better. This makes the config files a whole less daunting and confusing.
I found it very confusing to have things like movement items in two places let alone two files.
I found other items like LCD timeout that probably would be better in the config files than in the ultralcd file that most users would never look into.
I have a bat file that deletes all the files I do not use and that make my Marlin folder look a lot cleaner.
You may hate it all but it is helping me see the whole thing better - and I suspect it will help others also - and I am sure it will help to better maintain it.
What do you think? How can I get this to someone to look at?
After all this my last compile has an error that is confusing me. I have 15 #if statements and 15 #endif but the compiler says one #if is unterminated. Optically, they all match up.
thx all.

thinkyhead, 2017-03-01:
> only 2 hotend preheat sections

These are for different material profiles, not for each extruder.

> Where is the gear ratios set?

`DEFAULT_AXIS_STEPS_PER_UNIT` embodies gear ratios, Z rod pitch, etc.

> If fewer factors are given, the last will apply to the rest.
what exactly does this mean?

For a setting applying to XYZE axes such as `{ xval, yval, zval, e0val, e1val }` if you had 4 extruders, the `e1val` will apply to E2 and E3 as well.

> defines for the endstops changed, I assume between RC7 and RC8 and I totally missed that

`RCBugFix` is supposed to throw an error to inform you of this change (assuming there are no other compile errors).

thinkyhead, 2017-03-01:
> I found it very confusing

Documentation is coming together at http://marlinfw.org/

ruggb, 2017-03-13:
@thinkyhead - Documentation is part of the code. That might be OK if pieces were not spread all over the files. If it is separated from the code as it stands it becomes impossible to properly document. There are stepper, extruder, temperature, etc., related items in different parts of the file and in different files. There might be some logic for that in developer minds but it is not user friendly. It is also not user friendly - or developer friendly - to have OPTIONAL functions like special stepper drivers, nozzle cleaning, bed leveling, and LCD selection making a config file so large it becomes overwhelming to both user and developer.
I have put together a plan and tested it and it works perfectly. It isn't complete, but I am not understanding why you don't even want to consider it. There are no code changes, just reorganization. I had to go back to YOUR way on the last one and it took me 4 times longer to configure - and I have been doing this a lot longer than most users.
If the config files are organized and optional functions are set as include files, you could actually create a separate document - that would make the config files a lot easier to navigate.
And it will probably never be perfect, but IMHO what you have is overwhelming for most users. I am betting over 60% of your issues are related to improper configuration. That is a big waste of time for you, when you could eliminate most of it.

lrpirlet, 2017-03-14:
hi @ruggb 

Why don’t you PROVE to the community that your concept is better? Unless I completely missed it, I could not find any branch that expose such easy configuration files...
If your branch with that new configuration structure drains a lot of interest, I am sure that the MarlinFirmware community will switch to it...

You know, this is OPEN software with many UNRELATED developers... The possibility to plan is nearly nill... BUT it is extremely agile and MANY new features do get included that have proven fantastic... example LIN_ADVANCE, MESH levelling, UBL, M600... not to mention optimization (more code but less space in limited ROM and more performance for that poor 8 bits processor)... and we should not forget the documentation that is growing daily while the configuration (sorry, I have a totally diverging experience than you) is coherent enough since a long long time so that I only need to COMPARE from a previous version to the current to find what may be copied, what should be transposed what need a closer look...

Conclusion: @thinkyhead please would mind to consider my request? 
Do not change how those config files are structured unless pushed by a huge number of users...

emartinez167, 2017-03-14:
Let me jump in here. Let's not forget one key thing: we are trying to release a STABLE release. I was going through my Google+ (yes, some of us use it) and I read about a brand new 3D printer. Went to their website and clicked on the firmware link which took me to a Github Page with a very old version of Marlin. I wrote: "Why are these guys still using an old release of Marlin?" and one user replied: "Because Marlin has not been able to release a stable version in over a year (or two?)". - Head shook -

Is it still our intent to release a stable version, or how many more RCs are we going to have? If we want Marlin to stay as one of the main firmwares out there, we should really try to get there soon. Trying to do all this reshuffling right now is not helping, plus we have the knock on effect on the dozens (or hundreds?) of repositories that forked out from this one... Just my two cents.

Regards,

Ernesto.

lrpirlet, 2017-03-15:
Hi @emartinez167 emartinez167

**STABLE** version is something of the past... Well, sort of... When there was ONE heavy rigid process driving the project with foreseen steps and milestones..

Agile development is today **THE preferred** way, even by huge companies... have a look here https://en.wikipedia.org/wiki/Agile_software_development... 
Please note how well this can be applied to OPEN software with so many developers interested with a particular aspect rather than interested to produce a STABLE version (with features designed a long time ago that may be obsolete by the time the software is released)...

The problem is to make sure that users understand that we have a GOOD WORKING FIRMWARE (with maybe a few bugs)... People like @thinkyhead do make a fantastic job by restricting modifications that are NOT tested enough, while authorizing repair patches... 

Think: current RCBugFix is ALWAYS requested when someone reports a bug...

emartinez167, 2017-03-15:
Irpirlet,

I can understand that, the problem is the perception users might have of it. Having said that, it is their loss if they choose not to use the latest revision. Perhaps we could update the documentation/GitHub main page with that statement so that users feel confident about using it? I just think it would dispel these concerns from potential users.

Regards,

Ernesto

ruggb, 2017-03-15:
@thinkyhead Documenting Marlin as it stands will be very difficult. It will end up with many things the user would like to be aware of undocumented. Or worse, with documentation that is harder to read than the code. Things like the BOOTSCREEN_TIMEOUT issue is a perfect example. You don't want a user to have to wade thru lots of files to configure items, right? - the only exception would be in defining OPTIONAL functions like special steppers, bed leveling, etc. There are other things, like the welcome message in languages that I change, or the delay to return to status screen. I would say most users don't even know they can change those things. The fact that none of the developers knew that babystepping_Z did, in fact, work on coreXY machines if you disabled the XY part of it, says something.

@emartinez167 Consider the fact that a long time ago Marlin was probably very clean code with few functions and few developers. Based on things I am finding in the code (that I don't believe are operational bugs) it is obvious lots of people are working on this because it isn't consistent - I assume because the rules were lost in the shuffle to add functionality. It works! as long as the user doesn't try to configure it too much - THAT makes it unstable. I am a BIG proponent of organization, but don't look at my desk right now.
If H files contain declarations and CPP files contain definitions, why do H files contain includes?
I have a good reason for putting includes in the config files because they make the config file more manageable for a user. But having includes in a CPP file, then repeating them in an H file that the CPP file includes is sloppy and unnecessary as far as I can see - but you may be able to tell me why it isn't. Look at the nozzle files for an example.

It is also bad for user friendliness to include user definable items in files you don't want the user to have to look at.

It is also bad for user friendliness to format a file the user must deal with visually in a manner that causes difficulty navigating, comprehending and difficulty visually finding items needed.

Fix those things and Marlin will become a LOT MORE STABLE to users.

There may be functions that don't quite work yet - or may never work, but declared as such, they don't make Marlin unstable - just a work in progress that users know how to deal with. And if you don't include them in a released product, they don't have to deal with it.

But if the code isn't clean such that one change causes an issue - like the BOOTSCREEN_TIMEOUT thing - then it is unstable. That is an organization issue, not an operating bug.

An operating bug WAS the fact that Marlin could not handle fast printing - and there was and maybe still is - no published spec on what it can handle - Though I think someone stated it in one of these issues.
It is now able to handle at least 60MM/sec on my printer. It really should be published. Not doing so creates an instability to the user.

closed 2017-10-31 by thinkyhead