MarlinFirmware/Marlin #5914: Delta tower angle

asterycs, 2017-02-26:
Hello! I just got started with printing, and the calibration of my delta is still lacking. I believe, after some googling, that this is due to the tower angles not being perfect. I read some old forum post that said that some development branch of Marlin had this calibration feature implemented with code M665. The posts were years old, and I can't seem to find this feature in any of the current branches. Has this ever been implemented?

james94jeans2, 2017-02-26:
M666 is available in the RCBugFix Branch as far as I can tell, but I have no clue on how to use it as I don't have a delta.

asterycs, 2017-02-26:
Yes it is available, but this particular feature seems to be missing. From configuration_store.cpp:
` * DELTA (if deltabot):                            36 bytes
 *  348  M666 XYZ  endstop_adj                     (float x3)
 *  360  M665 R    delta_radius                    (float)
 *  364  M665 L    delta_diagonal_rod              (float)
 *  368  M665 S    delta_segments_per_second       (float)
 *  372  M665 A    delta_diagonal_rod_trim_tower_1 (float)
 *  376  M665 B    delta_diagonal_rod_trim_tower_2 (float)
 *  380  M665 C    delta_diagonal_rod_trim_tower_3 (float)`
What I'm looking for is a way to change the angle between the towers when the printer is seen from above. This is not the same as diagonal rod trim. I'll give a try to implement it if no one has heard about it.

thinkyhead, 2017-03-01:
So, you've got a setup where the towers are not equally-spaced 120° apart? I'm not sure we have the hooks to do that kinematic variant yet, either.

asterycs, 2017-03-01:
I implemented it in my fork allready. I'll do some testing and verify that it works as it should.

thinkyhead, 2017-03-08:
Merged!

closed 2017-03-08 by thinkyhead