MarlinFirmware/Marlin #5915: g29 without endstop???

lacosa83, 2017-02-26:
As title...please help :)

james94jeans2, 2017-02-26:
As far as I know you would need to change some code to do this. I can maybe give you a hint on what to do if I find the time later.

lacosa83, 2017-02-26:
thanks dude :)

Bob-the-Kuhn, 2017-02-26:
Try the solution in issue #5909.

lacosa83, 2017-02-26:
it was the mine and didn't work :D

james94jeans2, 2017-02-26:
I can not test this as my printer is currently printing, but there might be a very simple:

In line 3785 of file Marlin_main.cpp:

remove "G28\n"

I am quite new to github and don't know how I could show it in a stylish way, but the describtion should be enough.

change from: 
3785 enqueue_and_echo_commands_P(PSTR("G28\nG29 S2"));
to:
enqueue_and_echo_commands_P(PSTR("G29 S2"));

Please report back on any issues.

lacosa83, 2017-02-26:
That's work, but not from lcd. I must do from repetier M420 S1 then g92 x0y0z0 and then i start it :)

james94jeans2, 2017-02-26:
You can remove the homing from lcd by editing ultralcd.cpp

as suggested in #5909 set the homed values to true:
1285 axis_homed[X_AXIS] = axis_homed[Y_AXIS] = axis_homed[Z_AXIS] = true;
and remove the line that calls g28:
enqueue_and_echo_commands_P(PSTR("G28"));

However it will still be doing a homing at the end of leveling

lacosa83, 2017-02-26:
umh, can i save in eeprom the result first of homing?

james94jeans2, 2017-02-26:
I don't think thats possible, but I am not sure where it starts the homing after the leveling, maybe try the above mentioned edit and tell me if it starts homing after the leveling

lacosa83, 2017-02-26:
all work well but at the end of the cycle will do g28.. :(

lacosa83, 2017-02-26:
seem work: first i launch from lcd the mesh, the, from repetier host i can see the grid using g29 s0

james94jeans2, 2017-02-26:
You could try deleting line 
1196             enqueue_and_echo_commands_P(PSTR("G28"));

in ultralcd to stop the g28 from happening, but I don't know if the leveling will work as intended when this is removed.

lacosa83, 2017-03-01:
now i have the grid on eeprom but don't seem work at all: i tried print something but nothing.....any idea?

Bob-the-Kuhn, 2017-03-01:
Looks like you're running RC7 and mesh leveling.

I don't think you need to do the M501.  That's done automatically when the controller powers up or goes through reset.

Please give us the steps you've taken to verify that the mesh leveling is or is not working.

We'd also like to see the steps you take to get ready for a print and the start of the g-code file through the first print command.

lacosa83, 2017-03-01:
;Project GRIGLIA3
;Created by Estlcam version 10 build 10,012
;Machining time about 02:02:00 hours

G90
M03 S24000
G00 Z5.0000

;No. 1: Engraving 4
G00 X0.0000 Y0.0000
G00 Z0.5000
G01 Z-1.0000 F1200 S24000
G01 X5.0000
G01 Y600.0000
G01 X10.0000
G01 Y0.0000
G01 X15.0000
G01 Y600.0000
G01 X20.0000
G01 Y0.0000
G01 X25.0000
G01 Y600.0000
G01 X30.0000
G01 Y0.0000
G01 X35.0000

lacosa83, 2017-03-01:
Could help set

Bob-the-Kuhn, 2017-03-01:
Please do the following:
- M420 V - should print out the leveling matrix, if not then do M420 S1
- move your head to the (0,0,0) location
- G92 X0 Y0 Z0 ; sets the current location as (0,0,0)
- Go to 5 locations within the mesh area and measure how far off the bed the head is.  That'll tell you if mesh leveling is doing a decent job.

When you set (0,0,0), is the nozzle just touching the bed or is it some height above the bed?

Are you doing 3D printing or some type of CNC work?  The **M03 S24000** command is only used in CNC work.

lacosa83, 2017-03-01:
- 00:13:38.458 : N17 M112*23
00:13:39.733 : Printer reset detected - initalizing
00:13:39.733 : start
00:13:39.733 : echo:Marlin 1.1.0-RC7
00:13:39.738 : echo: Last Updated: 2016-07-31 12:00 | Author: (none, default config)
00:13:39.738 : Compiled: Mar  1 2017
00:13:39.742 : echo: Free Memory: 3261  PlannerBufferBytes: 1232
00:13:39.746 : echo:V24 stored settings retrieved (398 bytes)
00:13:39.746 : echo:Steps per unit:
00:13:39.746 : echo:  M92 X160.00 Y160.00 Z5120.00 E160.00
00:13:39.750 : echo:Maximum feedrates (mm/s):
00:13:39.750 : echo:  M203 X190.00 Y190.00 Z8.50 E25.00
00:13:39.754 : echo:Maximum Acceleration (mm/s2):
00:13:39.754 : echo:  M201 X500 Y500 Z35 E10000
00:13:39.754 : echo:Accelerations: P=printing, R=retract and T=travel
00:13:39.758 : echo:  M204 P400.00 R3000.00 T500.00
00:13:39.767 : echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
00:13:39.767 : echo:  M205 S0.00 T0.00 B20000 X4.00 Z0.40 E5.00
00:13:39.767 : echo:Home offset (mm)
00:13:39.771 : echo:  M206 X0.00 Y0.00 Z0.00
00:13:39.771 : Mesh bed leveling:
00:13:39.771 : echo:  M420 S1 X3 Y3
00:13:39.772 : echo:  G29 S3 X1 Y1 Z3.95000
00:13:39.776 : echo:  G29 S3 X2 Y1 Z3.92500
00:13:39.776 : echo:  G29 S3 X3 Y1 Z3.92500
00:13:39.779 : echo:  G29 S3 X1 Y2 Z3.60000
00:13:39.779 : echo:  G29 S3 X2 Y2 Z3.92500
00:13:39.779 : echo:  G29 S3 X3 Y2 Z3.87500
00:13:39.779 : echo:  G29 S3 X1 Y3 Z3.20000
00:13:39.783 : echo:  G29 S3 X2 Y3 Z3.22500
00:13:39.783 : echo:  G29 S3 X3 Y3 Z3.40000
00:13:39.783 : echo:Material heatup parameters:
00:13:39.787 : echo:  M145 S0 H190 B50 F0
00:13:39.787 : echo:  M145 S1 H240 B110 F0
00:13:39.787 : echo:PID settings:
00:13:39.791 : echo:  M301 P17.98 I0.98 D83.62
00:13:39.791 : echo:Filament settings: Disabled
00:13:39.791 : echo:  M200 D3.00
00:13:39.791 : echo:  M200 D0
00:13:39.851 : N1 M110*34
00:13:39.851 : N2 M115*36
00:13:39.851 : N4 M114*35
00:13:39.859 : N5 M111 S6*98
00:13:39.860 : N6 T0*60
00:13:39.860 : N7 M20*22
00:13:39.860 : N8 M80*19
00:13:45.374 : ok
00:13:45.452 : FIRMWARE_NAME:Marlin 1.1.0-RC7 (Github) SOURCE_CODE_URL:https://github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:MPCNC EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff
00:13:45.452 : ok
00:13:45.452 : N11 M220 S100*81
00:13:45.453 : N12 M221 S100*83
00:13:45.456 : X:0.00 Y:0.00 Z:0.00 E:0.00 Count X: 0 Y:0 Z:0
00:13:45.456 : ok
00:13:45.456 : echo:DEBUG:INFO,ERRORS
00:13:45.456 : N13 M111 S6*85
00:13:45.456 : ok
00:13:45.457 : N14 T0*15
00:13:45.460 : echo:Active Extruder: 0
00:13:45.460 : ok
00:13:45.460 : Begin file list
00:13:45.460 : End file list
00:13:45.460 : ok
00:13:45.460 : ok
00:13:45.464 : ok
00:13:45.464 : ok
00:13:45.464 : echo:DEBUG:INFO,ERRORS
00:13:45.464 : ok
00:13:45.464 : echo:Active Extruder: 0
00:13:45.464 : ok
00:14:18.236 : N26 M420 V*103
00:14:18.241 : ok
00:14:30.188 : N31 M420 S1*85
00:14:30.194 : ok

Bob-the-Kuhn, 2017-03-02:
What software are you running?  When I look at the RC7 code I see a slightly different layout for the mesh report than what you posted.

Does the g-code command **M03 S24000** do something?  That command isn't yet supported in the main Marlin code base.  If yes then please include the URL where you got your code from.

I went over to the Vicious Circle site and downloaded the MPCNC software.  It doesn't match the mesh report.

----

To me the results of your **G29 V4** command indicate that the mesh leveling is on.  Have you done any measurements to prove/disprove that mesh leveling is working?

lacosa83, 2017-03-02:
i download the code from vicious1.com: i onlly modify what i said in the post and movment data (i have t20pulley and not 16 tooth). The mesh isn't working:i tried to manual g0 Xx Yx Zx the same point i misured with the mesh and i have the same z
IMO it's because i can't do g28....

Bob-the-Kuhn, 2017-03-02:
We really need to get you using the latest software.  Please download [RCBugFix](https://github.com/MarlinFirmware/Marlin/tree/RCBugFix).

Don't overwrite your current software.  You'll need some files from it.

You'll need to port your code modifications over, modify the Configuration.h & Configuration_adv.h files and maybe copy another file over.

If you zip Configuration.h and Configuration_adv.h and attach then I'll update them.

lacosa83, 2017-03-02:
http://www.filedropper.com/rc7mpcnclcd9916

Bob-the-Kuhn, 2017-03-02:
Attached are the updated config files.

You'll need to copy your **_Bootscreen.h** file over to the new directory.

You may also need to copy **pins_RAMPS.h** over.  Sometimes there's custom settings in there.

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/812607/Configuration.zip)

lacosa83, 2017-03-02:
wow bob,  I don't know how to thank you for being so quick in helping me.

lacosa83, 2017-03-02:
can i slow down the movment speed when probing?

Bob-the-Kuhn, 2017-03-02:
You're right - that's moving fast.

There's settings for the speeds for the various operations.  The trick is finding them. 

Feel free to customize these for your machine.  That's the purpose of Configuration.h and Configuration_adv.h.

----

You can limit the max speed for the system by changing this setting:
`#define DEFAULT_MAX_FEEDRATE          {190, 190, 8.5, 25}    // (mm/sec) MPCNC`

These are the settings for the homing and probing speeds.
```
// Homing speeds (mm/m)
#define HOMING_FEEDRATE_XY (50*60)
#define HOMING_FEEDRATE_Z  (4*60)
// X and Y axis travel speed (mm/m) between probes
#define XY_PROBE_SPEED 8000
// Speed for the first approach when double-probing (with PROBE_DOUBLE_TOUCH)
#define Z_PROBE_SPEED_FAST HOMING_FEEDRATE_Z
// Speed for the "accurate" probe of each point
#define Z_PROBE_SPEED_SLOW (Z_PROBE_SPEED_FAST / 2)
```

lacosa83, 2017-03-02:
Now the probe work with an error of 10 micron, thanks alot dude :)

Bob-the-Kuhn, 2017-03-02:
+ - 0.1mm is OK for CNC work.

Is that figure from the M48 command?    You'll need to enable **Z_MIN_PROBE_REPEATABILITY_TEST** in order to use it.

At 5120 steps per mm on your Z axis I'd expect better accuracy. 

Wht kind of a probe are you using?

lacosa83, 2017-03-02:
A pieve of paper ;D

Bob-the-Kuhn, 2017-03-02:
Tried & true and perfectly acceptable for CNC work  :-)

Allted, 2017-03-04:
Wow, nice work!

Sorry we aren't on the latest RCBugFix yet, the updates are happening so fast I decided to wait for the next RC. I hope we can implement this on the next MPCNC firmware update, or at least put up a how to guide.

Very impressed.

Bob-the-Kuhn, 2017-03-05:
Once you have a working system then stick with it.

Usually people only switch if there is a bug fix they need or a new feature they want.

closed 2017-10-31 by thinkyhead