MarlinFirmware/Marlin #5916: Possible bug in heating protection

Pipshag, 2017-02-26:
I just broke my thermistor on Heater 0 and found a bug that is possibly a safety issue. 

On RC6 I get an ERR: MAXTEMP error as soon as the printer powers on but on f9e4717 the printer powers on very happily with a temp reading of 100C on the LCD (but it shows 1100C in OctoPrint). Might this be case of the value being too great to be picked up as unsafe?

My configuration is located here https://github.com/Pipshag/Marlin/tree/BOWDEN_IDEX_RCBugFix

```
#define THERMAL_PROTECTION_HOTENDS
...
#define HEATER_0_MINTEMP 5
#define HEATER_0_MAXTEMP 400
```

Tannoo, 2017-02-27:
Broken should result in MINTEMP error not a MAXTEMP error.

I tested this on mine by pulling the thermistor plug and setting the hotend to 1C. I got:
```
Err: MINTEMP
PRINTER HALTED
Please reset
```
Just as expected.
I am running RCBugFix.

Roxy-3D, 2017-02-27:
@Tannoo   Thank You!

Tannoo, 2017-02-27:
Np. I didn't think there was an issue, I just thought I would try it because it's been a while since I had a broken thermistor. (Usually the bed for me).

Pipshag, 2017-02-27:
So I understand, the checks don't trigger before we try to actually heat the tool. Reflashed and tried once again and it works just as you tested. It went against my better judgement to start heating with a defective temperature sensor.

I'm not really certain that this behavior is better than just shutting down with an error. Sure, aids in troubleshooting but are we never seeing a case where it might be bad?

On RC6 I got what I expect if my temperature is reading above normal, 1100C - an ERR.
Running latest (referenced above) boots up happily with T:1100.0.

closed 2017-02-27 by Pipshag

Tannoo, 2017-02-27:
>So I understand, the checks don't trigger before we try to actually heat the tool.

Correct

Pipshag, 2017-02-27:
In this case I have a value of 1100C (TOOL 0) on startup. The machine boots, runs and operates until I try to set a temperature on TOOL 0. I am just now printing something with TOOL 1 while TOOL 0 is showing 1100C (100C on the display).

What you are describing is totally correct. What I referred to as "normal" was the ambient temp, or whatever the temperature would actually be.

Tannoo, 2017-02-27:
hmmm... what thermistor do you have setup? When mine breaks, it goes low, not high.

Pipshag, 2017-02-27:
It's a PT100. There might be a false contact and not a total break.

Tannoo, 2017-02-27:
If you unplug it from the board, does it go high also?

Pipshag, 2017-02-27:
Yes, I unplugged both sensors from the amplifiers and got the same temp reading (1100C).

Tannoo, 2017-02-27:
Going to have to leave this one for the pros. Sorry. I'm at a loss.

Blue-Marlin, 2017-02-27:
There are two main failures with resistance based temperature sensors:
* shortcut  -> ~0 Ohm -> ~0 Volt at the ADC -> low ADC reading
* break -> infinite Ohm -> ~ VCC Volt at the ADC -> high ADC reading

Depending of the __T__ emperature __C__ oefficient of the resistor the ADC readings have to be interpreted different.
* __P__ ositive __TC__ -> high temperatures high resistance -> low ADC values (f.e. PT100 without amplifier)
* __N__ egative __TC__ -> high temperatures low resistance -> high ADC values (f.e. normal thermistors)

-> a broken wire results in a MINTEMP_ERROR with a Thermistor but a MAXTEMP_ERROR with a PT100 (without amplifier). 

-> the temperatures you can usually read on the display when a error occurs is determined by the first and last line of the table what is correlating ADC value and temperature (thermistortables.h).

(If there is an amplifier it depends on, if the amplifier is inverting or not)