MarlinFirmware/Marlin #5918: Atmega1284p AccelStepper.h and NEMA 17 Bipolar Stepper

Shrikant888, 2017-02-27:
Hi All,

I am creating a demo project where I am using Atmega1284p and A4988 stepper driver. 

The requirement is that threaded rod has to move nearly 10 cm within 30 seconds and after reaching 10cm it should come back. Please tell me how to use forwards and backwards.

Please help me.

Bob-the-Kuhn, 2017-02-27:
Assuming that the rod is attached to the X axis stepper motor then the following gcode script will move the carriage back & forth by 100mm:
```
G92 X0   ; set current position to zero
G0 X100  ; move 100mm in one direction
G0 Z0    ;  move it back to the zero position
```

Bob-the-Kuhn, 2017-02-27:
I just noticed the **AccelStepper.h** in the title.  Another example of why I shouldn't post late at night.

If all you're interested in is turning the stepper motor then the standard Arduino & Marlin will do just fine.

If you supply some details as to what you're trying to do then I should be of some help.

closed 2017-10-31 by thinkyhead