MarlinFirmware/Marlin #5920: multiple definitions

ruggb, 2017-02-27:
in configuration.h you have:
#define CUSTOM_MACHINE_NAME "CC_BOT" == my mod
and
#define MACHINE_UUID "277494b7-b7c8-4eb1-a653-c2e695c24911" == my mod
then in version.h you have:
  #define MACHINE_NAME "3D Printer"
and
  #define DEFAULT_MACHINE_UUID "cede2a2f-41a2-4748-9b12-c55c62f367ff"

Why the need to preserve DEFAULT and to add CUSTOM?
can't it just be:
#define MACHINE_UUID "277494b7-b7c8-4eb1-a653-c2e695c24911" == my mod
and
 #define MACHINE_NAME "CC_BOT" == my mod
in config...h?

ruggb, 2017-02-27:
OK, the Sanity check looks at the MACHINE_NAME and DEFAULT_MACHINE_UUID. So why not just use those in the config and delete the ones in version. Does not look like 2 sets are needed, are they?

thinkyhead, 2017-02-28:
Machine name may also be overridden by pins files. Everything comes together in `language.h` with…
```cpp
#ifdef CUSTOM_MACHINE_NAME
  #undef  MACHINE_NAME
  #define MACHINE_NAME CUSTOM_MACHINE_NAME
#else
  #ifdef DEFAULT_MACHINE_NAME
    #undef  MACHINE_NAME
    #define MACHINE_NAME DEFAULT_MACHINE_NAME
  #endif
#endif

#ifndef MACHINE_UUID
  #define MACHINE_UUID DEFAULT_MACHINE_UUID
#endif
```

ruggb, 2017-02-28:
I still don't see why it isn't handled with one definition. There isn't 3 different uses for MACHINE_NAME is there. It is all the same machine. One def in config file with a default name in case the user doesn't have one.
The same with the UUI. Why would the machine have 2 different UUIs?

ruggb, 2017-02-28:
@thinkyhead - I have the latest build working with most all my mods in it. The rest of them are just completing the format changes and maybe relocating a few more #defines. I think it makes this a lot easier for everyone. I would really like you to take a look at it and see what you think.

thinkyhead, 2017-03-01:
The idea is that a Version file could supersede a Configuration file in some theoretical vendor scenario.

thinkyhead, 2017-03-01:
> I still don't see why it isn't handled with one definition. There isn't 3 different uses for MACHINE_NAME is there. It is all the same machine. One def in config file with a default name in case the user doesn't have one.

We have a default name that is provided by `Version.h` and pins files (`DEFAULT_MACHINE_NAME`), a custom name provided by a user to override without the need for `#undef` (`CUSTOM_MACHINE_NAME`), and the final value, determined in `language.h`, that is actually used (`MACHINE_NAME`). The only one the user ever sees is `CUSTOM_MACHINE_NAME`.

closed 2018-02-26 by thinkyhead