MarlinFirmware/Marlin #5921: Dual Extruder, Secondary Extruder: Servo Actuated, Dual Bowden Steppers

MrJulz, 2017-02-27:
(have searched thoroughly but can not yet find how to do this)

This printer is custom, One printhead/carriage, Dual Extruders/Hotends, Dual Bowden Stepper Motors (one for each filament). E1 (second) extruder is servo-actuated, to raise/lower it above/below the height of E0, so a Z-Offset will also be required.

I am seeking guidance on how to implement Dual Hotends/Dual Drivers/E1 Servo Actuated in Marlin - is this configuration already supported?

I note this confuig is almost, but not quite, the same as SWITCHING_EXTRUDER - in that a servo must be actuated to switch the hotend in both, but SWITCHING_EXTRUDER assumes a single direct-drive stepper motor - not two independent Bowden stepper motors.  So it appears SWITCHING_EXTRUDER is very close to what I need, but not quite?

I also see that  `#definine EXTRUDERS 2 `allows only for X & Y offsets only.  AFAICT only SWITCHING_EXTRUDER includes support for` HOTEND_OFFSET_Z`.

Question: How to implement a dual extruder/dual bowden with servo-actuation? The following is a best guess at my options.  Any pointers/confirmation of direction appreciated...
a) It is already supported in Marlin, here is a link with details on how to implement it...
OR
b) Use SWITCHING_EXTRUDER, as it already supports servo control & Z-Offset, and I will need to change <INSERT HERE> to make it support dual stepper bowden drivers
OR
c) just `#define EXTRUDERS 2`, and alter <INSERT HERE> in Marlin_main.cpp to add servo activation and Z-offset support.
OR
d) Something else entirely.

Appreciate any pointers/guidance.

christianh17, 2017-02-28:
Hi! 

Intersting, I have almost the same problem. My extruder does not move up and down but the extruder is tilted by 18° via a servo so there is only one of them in contact to the printed material. I am still not totally happy with my configuration so I am interested in your solution. Can I find your extruder on thingiverse? For now my printer works with manual turning the servo and using both extruders. I wanted to have a look at the configuration next weekend and always assumed it would be no problem to do a configuration to support this setting. But you are right it does not seem to be as easy as I thoguht.

To answer your question: 
I would try at the first step to comment out the following lines

```
/**
 * Extruder indirection for the single E axis
 */
#if ENABLED(SWITCHING_EXTRUDER)
  #define E_STEP_WRITE(v) E0_STEP_WRITE(v)
  #define NORM_E_DIR() E0_DIR_WRITE(current_block->active_extruder ?  INVERT_E0_DIR : !INVERT_E0_DIR)
  #define  REV_E_DIR() E0_DIR_WRITE(current_block->active_extruder ? !INVERT_E0_DIR :  INVERT_E0_DIR)
```

in stepper_indirection.h and define SWITCHING_EXTRUDER in config.h. 

This should bring back both extruder steppers. Unfortunately I will not be at home till Friday so I cannot test it. If it works I would try to do a PR, it is always nicer to be close to the original. But of course this is a lot more work...
Kind regards
Christian

MrJulz, 2017-02-28:
Great, thanks for the tip! I'll test & experiment and let you know...

MrJulz, 2017-02-28:
Great, thanks for the tip! I'll test & experiment and let you know...

Unfortunately I do not have this print head on thingiverse. I can offer you a brief description though if that helps? Along the power and control cable channel housing to the print head, also have a bike brake cable in bike cable housing, so a servo (mounted on the housing) can pull the cable and lower/raise the secondary extruder as required. (The secondary extruder is spring-mounted). Feel free to pm me with any questions. Cheers.

thinkyhead, 2017-03-01:
Reposting from the other topic… This would be in every way just like the standard `EXTRUDERS 2` except there will be a servo actuation on tool-change. Everything else that `EXTRUDERS 2` alone does will also apply here.

MrJulz, 2017-03-01:
Thanks Scott, understand more now, greatly appreciate the pointers.

Agreed, going forward having the ability to set a servo (and thus also a Z Offset) for each extruder sounds great, and extensible.  As my printer is currently of this configuration, but looking to extend to 4 extruders in the next few months, I will see what I can do here as my understanding of the code improves.  (Marlin appears "not small", is taking me a little time to grok).  I note (on many occasions) and appreciate the way you invite people to contribute.

--

Additionally, as I see it at present, `SWITCHING_EXTRUDER` is for all purposes identical to "dual hotends, secondary hotend servo actuated.  Identical in that both benefit from:
- Automatic servo actuation-on-tool-change
- Hotend Z Offset.

The only difference being: `SWITCHING_EXTRUDER` has the new code (#3994) to use a single stepper, and to reverse stepper  direction on-tool-change.

So, as I am currently viewing it, there are two cases of `SWITCHING_EXTRUDER`:
1. `SWITCHING_EXTRUDER` (with one stepper per each hotend)
2. `SWITCHING_EXTRUDER_WITH_SINGLE_STEPPER`

If my understanding above is correct, does that mean I should be able to, for now, given time constraints, simply use `SWITCHING_EXTRUDER` and add an additional #define **within** e.g `#define `SWITCHING_EXTRUDER_WITH_SINGLE_STEPPER`, and simply use that #define to enable/disable the new "single reversing stepper" code?

Does that seem like a reasonable immediate solution?

MrJulz, 2017-03-02:
Thanks very much for the help gents. With you pointers, I got it up and running today pretty fast. @christianh17 I'll provide what I did for refererence, it isn't pretty or ideal, but it works for now. (no manual servo turning, z_offset etc)

Summary: `SWITCHING_EXTRUDER` (S_E) does exactly what we require, except we also require one-stepper-per-extruder, as per normal. e. I.e simplified can just treat these extruder as a "subclass of S_E"

Configuration.h:
`
#define EXTRUDERS 2
#define SWITCHING_EXTRUDER
#if ENABLED(SWITCHING_EXTRUDER)
  ...
  #define SWITCHING_EXTRUDER_WITH_MULTIPLE_STEPPERS
#endif`

At the bottom of ConditionalsLCD.h (strange place for this code???)
`  #if ENABLED(SWITCHING_EXTRUDER_WITH_MULTIPLE_STEPPERS)
    #undef HOTENDS
    #undef E_STEPPERS
    #undef E_MANUAL
    #undef TOOL_E_INDEX
    #define HOTENDS      EXTRUDERS
    #define E_STEPPERS   EXTRUDERS
    #define E_MANUAL     EXTRUDERS
    #define TOOL_E_INDEX current_block->active_extruder
  #endif
#endif //CONDITIONALS_LCD_H`
Note: I realise the above is not efficient, but it provided an effective quick and fast way for me to "just add code to the end" for testing, rather than interspersing my testing code within Marlin code - I would save that work for a PR)

indirection.h (changed just the one line you suggested, thanks!)
`#if ENABLED(SWITCHING_EXTRUDER) && (!ENABLED(SWITCHING_EXTRUDER_WITH_MULTIPLE_STEPPERS))
  #define E_STEP_WRITE(v)..........`

Thats it! Works like a charm.  Obv. I wouldn't do it this way for a PR, I'm not a great coder, but it works for now.

MrJulz, 2017-03-02:
One caveat:

When switching from E0 to E1, the sequence should be (and is):

1. Raise Z

christianh17, 2017-03-04:
Today I had time to do the changes MrJulz suggested: It works in my setting, too! Perfect!
I am glad you were some days before me so as soon as I needed a solution it was there.

Only problem: I have some problems getting the height correct.  I tested so much but in the meantime I think the printer has its own ideas, sometimes it works and sometimes not. The z-probe is highly reliable, I tested it with M48. I am still trying to investigate the problem.

MrJulz, 2017-03-04:
The printing height of either/both nozzles is giving your problems, or just the printing height of the second nozzle E1?

christianh17, 2017-03-05:
Hi! I just spent severl hours with this issue and I think I know the problem now. As it has not much to do with SWITCHING_EXTRUDER_WITH_MULTIPLE_STEPPERS I will open a new issue (just now). BTW: I am still interested in knowing more about your extruder but unfortunately there is no pm at github.

thinkyhead, 2017-03-17:
> When switching from E0 to E1, the sequence…

@MrJulz I've submitted a fix that fixes the ordering so the Z lowering is last, and it happens to also solve the wrong-z-after-toolchange issue at the same time. https://github.com/MarlinFirmware/Marlin/pull/6042/commits/421468ab0acffe0d948b9952c1107bd8e7ef13c8

MrJulz, 2017-03-17:
That is great news, gratitude to you.

closed 2017-10-31 by thinkyhead