MarlinFirmware/Marlin #5922: CUSTOM_BOOTSCREEN_TIMEOUT

ruggb, 2017-02-27:
this appears in ultralcd_impl_DOGM.h on line 278, but I can't find it defined anyplace.
Can you tell me where it is defined?

Blue-Marlin, 2017-02-27:
https://github.com/MarlinFirmware/Marlin/search?utf8=%E2%9C%93&q=CUSTOM_BOOTSCREEN_TIMEOUT

ruggb, 2017-02-28:
Well now, there ya go. Thank you -- however it doesn't do ME or a lot of other ppl any good as I do not use ANY of those preconfigured builds. Can you say **BUG**. This needs to be defined someplace in the general config files - I have placed it and would suggest configuration_adv.h as it is not critical to operation.

Blue-Marlin, 2017-02-28:
If you don't have a custom bootscreen you don't need `CUSTOM_BOOTSCREEN_TIMEOUT`.
So `_Bootscreen.h` is the exact right place for the definition.

ruggb, 2017-02-28:
OH, it is only for a REAL custom bootscreen. not a custom timeout for a bootscreen. I got it confused with MACHINE_NAME and CUSTOM_MACHINE_NAME. I still don't know why there are two defined. The same for the MACHINE_UUI

closed 2017-03-01 by thinkyhead