MarlinFirmware/Marlin #5923: UBL issues

scott1326, 2017-02-28:
Tried to get UBL working again today, but I've run into a couple issues.

1) Config file asks to define one of several z probe variants:  servo, sled, fixed, etc.  You cannot define FIX_MOUNTED_PROBE without getting compile errors, as there is no way to undefine servos.  I tried to put NUM_SERVOS to 0, or to comment it out, but that leads to more define errors.  It will compile if I leave FIX...  undefined, but my machine has a fix mounted induction probe and no servos.

2) UBL docs clearly state: 
     "If you have too little clearance and not much plastic was extruded in an area, you want to LOWER the Mesh Point at the location.  If you did not get good adheasion, you want to RAISE the Mesh Point at that location."
     I have found that to be the exact opposite.  I have to make the mesh points more negative in order to get the hotend closer to the bed (ie, lower Z).

3) After generating what I think is a valid mesh, and doing several G26 tests to verify, I tried to do a quick calibration print which air-printed several mm above the print bed.

Cheers-
    Scott1326

Roxy-3D, 2017-02-28:
> UBL docs clearly state:
> "If you have too little clearance and not much plastic was extruded in an area, you want to LOWER the
>  Mesh Point at the location. If you did not get good adheasion, you want to RAISE the Mesh Point at 
> that location."

Yes.   You are correct.   That needs to be changed!

scott1326, 2017-02-28:
Hi Roxy,
     Thanks for the prompt reply.  All commands at power on.  I realize the top row of the mesh is unpopulated.  I'll worry about that later when the rest of it is working.  G29 N W below:

Connecting...
start
Printer is now online.
echo:Marlin 1.1.0-RCBugFix
echo: Last Updated: 2016-04-27 12:00 | Author: Scott
Compiled: Feb 28 2017
echo: Free Memory: 2199  PlannerBufferBytes: 1232
echo:V30 stored settings retrieved (263 bytes)
 UBL Active!
Initializing Bed Leveling State to current firmware settings.
Mesh loaded from slot 1  at offset 0x0EF4
Mesh 1 loaded from storage.
echo:Steps per unit:
echo:  M92 X80.00 Y80.00 Z4000.00 E616.00
echo:Maximum feedrates (mm/s):
echo:  M203 X300.00 Y300.00 Z2.60 E25.00
echo:Maximum Acceleration (mm/s2):
echo:  M201 X3000 Y3000 Z100 E10000
echo:Accelerations: P=printing, R=retract and T=travel
echo:  M204 P3000.00 R3000.00 T3000.00
echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
echo:  M205 S0.00 T0.00 B20000 X20.00 Z0.40 E5.00
echo:Home offset (mm)
echo:  M206 X0.00 Y0.00 Z0.00
Unified Bed Leveling:
echo:System is: Active
Active Mesh Slot: 1
z_offset: 0.000000
MESH_NUM_X_POINTS  5
MESH_NUM_Y_POINTS  5
MESH_MIN_X         0
MESH_MIN_Y         0
MESH_MAX_X         200
MESH_MAX_Y         200
MESH_X_DIST        50.000000
MESH_Y_DIST        50.000000
echo:Material heatup parameters:
echo:  M145 S0 H180 B70 F0
echo:  M145 S1 H240 B110 F0
echo:PID settings:
echo:  M301 P36.07 I3.67 D88.65 C100.00 L20
echo:  M304 P352.23 I67.61 D458.77
echo:Filament settings: Disabled
echo:  M200 D1.75
echo:  M200 D0
echo:Z-Probe Offset (mm):
echo:  M851 Z-2.20
>>>g29 o
SENDING:G29 O
echo:XYZ need homing.
X:100.00 Y:140.00 Z:4.00 E:0.00 Count X: 8000 Y:11200 Z:16000
Bed Topography Report:
(0,4)                                                                        (4,4)
(0,200)                                                                    (200,200)
       .                 .                 .                 .                 .

 -0.76225    -0.26800   [ 0.03325]   0.18025     0.10250

 -0.72500    -0.23425     0.01825     0.08475    -0.10425

 -0.61625    -0.18300     0.08475     0.19425    -0.02050

 -0.52650    -0.14700     0.14725     0.21625     0.03800
(0,0)                                                                        (200,0)
(0,0)                                                                           (4,0)

>>>g29 N W
SENDING:G29 N W
Version #4: 10/30/2016 branch
Unified Bed Leveling System Active.  -------------------------------------       <----<<<
Mesh: 0001 Loaded.   --------------------------------------------------------       <----<<<
G29_Correction_Fade_Height : 10.00  -------------------------------------       <----<<<
z_offset: 0.000000  ------------------------------------------------------------       <----<<<
X-Axis Mesh Points at: 0.0  50.0  100.0  150.0  200.0
Y-Axis Mesh Points at: 0.0  50.0  100.0  150.0  200.0
Kill pin on :41  state:1
UBL_state_at_invokation :0
UBL_state_recursion_chk :0
Free EEPROM space starts at: 0x0120
end of EEPROM              : 0FFF
sizeof(blm) :  134
z_value[][] size: 100
EEPROM free for UBL: 0x0EDF
EEPROM can hold 0x0026 meshes.
sizeof(stat)     :0043
MESH_NUM_X_POINTS  5
MESH_NUM_Y_POINTS  5
MESH_MIN_X         0
MESH_MIN_Y         0
MESH_MAX_X         200
MESH_MAX_Y         200
MESH_X_DIST        50.000000
MESH_Y_DIST        50.000000
sizeof(block_t): 77
sizeof(planner.block_buffer): 1232
sizeof(char): 1   sizeof(unsigned char): 1
sizeof(int): 2   sizeof(unsigned int): 2
sizeof(long): 4   sizeof(unsigned long int): 4
sizeof(float): 4   sizeof(double): 4
sizeof(void *): 2   sizeof(struct pf): 1   sizeof(void *()): 1
Unified Bed Leveling sanity checks passed.

Roxy-3D, 2017-03-02:
When I set up a mesh...  I first get my Z_PROBE_OFFSET_FROM_EXTRUDER set very closely to correct.   And I start with the z_offset value at 0.0000    

In theory...  There isn't even a need for the z_offset number.    Is that the process you followed?   Can you tell us how you did things? 

But even so...  Even if you have your z_offset and Z_PROBE_OFFSET_FROM_EXTRUDER wrong...    You should be able to edit and adjust the mesh such that everything works out correctly and you get good adhesion across 100% of your bed.

scott1326, 2017-03-03:
My Z_PROBE_OFFSET_FROM_EXTRUDER is -2.20.  I believe the probe reads at about this level below the hotend.
I run a G28 command, and then a G29 P1.  It probes a 5x5 grid.  I save the result with G29 S1.  Then I run a G29 P6 C-x.xx.  I shift the entire mesh down by about .2 to .3 mm because the initial readings are nowhere close.  I save again with G29 S1.  Then G26 H188 (I have a heated bed, and I print PLA with the bed at 60).  I will try to edit individual points with G29 P4 or M421 (I think, can't remember if this works in UBL or not) and save again, run G26 again, etc, until I'm satisfied.
I then run a quick print (usually a rectangle border that I use for calibration), which starts printing at about 4 or 5 mm above the bed.  This print prints fine with other builds of Marlin as long as I have the bed leveled.  I usually use RC8 and Manual/Mesh bed leveling, I haven't yet got UBL to work satisfactorily.
I do have a LCD panel and inductor probe, and it's a Prusa i3 printer that I sourced parts for and assembled.

closed 2017-10-31 by thinkyhead