MarlinFirmware/Marlin #5925: timeout for babystepping

ruggb, 2017-02-28:
timeout for babystepping returning to status screen is still not functioning in RC8 493e738 - this has been in a couple threads and @Roxy-3D verified it but I have no idea if anyone it attending to it.

Roxy-3D, 2017-02-28:
I think what needs to happen is this:   I think a Git Ninja ought to look at how many changes have been merged into ultralcd.cpp since it was known to be working.   And with a couple binary searches to see where it works and doesn't work...   It probably will be obvious the piece of code causing the problem.

ruggb, 2017-02-28:
not that it is causing a big problem b/c you can click out of it. It just doesn't work like it used to. If that is design intent I can live with it. But I suspect it wasn't intentional.

Kaibob2, 2017-02-28:
See #5570

ruggb, 2017-02-28:
OK, leave it.

Roxy-3D, 2017-03-01:
Oooops....  I posted a reply in the wrong place:   https://github.com/MarlinFirmware/Marlin/issues/5570#issuecomment-283219079

thinkyhead, 2017-03-01:
The timeout is now configurable. If unset the default is 15s.

Tannoo, 2017-03-02:
How about a short note at the bottom of the screen to tell the user to Click to return? Not everyone is as keen on Marlin, nor active on git.
Or, how about a timeout for babystepping that's 1 or 2 minutes? Or, some multiple of the current LCD timeout period? (Surely, that's enough time to get a skirt around the full bed)

Roxy-3D, 2017-03-02:
It is easy enough to add a "Click to return to status screen" message at the bottom of the LCD screen.  

But isn't it obvious the display is alive and waiting?   If the encoder wheel is turned you see the numbers change.   The only other possibility is to click the encoder wheel to lock in the current value?

ruggb, 2017-03-02:
@Roxy-3D A more used function might be to return that "locked in" value the next time you went to BS during the same print. It doesn't really seem to be "locked in" at this point because it will start at 0.000 every time you go back to BS.

closed 2018-02-26 by thinkyhead