MarlinFirmware/Marlin #5927: Clearance between probes is too high

Jean-Fred-CEC, 2017-02-28:
Hi everybody, I updated the last marlin firmware (1.1.0) in my home made 3D printer and I don't know how to reduce the height that my probe raise between each probe point.
I have this kind of configuration:[https://thebreizmaker.wordpress.com/autoleveling-bed/](url)
so I defined a fixe mounted probe, and my Z offset is negative .
My problem is that we can not define a négative probe clearance, so the probe raises by the Z_offset value and it is far too long...
With the older marlin firmware I didn't have this problem, I think I didn't understand the new firmware.
Can someone please help me?

thinkyhead, 2017-03-01:
What version did you upgrade to?

The configuration option in the latest Marlin is called `Z_CLEARANCE_BETWEEN_PROBES`. See http://marlinfw.org/docs/configuration/configuration.html#probe-clearance

Jean-Fred-CEC, 2017-03-01:
Hi Thinkyhead, I upgraded to 1.1.0 (RC8). I agree with youbut I can't put a negative value, so my probe raises by at least 22mm between each point I would like to raise about 4mm max if possible: so my clearance would be -18mm but it is not possible

#define X_PROBE_OFFSET_FROM_EXTRUDER 20  // X offset: -left  +right  [of the nozzle]
#define Y_PROBE_OFFSET_FROM_EXTRUDER 50  // Y offset: -front +behind [the nozzle]
#define Z_PROBE_OFFSET_FROM_EXTRUDER -22.0   // Z offset: -below +above  [the nozzle]

#define Z_MIN_PROBE_ENDSTOP

#define Z_CLEARANCE_DEPLOY_PROBE   0 // Z Clearance for Deploy/Stow
#define Z_CLEARANCE_BETWEEN_PROBES   0// Z Clearance between probe points

#define Z_PROBE_OFFSET_RANGE_MIN 0
#define Z_PROBE_OFFSET_RANGE_MAX 0

I don't understand the offset range...

Jean-Fred-CEC, 2017-03-04:
Can someone explain me something about the offset range?

thinkyhead, 2017-03-06:
To get the correct value for `Z_PROBE_OFFSET_FROM_EXTRUDER`, measure the maximum Z height of the nozzle where the probe gets triggered, then negate that distance.

If you want the Z axis to raise for clearance between probe points set `Z_CLEARANCE_BETWEEN_PROBES`. For the minimum height of the initial probe deployment (and for stowing after probing is done) set `Z_CLEARANCE_DEPLOY_PROBE`.

Jean-Fred-CEC, 2017-03-06:
Thanks for answering.
That's what I did, but my problem is that I need a negative value on Z_CLEARANCE_DEPLOY_PROBE and I can't implement it below zero...
With old marlin the problem didn't occur

thinkyhead, 2017-03-06:
>  I need a negative value on Z_CLEARANCE_DEPLOY_PROBE 

How come? A negative value would make the probe press against the bed. Are you trying to do a special move to stow the probe by pressing against the bed?

thinkyhead, 2017-03-06:
> between each point I would like to raise about 4mm max

Set `Z_CLEARANCE_BETWEEN_PROBES` to `4.0` to get a 4mm raise between probes.

thinkyhead, 2017-03-06:
> Can someone explain me something about the offset range?
```cpp
//
// For M851 give a range for adjusting the Z probe offset
//
#define Z_PROBE_OFFSET_RANGE_MIN -20
#define Z_PROBE_OFFSET_RANGE_MAX 20
```
It limits the range of Z offset that you can set with `M851` or the LCD.

Jean-Fred-CEC, 2017-03-06:
I need this negative value because my probe is self retractable and between each probe it raises by the  Z_offset value  (22mm). I just need far less..(approx 7mm)

thinkyhead, 2017-03-06:
If it self-retracts, you probably still need a small raise so when it re-deploys it is off the bed. A `Z_CLEARANCE_BETWEEN_PROBES` of `2.0` should be sufficient.

thinkyhead, 2017-03-06:
> between each probe it raises by the Z_offset value (22mm)

22mm should be the (approximate) height of the nozzle when the probe triggers. It should not raise an additional 22mm between probings unless `Z_CLEARANCE_BETWEEN_PROBES` is set to `22.0` (or something like that).

Jean-Fred-CEC, 2017-03-06:
That's clearly what I don't understand:
 my Z_CLEARANCE_DEPLOY_PROBE=0 and Z_CLEARANCE_BETWEEN_PROBES=0 and it raises by 22

thinkyhead, 2017-03-06:
The probe extends to 22mm below the nozzle. So there is no point where it should go below Z=22, other than the actual bed probe.

Are you saying your nozzle raises to Z=44?

Jean-Fred-CEC, 2017-03-06:
Thanks for your help!
Right, we can say that, my probe triggering point is at -22 below the nozzle and as it raises by 22 between each point it doubles the value.

Jean-Fred-CEC, 2017-03-06:
So how can I ask to the firmware to raise the nozzle to 25 instead of 44?

thinkyhead, 2017-03-06:
The correct settings are as described.
```cpp
#define Z_PROBE_OFFSET_FROM_EXTRUDER -22.0   // Z offset: -below +above  [the nozzle]

#define Z_CLEARANCE_DEPLOY_PROBE   4.0 // Z Clearance for Deploy/Stow
#define Z_CLEARANCE_BETWEEN_PROBES   2.0// Z Clearance between probe points
```
If these settings are not working, try first using `M502` followed by `M500` to ensure these values are being used. If that fails to resolve the issue then update to `RCBugFix` and start from the most reliable code-base.

Jean-Fred-CEC, 2017-03-06:
It didn't resolve the issue, the probe is now raising by 22+2=24 between each point.
I have already the last RCBugfix code-base (1.1.0)

Jean-Fred-CEC, 2017-03-06:
Are these lines a part of my problem? (Marlin_main.cpp lines 2181 & 2182)
 // Ensure a minimum height before moving the probe
    do_probe_raise(Z_CLEARANCE_BETWEEN_PROBES);

In the old marlin that was working these lines didn't exist

closed 2017-10-31 by thinkyhead