MarlinFirmware/Marlin #5936: meaning of the number in the bilinear leveling grid (just a question not a bug)

jfpion, 2017-03-02:
with pronterface and rc8  I got

Bilinear Leveling Grid:
      0     1     2
 0 +0.01 -0.13 -0.25
 1 +0.08 +0.01 -0.01
 2 -0.09 +0.04 +0.20

after the g29 (witch work perfectly on my prusa 2 by the way, thank you again)
but what is the meaning  ? is it the diference from the g28 z0 just before ? in mm ? i can't find the answer in the documentation

thank you

FHeilmann, 2017-03-02:
The points represent the height of the probed points in relation to the "zero-point" your G28 found. 

Make sure to close the issue if this answers your question.

jfpion, 2017-03-02:
I suppose it is in millimeters ?

closed 2017-03-02 by jfpion

FHeilmann, 2017-03-02:
Yes

aplocher, 2018-06-08:
Does point 0, 0 represent the first point probed?  And point 2, 2 the last point probed?  I'm trying to figure out what the numbers represent, as well.  Which direction is my bed represented with by those numbers...  Thanks

thinkyhead, 2018-06-08:
@aplocher — 0,0 is the mesh point closest to the smallest XY position in the measured bounds. So, if the bounds specified for `G29` were given as L20 F20 R180 B180 then...
```
X020 Y180     X100 Y180     X180 Y180
   0    2        1    2        2    2

X020 Y100     X100 Y100     X180 Y100
   0    1        1    1        2    1

X020 Y020     X100 Y020     X180 Y020
   0    0        1    0        2    0
```

aplocher, 2018-06-10:
Thank you that makes perfect sense and is exactly what I was looking for!