MarlinFirmware/Marlin #5939: Screen issue with i2c 2004 lcd display

nielsnl68, 2017-03-02:
Hi, 
i have hooked up an i2c lcd display to follow the print process.  Using the dcf8574t
Sadly the shown result is not displayed correctly.  For example the temperature indications are in correct and not shown correctly. And the x y z values are not shown at all. 

I have already been looking into it for some time and i know where it displays the values but when i try to add lcd.print (T1); \  then i get a one digid value instead a 2 or 3 digid value. 

See attatched image.
[image]

nielsnl68, 2017-03-05:
Nobody else has this issue?

thinkyhead, 2017-03-05:
It does appear, only the first digit is coming through on each individual call to 'lcd.print`. This sounds like it could almost be caused by a program error, but it could also be caused by some mis-configuration where the character mapping is not correct, or where the display is expecting more bits to come through per-character. At the outside, it could be caused by bad timing to the display, but this seems the most unlikely.

I note that the string "RigidBot ready." comes through fine. And, it happens to use the `lcd_print` function to print that line instead of the (more direct) `lcd.print` function. So it's possible that replacing all the `lcd.print(char*)` calls with `lcd_print(char*)` will solve the issue for your situation. I'm not sure whether that's a universal solution, but it may be.

nielsnl68, 2017-03-05:
Thanks Scott,

You are totally right i modified the code and it works now. Should I add a
PR  for this?

closed 2017-10-31 by thinkyhead

nielsnl68, 2017-10-31:
Is this now fixed?