MarlinFirmware/Marlin #5940: Cooling FAN not started with extruder heating

walterX12, 2017-03-02:
I have RAMPS 1.4 and I have only 1 fan. The fan is for cooling down the "cool part" of extruder (not the printed parts). This fan is in D9 pins in RAMPS.

I can start and stop the fan using Repetier GUI or M106 or M107. 

I have found that when I start heating the extruder (using Marlin-RCMarlin 1.1.0-RC8 or Marlin-RCBugFix ) the fan does not start at all (at no extruder temperature).

I have tried:
1. #define E0_AUTO_FAN_PIN -1 - it ends up with described behaviour
2. #define E0_AUTO_FAN_PIN 9 - it ends up with error "You cannot set E0_AUTO_FAN_PIN equal to FAN_PIN."

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/815307/Marlin.zip)

It seems that there were many changes related to FAN PINS in the past few months. I assume that cooling down extruder is not something that needs to be started from GCode, right ? Is it a bug or do I have some settings wrong.

KevMags, 2017-03-02:
D9 is for the part cooling fan.  Wire your extruder fan straight to 12V or use a fan extender to automatically turn it on and off.

walterX12, 2017-03-03:
D9 is supposed to be 12 V (http://reprap.org/wiki/File:Arduinomega1-4connectors.png).
Do I understad correct that D9 will be used for fan to cool down the printed part ? 

I assume that 100% users have fan for cooling down HE , but not all users have fan for cooling down printed model.

It would make sense to have "out of the box" solution for HE cooling rather then for printed parts cooling.

I assume that many users were used to have fan on D9 and it worked previously fine (according to my colleague, because I moved from Repetier to Marlin recently).

Bob-the-Kuhn, 2017-03-03:
@KevMags is correct - FAN_PIN is now just a fan that you can manually control.

I could have sworn that at one time it defaulted to the E0 extruder fan.

@walterX12  - to use D9 to cool the extruder you'll need to do two things:
- Configuration_adv.h add **#define E0_AUTO_FAN_PIN 9** (as you had tried earlier)
- In pins_RAMPS.h add **#undef FAN_PIN** at line 196

walterX12, 2017-03-03:
Hi Bob-the-Kuhn and KevMags,

thank you, both for helping to solve the issue. Now the fan is started when HE is above 50 degrees without the Fan extender.

If anyone is trying to solve the same issue I am attaching my configuration changes.

I am wondering whether I am the only one who will need to do similar changes or there are many users having the same HW and being surprised that they need to buy fan extender until they will figure out that new line of code (rather than commenting / uncommenting lines) needs to be written.

It makes much more sense to edit only Configuration.h and Configuration_adv.h than to edit "low-level config file" pinsXXXX.h

This is my suggestion, but I am grateful for your help to solve my issue.

Have a nice weekend.

PicoPi, 2017-03-30:
Thanks. |I was having the same issue as above. Now fixed thanks to you.

vMeph, 2017-10-11:
it works but then you cant work then stops working this section

#define USE_CONTROLLER_FAN
#if ENABLED(USE_CONTROLLER_FAN)
  #define CONTROLLER_FAN_PIN 4 // Set a custom pin for the controller fan
  #define CONTROLLERFAN_SECS 60          // Duration in seconds for the fan to run after all motors are disabled
  #define CONTROLLERFAN_SPEED 255        // 255 == full speed
#endif

closed 2017-11-15 by thinkyhead

Ladvien, 2018-02-18:
@Bob-the-Kuhn hey man, that's what I needed.  One note, it looks like they've moved where the FAN_PIN is declared, so I thought I'd just add what I put in to make it work

```
#if ENABLED(IS_RAMPS_EFB)                      // Hotend, Fan, Bed
  #define FAN_PIN        RAMPS_D9_PIN
  #define HEATER_BED_PIN RAMPS_D8_PIN
#elif ENABLED(IS_RAMPS_EEF)                    // Hotend, Hotend, Fan
  #define HEATER_1_PIN   RAMPS_D9_PIN
  #define FAN_PIN        RAMPS_D8_PIN
#elif ENABLED(IS_RAMPS_EEB)                    // Hotend, Hotend, Bed
  #define HEATER_1_PIN   RAMPS_D9_PIN
  #define HEATER_BED_PIN RAMPS_D8_PIN
#elif ENABLED(IS_RAMPS_EFF)                    // Hotend, Fan, Fan
  #define FAN_PIN        RAMPS_D9_PIN
  #define FAN1_PIN       RAMPS_D8_PIN
#elif ENABLED(IS_RAMPS_SF)                     // Spindle, Fan
  #define FAN_PIN        RAMPS_D8_PIN
#else                                          // Non-specific are "EFB" (i.e., "EFBF" or "EFBE")
  #define FAN_PIN        RAMPS_D9_PIN
  #define HEATER_BED_PIN RAMPS_D8_PIN
  #if HOTENDS == 1
    #define FAN1_PIN     MOSFET_D_PIN
  #else
    #define HEATER_1_PIN MOSFET_D_PIN
  #endif
#endif
#undef FAN_PIN 
```

Bob-the-Kuhn, 2018-02-19:
We'd only move things around if you weren't looking.  😉

Glad to hear that you're up & running.

opodecano, 2018-06-06:
thanks, worked for me...

Configuration_adv.h add #define E0_AUTO_FAN_PIN 9 (as you had tried earlier) and..
pins_RAMPS.h add #undef FAN_PIN  (line 239)

Mangosniper, 2018-08-17:
If anyone has the same problem with a RAMPS 1.3 and Marlin 1.1.9, I just moved my extruder fan to PIN 44 and the print object fan to PIN 9. Then I set in the advanced config:

`#define E0_AUTO_FAN_PIN 44`

I did this as 

`#undef FAN_PIN`

in the pins_RAMPS.h disabled me to send M106 or M107

This is my pinout: https://drucktipps3d.de/wp-content/uploads/asgarosforum/195/anycubic_i3_mega_trigorilla_board.jpg

thinkyhead, 2018-08-21:
We have two options for Trigorilla — 1.3 and 1.4. The 1.4 version already has `E0_AUTO_FAN_PIN` set to 44.

sidhant, 2019-01-04:
I agree with @walterX12 about the default being able to correctly control the extruder/hot-end fan as that is basic functionality. Control of the part cooling fan is a perk. I ran into the exact same issue where my board has one connector labeled FAN and is a D9 pin. Not changing anything in the config, Marlin fails to kick the fan on when the hot-end even gets to ~190C. Turning on the `E0_AUTO_FAN_PIN ` results in compile error complaining that it cannot be the `FAN_PIN`.

I disagree that the solution is to connect extruder fan to 12V rails and have it be on at all times. The default handling of fan control is counterintuitive in Marlin.

Richie800, 2019-04-28:
in 1.1.9, can you change:

  #define RAMPS_D9_PIN       9  
  #define RAMPS_D9_PIN      -1  // Change made 

to fix the problem? Now you can use D9 however you choose.

itaycohen73, 2020-06-03:
Hello,
I was trying all suggestions here and no matter which one I choose, I get that same error message "You cannot set E0_AUTO_FAN_PIN equal to FAN_PIN"
These are the changes I was trying :
1. Set RAPS_D9_PIN to -1
2. Set #undef FAN_PIN
3. Commented all #define FAN_PIN lines 

Any suggestion why I'm still getting the error ?

Richie800, 2020-06-04:
Set:
Config.h.adv
#define E0_AUTO_FAN_PIN PD12
Commet out in pin.h
//#define FAN_PIN                             PD12   // Fan0