MarlinFirmware/Marlin #5944: Extruder heating message only shows 'S' on Full Graphics LCD status line

WheresWaldo, 2017-03-03:
I am trying to use RCBugFix up to commit eeb4902 (May 2nd) and appear to have an issue that I am not sure is configuration or something else. 

Bed heating shows the proper message of 'Bed Heating...' while the bed is coming up to temperature. As soon as it switches to hotend heating (there is only one extruder) the LCD status message changes to just the letter 'S' and not the 'E0 Heating...' that I expected. I have tried to go through _Configuration.h_ with a fine tooth comb and cannot find any part of it that would change the way the message should read. All other functions of the printer appear to be working well.

The slicer is S3D and the start up gcode is always the same (except for temps obviously) and it looks like this:

```G90
M82
M106 S0
M140 S90
M190 S90
M104 S235 T0
M109 S235 T0
M82
G1 Z15
G21
G28
G29
G90
M117 Printing ...

Tannoo, 2017-03-03:
Can you see if you have the same issue I had here: #5876?

WheresWaldo, 2017-03-03:
You did a lot of testing let me try some of the same and let you know. This might be a duplicate issue. I'll get back to you in about an hour (my family is still sleeping).

benlye, 2017-03-03:
Also looks like the same issue (but different message) as #5894.  I get 'LLED.' when heating the extruder.

WheresWaldo, 2017-03-03:
I haven't had a chance to test yet, but I just got a report that said someone else using the exact firmware and configuration had "h 1m 9089s" on the status line when heating switched to extruder (T0).

Tannoo, 2017-03-03:
I did a partial reversion to get the message back until this is resolved.

In `marlin_main.cpp`, I commented out this line:
```
//if (code_value_temp_abs() > thermalManager.degHotend(target_extruder)) status_printf(0, "E%i %s", target_extruder + 1, PSTR(MSG_HEATING));
```
and inserted this line:
```
if (code_value_temp_abs() > thermalManager.degHotend(target_extruder)) LCD_MESSAGEPGM(MSG_HEATING);
```

Also commented out this line:
```
//if (code_value_temp_abs() > thermalManager.degHotend(target_extruder)) status_printf(0, "E%i %s", target_extruder + 1, PSTR(MSG_HEATING));
```
and inserted this line:
```
if (code_value_temp_abs() > thermalManager.degHotend(target_extruder)) LCD_MESSAGEPGM(MSG_HEATING);
```

This is temporary, but it does put the correct message of `Heating...` back on the status line.

thinkyhead, 2017-03-03:
#5948 should fix this.

closed 2017-03-03 by thinkyhead