MarlinFirmware/Marlin #5945: Heating Failed Problem

AlexDoran, 2017-03-03:
Hi there,

I have a gMax 1.5 at work that i recently decided to upgrade to the latest BugFix, i'm getting a random Heating Failed Error always on the second layer.

The printer is configured as dual extruder, no heatbed. I tried increasing the watch window from 20 to 60 seconds with no change. The temperature does not actually drop or creep over target temp when the print is killed with heating failed.

Am i missing something here?? 

Thanks

Alex

Sebastianv650, 2017-03-03:
Please provide your configuration and configuration_adv files. Do you have it connected over USB, so you could get a temperature graph from Pronterface or a simmilar program?

When it's failing is the extruder number it reports equal to the realy used (heated) extruder number?

WheresWaldo, 2017-03-03:
[AlexDorian](https://github.com/AlexDoran) I had to redo the PID tuning moving to the latest bug fix. Not sure if it was purely coincidence but I was overshooting and undershooting a lot before stabilization . New PID parameters seems to have mitigated it.

AlexDoran, 2017-03-03:
I will try that thank you, i am printing from SD currently. I am watching the printer when the failure occurs and there temperatures are dead on target when it kills with Heating Failed.

bond4u, 2017-03-14:
I'm also getting "Heating failed" errors when printing stuff sliced with Cura-2.4.
No problems with prints sliced with Cura-2.3.1.
Marlin-1.1.0-RC8.

Sebastianv650, 2017-03-14:
@bond4u please try again with RCBugFix, there were some bugs in RC8 regarding such errors.

AlexDoran, 2017-03-15:
I switched from Cura to S3D and my problems went away.

bond4u, 2017-03-17:
Tried RCBugFix branch and still get "heating failed" error with Cura-2.4. No problems with Cura-2.3.1.
I did not touch settings in EEPROM, does it matter?
[files.zip](https://github.com/MarlinFirmware/Marlin/files/851081/files.zip)

bond4u, 2017-03-17:
As suggested in Cura issue https://github.com/Ultimaker/Cura/issues/1540, I enabled heated bed and set temp to zero and print succeeded.

Sebastianv650, 2017-03-17:
Than this looks like an Cura issue. Due you have the gcode from Cura 2.4 that crashes the printer? I guess it tries to set a bed temperature where you don't have a heated bed.
If so, maybe Marlin could check if there is a heated bed before it accepts a bed target temperature.

bond4u, 2017-03-17:
There is zip with gcode files attached to Cura issue.

Sebastianv650, 2017-03-17:
Indeed there is a `M140 S60` in row 240.

@ all coders, what's about inserting a #if TEMP_SENSOR_BED == 0 in `inline void gcode_M140()`. When the heat bed has no temperature sensor (no heat bed available), it could return an error message when a M140 is received?
Another option would be to remove the hole command when there is no heated bed?

Pucho586, 2017-12-16:
I have the same problem but with the active hot bed (RAMPS 1.4 EFB) and a Cyclops extruder. Randomly between the first and second layer the printer stops, alleging a temperature problem.
I also use curaengine but I would not know what version it is.
This is the right place to ask for help about this? Forgive my english, I am helped by google translate
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/1565192/Marlin.zip)

thinkyhead, 2017-12-17:
@Pucho586
- Does your fan start up between layers?
- Have you tried doing a new `M303` to re-tune the hotend?
- Have you tried relaxing your thermal protection settings?

Pucho586, 2017-12-17:
The fan works correctly
Never use or know how to use the M303 command
Try to extend the temperature range to. What happens is that the heating resistance of the hotend goes out
Thanks for such a quick response!!

thinkyhead, 2017-12-18:
> Never use or know how to use the M303 command

The simplest way to use it is to send `M303 U` followed by `M500`. When `M303` is finished, it also gives you some values to paste into your `Configuration.h`.

Pucho586, 2017-12-18:
Configure the PID and the error persisted. I wanted to do slicing with Slic3r (in case it was a problem of Curaengine) and when I finished working, and before I started printing, the error returned and the hotend went out without even starting to print. This error does not happen when I use it with a single extruder. Suddenly the X and Y motors make very short movements in place and the heating of the extruder stops

Pucho586, 2017-12-22:
The problem with version 1.1.7 was solved but the z axis does not obey the set offset. I saw a thread that carries that theme

thinkyhead, 2017-12-25:
> before I started printing, the error returned and the hotend went out without even starting to print.

Try increasing `WATCH_TEMP_PERIOD` to 40.

Show us a temperature graph from Repetier Host.

closed 2018-02-25 by thinkyhead