MarlinFirmware/Marlin #5946: Discussion: Protect Babystepping against extruder ISR interrupts

Sebastianv650, 2017-03-03:
There is a possibility to run into a bug with enabled babystepping I wasn't aware of up to now.
In current RCBugFix, inside the temperature ISR (which is executing the babysteps) Marlin re-enables serial and stepper ISRs.

If babystepping is enabled, they will executed [here](https://github.com/MarlinFirmware/Marlin/blob/RCBugFix/Marlin/temperature.cpp#L1923-L1936). `stepper.babystep` isn't allowed to be interrupted by another stepper ISR, it would harm the stepper signal.

I see two options to fix it:
- Encapsulate the highlighted lines by cli() and sei().
This will block all ISRs during the babystep execution and therefore restore the original behaviour as it was before the PR enabling the stepper and serial ISR inside the temperature ISR.
- Disable the stepper ISR completely inside the temperature ISR.
While the UART ISR is necessary to prevent lost chars, I added the stepper ISR "only" because it should provide a smoother step distribution especialy at high step rates (print speeds). The temperature ISR takes about 20µs, so that's the amount of time it can delay a step execution in worst case. There  will be a stepper ISR up to each 100µs (10kHz), so the delay is quite hughe if you state it in percentages: 20%!
On the other side as an advantage, this would reduce the complexity due to non-blocking ISRs and it would even eleminate the chance of temperature ISR re-entry where we need a boolean check flag at the moment. So this check could also be removed in this case.

Both solutions are easy, but I know ISRs is a topic with some pain for some people therefore I would like to get some feedback before creating a PR.

CONSULitAS, 2017-03-03:
@Sebastianv650 

Why not do both? Safety first.

thinkyhead, 2017-03-03:
I stated [here](https://github.com/MarlinFirmware/Marlin/commit/912704a0d85fdcdbe4f9783a9e007f6e83c11dab#commitcomment-21143032) that we seem to be hitting some kind of race condition, but that's not something I can really confirm. It's just the random nature of the freezes we've been seeing lately that has me thinking this is a possibility. Marlin doesn't have very complex "threading" but we still have to follow best principles. I can't confirm that these random freezes are related to interruptible-interrupts. It just seems one likely cause.

Sebastianv650, 2017-03-03:
@CONSULitAS What do you mean with both, removing stepper ISR and using the cli-sei thing around the babystepping? Possible, but I don't see an advantage of this configuration. The UART ISR is a very fast thing, or, in other words: how an ISR should be ^^ Therefore catching and storing an incomming char during a babystep wont do something.

@thinkyhead I'm used to that :) Since I started `LIN_ADVANCE` before it was known in Marlin nearly every person I talked about it and the problems I'm hitting stated I'm overloading the ATMega and such things - in the end of the day up to now it was always only a tricky bug (tricky for me, maybe not for a skilled developer).
But I also can't proove that it's not a race condition this time. So let's assume it is one and see if the bugs disapear. I will provide a PR, only allowing UART in stepper and temperature ISR. Temperature ISR will have a constant runtime again and extruder ISR can't be an issue anymore to re-enter temperature ISRs.

thinkyhead, 2017-03-03:
> overloading the ATMega and such things

I know we've already done a lot of offloading calculation to `Planner::_buffer_line` since Marlin 1.0.x so my hope is that the execution time of the stepper ISR in 1.1.x will be shorter than it is in 1.0.x., given the same configuration. Otherwise we will have some cases where printing at their old top speeds will invoke the watchdog.

As for the spurious freezes (often after several minutes of printing), that is more mysterious. When we see that, I tend to think "buffer overrun" or "stack overflow" — not specifically "race condition."

Too bad the stepper ISR cannot also handle babystepping. But, it only runs when there are real planner blocks, and I don't think we want to introduce a new planner block field (or flag) just for babystepping, necessarily. But that is another option, should we decide that the temperature ISR isn't the best place for it.

thinkyhead, 2017-03-03:
The freeze issue at #5660 was confirmed to be a stack corruption (due to a too large `BLOCK_BUFFER_SIZE` of 64). So, my faith in determinism is restored.

closed 2017-10-31 by thinkyhead