MarlinFirmware/Marlin #5952: How would I change the extruder axis input to degree?

killpop, 2017-03-04:
Hey guys, just a quick question. Is there any way that I can change the extruder axis input to accept degree command? I would like to use Marlin on my 4 axis CNC mill (currently on Grbl) and having my rotational axis recognize degree based commands would make my life a lot easier. I will be using a ramps 1.4 board. Ideally, I would want to keep XYZ in mm and just make E1 and E2 for degrees. Is this possible?

thinkyhead, 2017-03-04:
All you have to do is set the steps-per-unit for the extruder stepper based on degrees instead of millimeters.

For example, if you have a stepper with 200 full steps and 16 micro-steps, you have 3200 steps over 360 degrees. Divide by 360 to get the number of steps-per-degree. You can then set this in your configuration, or set it live with `M92 E8.888888` then save to EEPROM with `M500`.

You'll also need to change your E `DEFAULT_MAX_FEEDRATE` and `DEFAULT_MAX_ACCELERATION` so that the axis doesn't try to move too fast. Following our previous example, `M203 E180` would limit rotation to 180 degrees per second (30rpm). `M203 E360` gives 360 degrees-per-second (60rpm), etc. Set your max acceleration to something closer to a normal axis, such as `M201 E6000`.

The `DEFAULT_EJERK` may also need a change. For example, `M205 E2` will limit jerk to 2°.

killpop, 2017-03-04:
That's a lot simpler solution than what I though was going to need to be done. I wonder if it will make any difference if my CAM program starts spitting out any negative numbers or going from 350° to 10° and what the abs/inc expectation would be. Mastercam and their post processors can be weird like that sometimes. Also, I assume that I would need to disable the "advance" feature or any other kind of nozzle pressure control as these I think could lead to a slightly desynchronized movement; not too sure how these work. Are there anymore features like this in the code that are not immediately obvious that I need to look for?

thinkyhead, 2017-03-06:
If you keep your CAM program to using relative E movement, you should be ok. Of course if the axis should not be twerked past certain angles, then you'd need to add failsafes to the code to ensure that the angles are always within certain limits.

closed 2017-10-31 by thinkyhead