MarlinFirmware/Marlin #5954: Selecting a particular extruder

ssquinntran, 2017-03-04:
Kevin
How would you select a particular extruder to run? For example, picking extruder 2 instead of extruder 1 to run. I see that there is a T command, such as T2 that would specify which extruder, but nothing happened.
This is what my changes look like: 

Configuration.h:
#define EXTRUDERS 3

#define TEMP_SENSOR_0 1
#define TEMP_SENSOR_1 1
#define TEMP_SENSOR_2 1
#define TEMP_SENSOR_3 0

In pins_AZTEEG_X3_PRO.h:
#undef HEATER_0_PIN
#define HEATER_0_PIN     11 // Tie to Pin 11
#undef HEATER_1_PIN
#define HEATER_1_PIN     11 // Tie to Pin 11
#undef HEATER_2_PIN
#define HEATER_2_PIN     11 // Tie to Pin 11
#undef HEATER_2_PIN
#define HEATER_2_PIN     11 // Tie to Pin 11

Also, why would the extruder spin in different directions? I noticed that running the commands would have the extruder spin in these respective directions: 

g0 x17 e61 (clockwise)
g0 x20 e62 (clockwise)
g0 x30 e63 (counterclockwise)

Shouldn’t the extruder only spin in one direction? How would you fix this?

More specifically, what I want to do with the extruders is to push experimental materials out of a syringe. I know that in G0/1 XYZ E##, E is supposed to be extrude #amount of filament.

Bob-the-Kuhn, 2017-03-04:
Extruders are numbered 0, 1 & 2 so T0 command selects the first extruder, ...

G0 E25 should spin the extruder in one direction, G0 E-25 in the other.

I haven't heard of an extruder changing direction just by using a different number.  Is this reproducible?

ssquinntran, 2017-03-04:
We figured out the directionality of the extruder. Our remaining problem is we cannot figure out how to make extruder 1 (T1) and extruder 2 (T2) work. Our commands were:

g0 e10 t0 
g0 e20 t1

However, only the first extruder (T0) moved even though T1 was used. We attached the stepper too.

Bob-the-Kuhn, 2017-03-05:
They should be:
t0
g0 e10 
t1
g0 e20

ssquinntran, 2017-03-11:
@Bob-the-Kuhn We ran the commands on Pronterface, and this is what we saw:
SENDING:T0
echo:Active Extruder: 0
>>> g0 e10
SENDING:G0 E10
>>> g0 e40
SENDING:G0 E40
>>> g0 e60
SENDING:G0 E60
>>> g0 e80
SENDING:G0 E80
>>> t1
SENDING:T1
echo:Home XYZ first
No move on toolchange
echo:Active Extruder: 1
echo:Active Extruder: 0
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:Active Extruder: 1
echo:Active Extruder: 0
echo:busy: processing
echo:Active Extruder: 1
>>> t1
SENDING:T1
echo:Active Extruder: 1
>>> g0 e10
SENDING:G0 E10

We were able to call t1. However, the stepper for t1 never turned on. Both the stepper and the board work. Are you able to take a look at our config to see what could be wrong?

Configuration.h: https://github.com/kcheeeung/MarlinRC8git/blob/master/Marlin/Configuration.h
Configuration_adv.h: https://github.com/kcheeeung/MarlinRC8git/blob/master/Marlin/Configuration_adv.h
pins_AZTEEG_X3_PRO.h: https://github.com/kcheeeung/MarlinRC8git/blob/master/Marlin/pins_AZTEEG_X3_PRO.h

kcheeeung, 2017-03-11:
Hey Bob, a simple way of thinking about the problem is that we just want to understand how to use a dual/triple/multi extruders. But in our application, we are literally just spinning the extruder motors and that's it (the heating elements will be detached). We commented out the cold extrusion since temperature doesn't really matter.

Somehow, we cannot figure out what are the right configuration settings are.

Bob-the-Kuhn, 2017-03-11:
Attached are the modified files that should work for you if I understand what you want.
[updated files.zip](https://github.com/MarlinFirmware/Marlin/files/836077/updated.files.zip)

When Marlin starts up T0 is selected automatically.  

None of the steppers should have power applied to them until there is a movement command.

Marlin assumes there is a fan, heater and thermistor for every extruder.  You'll need to define these for each extruder.  Since they aren't functional you can assign them to a non-existent pin and re-use the real pins if needed.  You'll also need to assign, in configuration.h, a thermistor table to each of the used extruder slots.

Are you running a heated bed? I've assumed that you are not.

You're running single steppers for X, Y & Z so there's none of the tricky dual stepper issues/concerns/setup.

You can have only one extruder active at a time.  So **T1** followed by **G0 E10** will move the E1 stepper by 10mm.

Are all the extruder steppers the same?  If you need custom settings for each then enable **DISTINCT_E_FACTORS** and set all your movement setup defines appropriately.

Do you need to automatically change the head position when switching from T0 to T1 & back?  If yes there's a bunch of offsets that need to be defined.  Just give me the offsets & I'll add them in.

ssquinntran, 2017-03-12:
Thanks so much for your help, Bob! Your changes worked and we solved the offset problems.

As for the final question, in the firmware, is it possible to turn 2 extruder motors at the same time?

Bob-the-Kuhn, 2017-03-12:
In general, no but ...

If the steppers are identical,  always turn the same way & always the same amount then you can wire them to the same driver (if the driver can handle the current).

closed 2017-03-18 by ssquinntran

thinkyhead, 2017-04-02:
> I haven't heard of an extruder changing direction just by using a different number. Is this reproducible?

That's exactly what `SWITCHING_EXTRUDER` does, actually. But in that case there's a good reason.

> is it possible to turn 2 extruder motors at the same time?

It can be accomplished, but Bob's suggestion is the better one.

karakbot, 2019-04-19:
> That's exactly what `SWITCHING_EXTRUDER` does, actually. But in that case there's a good reason.
>
>
> It can be accomplished, but Bob's suggestion is the better one.

Is it possible to control each extruder stepper driver using G1/G0 along with X, Y movements ?