MarlinFirmware/Marlin #5960: heat bed temperature flickering

nikki-reprap, 2017-03-04:
I've just noticed  that the heat bed temperature is flickering  between 40 and 45c. I am using the latest bugfix on a ramps 1.4 with a reprapdiscount full graphic smart controller.

Bob-the-Kuhn, 2017-03-04:
Try enabling  PIDTEMPBED and then use the autotune command.  PID control is much better at keeping the temperature stable.

/**
 * M303: PID relay autotune
 *
 *       S<temperature> sets the target temperature. (default 150C)
 *       E<extruder> (-1 for the bed) (default 0)
 *       C<cycles>
 *       U<bool> with a non-zero value will apply the result to current settings
 */

nikki-reprap, 2017-03-06:
Thanks i've found it still does it after running M303 again. I will have a look at my hardware and my connections.

thinkyhead, 2017-03-06:
Try today's `RCBugFix`, also. It's always possible we accidentally fixed something.

Bob-the-Kuhn, 2017-03-06:
How fast are these variations?  Repetier Host and Pronterface have logging functions that will graph the temperature vs. time.  If the slope of the variations is a lot faster than the slope during the heat up period then the variations are not real.  We've had other reports of this.

Make sure SLOW_PWM_HEATERS is disabled.  If it is enabled then the heaters will only update every 8 seconds.

Long shot - you might also try speeding up the temperature control loop by changing SOFT_PWM_SCALE.  The default is 0 which is 8 updates per second which is usually way faster than needed for the bed.
```
    // SOFT_PWM_SCALE to frequency:
    //
    // 0: 16000000/64/256/128 =   7.6294 Hz
    // 1:                / 64 =  15.2588 Hz
    // 2:                / 32 =  30.5176 Hz
    // 3:                / 16 =  61.0352 Hz
    // 4:                /  8 = 122.0703 Hz
    // 5:                /  4 = 244.1406 Hz
```

Did you save your M303 results to EEPROM and/or copy them into Configuration.h?  Maybe the old/default values are being used.

nikki-reprap, 2017-03-07:
I've just checked and LOW_PWM_HEATERS is disabled. I will have a go with changing the SOFT_PWM_SCALE. 

thanks

nikki-reprap, 2017-03-07:
I have tried all the things you  suggested but nothing worked but then i spotted i have been using thermistor 6 instead of thermistor 1. It seems to have fixed the problem.

thanks

closed 2017-10-31 by thinkyhead