MarlinFirmware/Marlin #5961: CoreXY Won't safe home with Y_Max

sbcrumb, 2017-03-04:
I am running Marlin 1.0.2. My core Xy has my homeing in the left rear. X Min Z Min Y Max.
When I install a proximity sensor, enable auto bed leveling and do a G28 the Z will not probe in the center of the bed. it probes in the rear left corner. If I swap my X Y and swap Y to a Min endstop instead of a max it works as it should.  How can I get Safe homing to work?

http://pastebin.com/fa6kU3PZ

This is my conifg.

Bob-the-Kuhn, 2017-03-04:
I suggest updating to the latest RCBugFix and see if that fixes it.  

The biggest problem will be transferring your config items over.  Lots of changes and some of the names have changed.

sbcrumb, 2017-03-05:
Is that my only choice?

Roxy-3D, 2017-03-05:
It is worth the effort...   Grab a Visual Diff program (like Notepad++ with the comparison plug-in) and just cross over your existing configuration.h file.

sbcrumb, 2017-03-05:
Alright I will take a look at it.

thinkyhead, 2017-03-06:
There are some bugs in Marlin 1.0.2 associated with CoreXY (and many other things) which we know are fixed in `RCBugFix`, so that's the recommended course at this time. See http://marlinfw.org/docs/configuration/configuration.html for a more complete overview of all current options, and some advice on upgrading from earlier versions.

It's easiest to start with new configuration files and bring over just your customized settings, which you can determine by comparing your configs to the default 1.0.2 configs.

The new configuration files are mostly in the same order, but have some new options and changes. If you simply try to build `RCBugFix` using your old configuration files, first it will ask you to set a version (See http://marlinfw.org/docs/configuration/configuration.html#configuration-versioning) and if you fix that then it will throw a bunch of helpful errors about which options have changed. Following and fixing these errors can possibly get you to a point where Marlin will build, without needing to meticulously copy settings into the new configs. But in the long run, it's good to start with the new config files and set them up fully.

ozligia, 2017-03-07:
Confirm, that latest RC and RC bug fix can  home to Y_Max on CoreXY

closed 2017-10-31 by thinkyhead