MarlinFirmware/Marlin #5962: 32-bit version

mperdue, 2017-03-04:
I see there are 2 versions of the 32-bit rcbugfix. How is this coming along? How current is it compared to the features of the regular rcbugfix? I'm thinking about ordering a 32-bit controller card but I want to know what is missing before I take the plunge.

p3p, 2017-03-04:
Unfortunately, as it stands now, Marlin doesn't have a 32-bit version, at least not one that you could use without being able to program and being willing to put some work in on the platform you choose. There is no build system for general use, and as far as I'm aware the Due HAL layer that is in the 32-bit branches has never been tested.

Keep in mind that I say this with my printer currently printing using a relatively feature complete 32-bit branch so progress is being made just not in a way that you can probably use yet.

mperdue, 2017-03-04:
Thanks for the reply. The board I'm considering is the Azteeg X5 and it comes with smoothieware support. Obviously I could use that for now but I really like what I'm seeing with Marlin. I have no idea what the pros and cons of smoothieware are so I guess I need to do some homework. Ultimately I would prefer to run Marlin because the support here has been pretty good.

mperdue, 2017-03-05:
I think I've decided to hold off for awhile. Looking forward to a 32-bit release here.

closed 2017-03-06 by mperdue